Avnu/OpenAvnu #172: Gstreamer Configuration issues

DattuMP, 2015-01-15:
Hi All,

I am working Live stream in Open/AVB.
I have tried to configure the gstreamer to create AVB pulgins avbsink,avbplaybin.
As given instructions in the README file in gstream , creation of .so file is succeded and also I added the library file to the path.But when I use gstreamer inspect It's showing no avb sink element.

Step follwed on Gstreamer and results:
===========================
1.Changes made in Makefile.am in gstream plugin source location.

# Note: plugindir is set in configure

##############################################################################
# TODO: change libgstplugin.la to something else, e.g. libmysomething.la     #
##############################################################################

plugin_LTLIBRARIES = libp.la

EXTRA_FLAGS=-I ../../lib/igb/
EXTRA_FLAGS+=-L ../../lib/igb/ -ligb -lpci -lrt -lpthread -lm  -lgstapp-1.0

##############################################################################
# TODO: for the next set of variables, name the prefix if you named the .la, #
#  e.g. libmysomething.la => libmysomething_la_SOURCES                       #
#                            libmysomething_la_CFLAGS                        #
#                            libmysomething_la_LIBADD                        #
#                            libmysomething_la_LDFLAGS                       #
##############################################################################

# sources used to compile this plug-in
#For Talker uncomment the below:

libp_la_SOURCES = gstavbsink.c gstavbsink.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h

#For Listener uncomment the below:
#libgstplugin_la_SOURCES = gst_avb_playbin.c gst_avb_playbin.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h gst_app_src.c

# compiler and linker flags used to compile this plugin, set in configure.ac

libp_la_CFLAGS = $(GST_CFLAGS)
libp_la_LIBADD = $(GST_LIBS)
libp_la_LDFLAGS = $(GST_PLUGIN_LDFLAGS) $(LDFLAGS)
libp_la_LIBTOOLFLAGS = --tag=disable-static
libp_la_EXTRA_FLAGS=$(EXTRA_FLAGS)

# headers we need but don't want installed
noinst_HEADERS = gstplugin.h

2. Configuration of gstreamer.
           		       
builduser@builduser-HP-Elite-7100-Microtower-PC:~/Desktop/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin$ sudo ./configure
[sudo] password for builduser:
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
[... 146 lines cut ...]

     8.Checked the avbsink plugin is added or not facing below issue.

          builduser@builduser-HP-Elite-7100-Microtower-PC:~/Desktop/Open-AVB-open-avb-next /examples/gstreamer-avb-plugins/gst-plugin$ gst-inspect-1.0 avbsink
        No such element or plugin 'avbsink'
 
  Kindly let me know whether I was in right direction in this working scenario.
  Kindly reply to the given query which will forward the Live streamer demo.
  Kindly let me know whether this is known issue and provide any inputs to go further 
  working on this  issue.Thanks in advance.

NOTE:
======
 Also tried configuration and install using the commands given below.
1. LDFLAGS='-L/Open-AVB-open-avb-next/lib/igb' CPPFLAGS='-I/Open-AVB-open-avb-next/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' sudo ./configure
2. LDFLAGS='-L/Open-AVB-open-avb-next/lib/igb' CPPFLAGS='-I/Open-AVB-open-avb-next/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' sudo ./autogen.sh
3. export GST_PLUGIN_PATH=/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src/.libs

System information :
-------------------------
# uname -a
Linux builduser-HP-Elite-7100-Microtower-PC 3.13.0-43-generic #72-Ubuntu SMP Mon Dec 8 19:35:06 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

Regards,
Dattu

pcub, 2015-05-28:
Hi DattuMP. Did you resolve the problem that you met?

DattuMP, 2015-05-28:
No.I'm not working on that now..if you have the solution,please update it here.

pcub, 2015-05-29:
I compiled the configue and autogen using CLIBS instread of LIBS. That worked for me. But I dunno that is right or wrong.

pinealservo, 2015-08-11:
Closed due to inactivity.

closed 2015-08-11 by pinealservo

rohan-lt, 2015-08-12:
We are working on Open-AVB-open-avb-next source code for a gstreamer plugins.
We have a two different pc with ubuntu 14.04 and NIC card(Intel i210) in both the system.

LISTENER side (WITHOUT SUDO)

1)Go to folder: cd /Open-AVB-Open-AVB-Next/examples/gstreamer-avb-plugins/gst-plugin/src 

and in Makefile.am file set
libgstplugin_la_SOURCES = gst_avb_playbin.c gst_avb_playbin.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h gst_app_src.c

2) LDFLAGS='-L/home/pcadmin/Downloads/AVB/lib/igb' CPPFLAGS='-I/home/pcadmin/Downloads/AVB/lib/igb' LIBS='-ligb -lpci -lrt -lpthread   -lm' ./configure

pcadmin@EMMYS1776:~/Downloads/AVB_fresh/examples/gstreamer-avb-plugins/gst-plugin$ LDFLAGS='-L/home/pcadmin/Downloads/AVB_fresh/lib/igb' CPPFLAGS='-I/home/pcadmin/Downloads/AVB_fresh/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm -lgstapp-1.0' ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert i686-pc-linux-gnu file names to i686-pc-linux-gnu format... func_convert_file_noop
checking how to convert i686-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
[... 888 lines cut ...]

1)Go to folder: cd /home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin/src
and in Makefile.am file set
libgstplugin_la_SOURCES = gstavbsink.c gstavbsink.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h

2)sudo LDFLAGS='-L/home/pcadmin/avb/avb_codeflow/lib/igb' CPPFLAGS='-I/home/pcadmin/avb/avb_codeflow/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' ./configure

3)sudo LDFLAGS='-L/home/pcadmin/avb/avb_codeflow/lib/igb' CPPFLAGS='-I/home/pcadmin/avb/avb_codeflow/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' ./autogen.sh

Now type 'make' to compile this module.

4)sudo make

5) export GST_PLUGIN_PATH=/home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin/src/.libs

6)Check weather gstreamer has installed the plugin or not:
gst-inspect-1.0 avbsink

root@pcadmin-OptiPlex-7010:/home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin# sudo gst-inspect-1.0 avbsinkNo such element or plugin 'avbsink'
root@pcadmin-OptiPlex-7010:/home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin# 

kindly help us what is missing or what need to be fixed in it. 

Thanks and Regards
Rohan

rohan-lt, 2015-08-13:
Hi all

the above mentioned problem is resolved using sudo -i.
talker is able to transmit the data using avbsink.

but on listener side we are using avbplaybin and it is not recieving any file, AS code is running but it is stuck at set state returned 2.

root@EMMYS1776:/home/pcadmin/Downloads/AVB_fresh/examples/gstreamer-avb-plugins/gst-plugin# gst-launch-1.0 avbplaybin talkerip=192.168.35.48 port=7600 interface=eth2
Setting pipeline to PAUSED ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
attaching to 0000:03:00.0
set state returned 2

kindly help us what is missing or what need to be fixed in it.

Thnaks & Regards
Rohan Gupta