Avnu/OpenAvnu #173: jack talker and listener application !

SumeetPJain, 2015-01-15:
Hi,

Can anyone shed some light on how actually the jack talker and jack listener applications can be ran ?
I am getting this prints constantly:-
Available writespace: XXXX
I guess may be I have to plug in some audio source into the IN ports.
Secondly I tried using 'zynaddsubfx' application to generate audio and feed it to the 'simple_talker' client using 'QjackCtl' by using the connect options in it. But still couldn't make it work.
I am attaching a snapshot of the experiment I did. 

If anyone can elaborate over this issue ? It will be really helpful.

[image]

SumeetPJain, 2015-01-19:
And one more observation is that for 'jackd_talker', the 'igb_get_wallclock' is failing that can also be seen the above snapshot. But for 'simple_talker' app, it works fine. 
Can anyone update me regarding the same ?

Thanks,
Sumeet

sumeet-jain-engineer, 2015-01-20:
I am now able to transfer audio packets from the talker hence closing the issue !

grandcat, 2015-01-20:
How did you fix the problem?

sumeet-jain-engineer, 2015-01-20:
There are some changes I made in the 'jackd_talker' application. They are:-
1) With the master package, my xmit was failing which I fixed.
2) As you can see in the above snapshot that 'Input Port' needs the data to be fed from 'Output Port' and for that you need to establish a connection between them for which I initially used 'qJackCtl' utility. Later I implemented this (establishing connection between them) using jack standardized API's in the 'jackd_talker' application itself.
After this, it worked tickety-boo !!

davidcemin, 2015-02-26:
By reading the comments i understand that this issue can be closed. Is that right ?

sumeet-jain-engineer, 2015-02-26:
Ya sure !

closed 2015-02-26 by andrew-elder