Avnu/OpenAvnu #181: igb_get_wallclock() does not measure PCIe read duration correctly

Matt-Edwards, 2015-01-29:
igb_get_wallclock measures the time it takes to read the NIC's timer register.  A PCIe read can take anywhere from 1us to 20us or more. igb_get_wallclock calls rdtscpll() to read the cpu's performance  timer.  rdtscpll() does not adjust for frequency of the CPU timer and it units will be more in the nanosecond range.  I believe the loop want's the read to be less than 80us.  The result is that the loop runs 32 times.

eric-mann, 2015-01-29:
The intent isn't to measure the latency of pci reads in nsec ... the intent is to return a cross timestamp of the wallclock vs the IA TSC counter (?).

Matt-Edwards, 2015-01-29:
OK, thanks for the clarification.  But is it true the loop will run 32 times?

christopher-s-hall, 2015-01-29:
This is true.  It returns the cross timestamp with the smallest TSC “spread”.  Previously, this code, ran *up to* 32 times and exited the loop when some threshold was reached.  The threshold was a fixed number of TSC cycles which, in testing, never seemed to scale across different CPU types.

Matt-Edwards, 2015-01-29:
Yes it would be difficult to find the right number it you don't consider the performance counter frequency.  I just tested a write to the tsauxc register on a 3.4Ghz i5 (unlocked) and it takes about 130ns which is 444 counter cycles.  Maybe  the loop is not needed.  

With a PCIe write does the CPU wait for a completion from the i210?  In other words, is the timer really latched when the routine read the CPU timer the second time?

eric-mann, 2015-01-29:
RDTSC isn't a serializing instruction ... so you can either do a memory fence - ie read a PCIe MMIO register (which adds MORE latency and jitter) or do a CPUID instruction which is serializing.

christopher-s-hall, 2015-01-29:
Currently, on Linux, I'm using lfence to serialize the rdtsc instruction.  According to the SDM: " If software requires RDTSC to be executed only after all previous instructions have completed locally, it can either use RDTSCP (if the processor supports that instruction) or execute the sequence LFENCE;RDTSC"  I found that this was faster than using CPUID.

The code is:

rdtscpll(&t0); // lfence;rdtsc
E1000_WRITE_REG(hw, E1000_TSAUXC, tsauxc );
rdtscpll(&t1); // lfence;rdtsc

Is there a problem with this logic that I'm not seeing?

Matt-Edwards, 2015-01-29:
The code and logic are good.  My question was more about hardware.  I should have asked, 'when E1000_WRITE_REG() returns, is the write request still travelling down the PCIe SERDES wire?'  PCIe reads take from 1 to 3us on my system but the write took 130ns.  I assume a read is more costly because you are waiting for something back but difference between a PCIe read and write times is much more than double. My concern was that the latch of the the system time into the aux time happens sometime after E1000_WRITE_REG() returns.   Sorry, I just don't know enough about PCIe transactions.

christopher-s-hall, 2015-01-29:
Hopefully Eric will weigh in, but I think a read to the same device will guarantee that the write is completed.  This flush macro already exists to do this.

rdtscpll(&t0); // lfence;rdtsc
E1000_WRITE_REG(hw, E1000_TSAUXC, tsauxc );
E1000_WRITE_FLUSH(hw)
rdtscpll(&t1); // lfence;rdtsc

eric-mann, 2015-01-29:
To answer Matt's original question - the memory writes to MMIO registers on a PCIe device are often 'posted writes' - which means the write it sent down to the PCIe fabric and a separate block within the CPU awaits the completion notification. The CPU execution from software's perspective continues to execute without waiting for the completion.

There are only a fixed number of these posted IO buffers available on the CPU, with server (Xeon) CPUs supporting many more outstanding IOs than client CPUs. Once the CPU runs out of posted buffers, the instruction pipeline stalls until a completion frees up a posted IO buffer.

To ensure the IOs are actually delivered/completed to the IO device, there are various CPU methods to enforce serialization. You can do the 'WRITE_FLUSH()' macro which does a read to the MMIO space - the CPU does not allow a read to complete out of order from a write. This is also portable across CPU architectures (e.g. Intel, ARM, Power, etc). Otherwise, you use native instruction-set methods like IA's CPUID, or Power's 'EIEIO' (enforce in-order execution of IO), or ...

Downside to the flush to a discrete PCIe device however is you must complete 2 PCIe TLPs to bracket the CPUs TSC. The TLP completions themselves can be highly variable, which impacts your cross-timestamp accuracy.

Matt-Edwards, 2015-01-30:
So considering worse case and the timer isn't latched anywhere near when we thought...  Would it be better to measure after the first read of the AUX timestamp?  If I understand correctly the read would force the write to complete. Sure, we have the delay of both the write and read, but we can measure and know when we should ignore the cross-timestamp.