Avnu/OpenAvnu #183: IPG issue with 100Mb/s link Speed

Hrusikesh4302, 2015-02-03:
Hi,

    I am trying to run simple_talker and simple_listener application with 100Mb/s link speed. I am using DSP4YOU(Marvell) 100Mb/s AVB switch and once auto negotiation completed I can see all the nodes set themselves to 100Mb/s speed. Now when I run simple_talker and simple_listener, I face the following issues...
- The IPG between 1722AVTP packets is  ~1.25msec which in idle case should be .125msec
- If I connect talker, listener one to one with out the AVB switch and force the link speed to 100Mb/s with the help of ethtool command; still I am facing the same 1.25msec IPG Issue.

    Does I have to modify the Qav control register values for 100Mb/s speed ? Please let me know if I am missing anything.

Regards
Hrusikesh

intel-ethernet, 2015-02-03:
I am assuming you have high confidence in the accuracy of your timestamps to measure the IPG  (e.g. you have turned off interrupt moderation on the receiving interface and running wireshark on a relatively modern CPU/platform) … the igb library should compensate for the link speed when programming the Qav shaper (it scales everything down by ’10’ if linkspeed is 100Mbps).

My first thought is to verify you have sufficient bandwidth to transmit the workload – how big are the 1722AVTP frames are you sending and at what rate (Class A ~ 125 usec?).

The other option is disable the Qav shaper completely and let the timed transmission feature handle the pacing. So long as software doesn’t attempt to send more than allocated, the Qav traffic shaper should never activate to back pressure.

salehsaleh, 2015-03-10:
gooood

Hrusikesh4302, 2015-03-11:
Hi,
 
Thanks for your quick responses, I have some updates on this. 

To use with 100Mbps speed I slightly modified in igb.c, and I am able to send the packets with the desired rate(125uSec). I am not scaling down the linkrate/10 in line no 1331 as below in igb.c, which actually increases the bw reservation. But I am not sure how much bw reservation happening now, can you give me some idea how to check that. But other scaling downs are still enabled, so it is certainly less than the bw reserved for 1000Mbps linkrate. 

1333    //	if (link.speed == 100) 
1334   //	    linkrate = E1000_TQAVCC_LINKRATE / 10;
1335   //	else
1336	    linkrate = E1000_TQAVCC_LINKRATE;

     Which I have modified to match line no 1359/1360 with  page-313 I210 spec Qav configuration.
1359  class_a_idle = (u_int32_t)(class_a_percent * 0.2 * (float)linkrate + 0.5);  
1360  class_b_idle = (u_int32_t)(class_b_percent * 0.2 * (float)linkrate + 0.5);

IdleSlope Configuration is calculated using the following equation:
100Mbps: BW * 0x7735 * 0.2
1000Mbps: BW* 0x7735 * 2   
As per spec only difference is 100/1000Mbps is 0.2/2 but linkrate credits i.e 0x7735 should be constant.  Please reply if I am doing any thing wrong. 

@intel-ethernet 
 
1> Yes, we have modern CPU platform and we are able to receive the packets at 1000Mbits/sec speed. So I think platform and interrupt moderation is not the issue in our case. Also if you can look at the IPG jump from 125uSec -> 1250uSec,  the bw reservation( more into scaling downs) looks suspicious. 
2>  Our pkt sz is 104 and priority class A.
3> I have not disabled the Qav for 100Mbps speed, but I am sure with Qav disable packets are going to flow with out back pressure as I have seen it with 1000Mbps speed.

Thanks
Hrusikesh

MarcinMiklasSTC, 2015-08-18:
I'm also using OpenAVB with 100Mbps link speed and seen the same error. 

My workaround for this issue was giving 10 instead of 1 to the igb_set_class_bandwidth() class_a or class_b parameter. It was working until I needed small percentage of bandwidth. Because otherwise the following check was triggered:

	if ((class_a_percent + class_b_percent) > 0.75) {
		error = EINVAL;
		goto unlock;
	}

Your patch is much better. Can you submit it to the OpenAVB?

Hrusikesh4302, 2015-08-21:
Hello,

     I am not a member of OpenAVB so I am not sure I can commit the patch or not. However I have tried with commentinf the following lines in  Open-AVB-master/lib/igb/igb.c file and the issue was solved I guess,

     //   if (link.speed == 100)
      //      linkrate = E1000_TQAVCC_LINKRATE / 10;
     //   else
            linkrate = E1000_TQAVCC_LINKRATE;
   We have not checked exactly 8000 frames/sec are delivering on listener or not. But it was near around that. Please let me know any other info I can provide.

Thanks
Hrusikesh

andrew-elder, 2015-08-24:
@Hrusikesh4302 - anyone can submit a pull request to fix a specific issue. There is a pending pull request that I suspect will take care of some 100 Mbps issues.