Avnu/OpenAvnu #186: IPG issue with simple_talker application

sumeet-jain-engineer, 2015-02-19:
Hi,

There is one observation while using the simple_talker application of the latest package related to the Inter packet Gap. We are using wireshark tool at the receiver side to monitor the received AVB packets.
Our observation is:-
--> In every ~248210 AVB packets, we are getting a delayed packet with the IPG of ~0.5 second.
--> This pattern (IPG of 0.5) keeps repeating after every ~248210 AVB packets
--> Apart from this, the IPG of every other AVB packet shows 127 usec instead of 125 usec.
--> We believe that these observation are relevant to section 7.2.2.2.3 in I210 spec.
--> We tried to understand the reason behind this but failed to develop any concrete theory to support this.

We need help to understand the above behaviour and how actually the launch time concept works.

Regards,
Sumeet Jain

intel-ethernet, 2015-02-19:
Sumeet –

“Our observation is:-
--> In every ~248210 AVB packets, we are getting a delayed packet with the IPG of ~0.5 second.
--> This pattern (IPG of 0.5) keeps repeating after every ~248210 AVB packets”

I thought this was fixed in the latest version of simple_talker … but the typical root cause for this behavior is the host software (and clock) gets out of sync with the I210 clock.

First I have to explain how the timed transmissions work. You can schedule a packet up to 0.5 second “in the future”. You can also schedule a packet to be sent right now (specifying a time which happens to be ‘now’). You can lastly schedule a packet “late” – up to 0.5 seconds in the past. Once a packet transmit time appears far enough in the past (more than 0.5 second) it becomes aliased to “the future” again.

You typically run into this weird behavior as follows:

a)      SW fetches the current LAN clock time (t_now)

b)      SW begins a loop transmitting frames every “125 microseconds” (e.g. t_now + 125*1, t_now + 125 * 2, t_now + 125 * 3, …)

c)       Unknown to SW, the I210 LAN clock runs “a little bit fast” (or slow) relative to software. The actual rate depends on device and temperature. Typically, the gPTP stack would/should act to adjust the frequency of the LAN clock to prevent the host CPU time and LAN time from drifting apart … but if gPTP doesn’t, software has to do the conversion between software’s idea of time and the current LAN clock.

d)      Without frequency compensation, SW runs long enough that the difference between what SW thinks ‘now’ is differs by more than 0.5 from what the LAN clock thinks ‘now’ is – transmissions halt for 0.5 second, which then acts to resync talker SW with the LAN clock (since transmissions stop – the tx queue fills up - and simple talker just spins waiting to resume).

--> Apart from this, the IPG of every other AVB packet shows 127 usec instead of 125 usec.

Not sure what “every other AVB packet” exactly means, but if the IPG looks like 127 usec instead of 125, sounds like the I210 LAN clock is running ‘really slow’. Normally I’ve only seen worst case 400 nsec drift / sec, whereas 2 usec * 8000/sec ~ 16 msec/sec drift, but I guess it is possible with crystal variation and ambient temperature … not sure if the Qav traffic shaper is also kicking in to throttle the transmit rate as well – may want to turn it off altogether to see how it affects the measurements.

Hope this helps.

ekm

Regards,
Sumeet Jain

sumeet-jain-engineer, 2015-02-24:
Thanks @intel-ethernet for the valuable response.
I tried one experiment in which I turned off CBS and observed IPG = 125usec.

Regards,
Sumeet Jain

intel-ethernet, 2015-02-24:
This suggests you didn’t reserve enough bandwidth with the credit based shaper on the I210 transmitter side.

ekm