Avnu/OpenAvnu #189: gPTP H/W timestamps don't account for PHY delay

andrew-elder, 2015-02-26:
i210 hooked up to Extreme switch with 6ft cable shoes neighbor delay of 500ns. This is too large. Perhaps we don't account for the PHY delay correctly?

christopher-s-hall, 2015-02-26:
Which OS are you using?  The PHY delays are currently set in the OS specific code.  If you're using Windows, take the I210 PHY delay numbers (they're default) from the Linux code.  In my opinion, compensation for PHY delay should be the business of the driver not gPTP.  As far as I know, in Linux, compensation is not done.

jdkoftinoff, 2015-02-26:
I agree that the compensation of the PHY delay should be responsibility of the driver, but would that mechanism work when the link is re-negotiated at 100baseT vs GigE?

intel-ethernet, 2015-02-26:
I thought we did apply the compensation offsets within the driver … but I’ll check … is this with a particular link speed?

ekm

andrew-elder, 2015-02-26:
I'm sure the linux driver is correct. We have our own driver running under Windows. Sounds like this is a driver issue we need to take care of ourselves. I'll review and most likely close this issue out.
  
 - Andrew
  
  
  

----------------------------------------

christopher-s-hall, 2015-02-27:
I don't know about the I210.  The last I knew (testing at UNH), the driver did not compensate for PHY delay.  The PCH driver (built-in Ethernet) definitely is pretty far off.  I'm sure that it doesn't.

This doesn't help Andrew because he isn't using Linux.  However, the I210 copper PHY numbers from the datasheet are #defined in the Linux code.

To Jeff's question:  I may be missing something, but wouldn't the network driver be aware of the link speed (and any changes) and compensate for the PHY delay appropriately?

jdkoftinoff, 2015-02-27:
> On Feb 26, 2015, at 17:28, Christopher Hall <notifications@github.com> wrote:
> 
> I don't know about the I210. The last I knew (testing at UNH), the driver did not compensate for PHY delay. The PCH driver (built-in Ethernet) definitely is pretty far off. I'm sure that it doesn't.
> 
> This doesn't help Andrew because he isn't using Linux. However, the I210 copper PHY numbers from the datasheet are #defined in the Linux code.
> 
> To Jeff's question: I may be missing something, but wouldn't the network driver be aware of the link speed (and any changes) and compensate for the PHY delay appropriately?
> 
Yes, I would expect it to. But it is a common implementation problem - Using the wrong PHY numbers for the negotiated speed because the speed changed.

Jeff

andrew-elder, 2015-02-27:
No further information required at this time. Thanks for all the input.

closed 2015-02-27 by andrew-elder

intel-ethernet, 2015-02-27:
Apparently we only added the PHY compensation code to the I210 MacOS driver … but Chris was right, the delays are documented in the public datasheet and are link-speed dependent.

ekm

andrew-elder, 2015-02-27:
Is the #define called TX_PHY_TIME?
And is the units nanoseconds?

I don't see any i210 specific #defines. Can only see
#if 0
#define TX_PHY_TIME 184
#define RX_PHY_TIME 382
#endif
#define TX_PHY_TIME 8000
#define RX_PHY_TIME 8000
in daemons\gptp\linux\src\linux_hal_generic.cpp (couldn't find anything in 
the linux drivers in Open-AVB).

The i210 PHY has 8000 ns delay in each direction?

christopher-s-hall, 2015-02-27:
Those are the right numbers for I210 copper.  The units are nanoseconds.

jdkoftinoff, 2015-02-27:
8000 ns for both directions for both gigE and 100baseT? that is very interesting…

Jeff

andrew-elder, 2015-02-27:
Are the numbers derived from
7.8.3.1 Capture Timestamp Mechanism
of the "i210-ethernet-controller-datasheet.pdf (v2.6) somehow?

Where are the 1G tables?

intel-ethernet, 2015-02-27:
Yes … its table 7-62 of http://www.intel.com/content/dam/www/public/us/en/documents/datasheets/i210-ethernet-controller-datasheet.pdf ... Although I will need to check why they took the 1G latencies out of the table (an earlier version of the document had them listed).

You will notice two items:

a)      The rx offsets are different from the tx offsets

b)      The Min/Max represents the jitter

To Jeff’s comment - I have no idea where 8 usec came from …

ekm

intel-ethernet, 2015-02-27:
The 1G compensation offsets we used (in the MacOS driver) were:

1G Tx: 178ns

1G Rx: 448ns

… although this doesn’t comprehend jitter, which depending on the application influences guard bands you may need to implement. It typically doesn’t come up in AVB usages, but it does in industrial Ethernet ones.

Back to your original question Andrew … 6 ft ~= 2 m, and 1 nsec = 0.7*30cm (over copper cabling) => expected delay of ~ 9.5 nsec due to the cable. So your original measurement (~500 nsec)  seems close to expected values (9.5 + 448 +/- jitter) @ 1Gbps.

Ekm

From: Mann, Eric
Sent: Friday, February 27, 2015 8:47 AM
To: AVnu/Open-AVB; AVnu/Open-AVB
Subject: RE: [Open-AVB] gPTP H/W timestamps don't account for PHY delay (#189)

Yes … its table 7-62 of http://www.intel.com/content/dam/www/public/us/en/documents/datasheets/i210-ethernet-controller-datasheet.pdf ... Although I will need to check why they took the 1G latencies out of the table (an earlier version of the document had them listed).

You will notice two items:

a)      The rx offsets are different from the tx offsets

b)      The Min/Max represents the jitter

To Jeff’s comment - I have no idea where 8 usec came from …

ekm

andrew-elder, 2015-02-27:
Ok, thanks Eric.