Avnu/OpenAvnu #192: IGB: event-based refill of xmit buffer

grandcat, 2015-03-24:
Hey all,

this issue is more a question than a feature request. 
As far as I stepped into the details, all examples use the following low-level mechanism to keep the send buffer filled up (please correct me if I'm wrong):

1. Reserve DMA space on I210 Ethernet card and create linked list based on `igb_packet` struct to manage the slots of this transmit buffer.
2. Set transmission time using the field `attime`. Enqueue prepared packets for transmission using `igb_xmit(...)`.
3. Try to gather already sent buffer slots using `igb_clean(...)`. This function is called in a busy-waiting loop until free space is found to enqueue some new packets. Goto 2.

So, my question is whether there is something in the driver to allow an event-based processing instead of probing with a busy loop?
Is there a signal to subscribe to in order to be notified if free slots in the tx-queue is available?

If not, are there any plans for a basic approach to realize this?

Thank you very much, guys =)

intel-ethernet, 2015-03-24:
There are a few methods for user-mode signaling. Ultimately, an interrupt must be fired from the device (either transmit done, or low-descriptor-threshold hit, or …), handled in the kernel, and driver code <does something> - typically set the state of a semaphore or socket / filedescriptor, which wakes a thread from select() – or somehow you hack the kernel to raise a signal to the process (but that isn’t something standard drivers do directly).

You could also take a completely difference approach and extend the skb on the standard Linux socket buffers to pass the ‘attime’ parameter and use the standard sockets APIs to implement something more clean using standard APIs.

However – I haven’t taken a formal poll of avb users whether they like the entire user-space busypoll() method better than figuring out some sort of eventing vs making AVB transmits more of a ‘standard’ Linux kernel feature. Years ago, the plan was to push patches to make it a standard socket option, but several well informed others in AVnu pointed out that high performance AVB applications with real-time processing cannot easily use sockets – not because of latency, but because the system call itself drops the priority of the thread.

Other comments/opinions welcomed.
ekm

pinealservo, 2015-09-30:
It has always seemed to me that AVB streams need a different interface to the kernel than via the standard networking stack. I am not very familiar with the details myself, as most of my AVB implementations have been on non-Linux systems, but perhaps the interfaces of drivers like ALSA and FFADO could provide some inspiration?

xcalger, 2015-10-07:
@pinealservo, I am thinking the same. Ethernet AVB devices are a network device and an audio device. Networking for connection, but the dataflow will require something like an audio driver.

eric-mann, 2015-10-15:
There are various methods to signal a user-mode process about an event. For example, the I210 (as like many other Intel adapters and I'm sure other vendor's LAN solutions) generates an interrupt on a variety of conditions when the transmit buffers are available to be cleaned up. Its usually "I've written back X descriptors" or "My tx queue is almost full" or somesuch. 

So you get an interrupt in the kernel mode driver. On this interrupt, the driver can set an event or memory location. You can generate a message (e.g. netlink on Linux). Or if you are really latency sensitive, user-mode software polls the completion status on the transmit descriptors themselves (as done with the Intel DPDK libraries and associated polled-mode-drivers). Ultimately its an architectural decision how much you want to spend on CPU utilization to minimize latency. A thread can spin at 100% CPU utilization waiting for a memory location to transition from 0->1. A thread can be scheduled as 'runnable' by the scheduler based on an event or signal (higher latency, less CPU utilization). A thread can also process a message from the kernel (even less CPU utilization, but yet higher latency). 

My other observation is some applications like direct user-mode access to the transmit queues of the LAN (to minimize latency), and others are quite satisfied with transmitting packets on traditional sockets (either depending on a HW-based Qav shaper to pace out the traffic by class, or passing along a launch time somehow with the packet). I can't say I've seen a consensus form yet.

my 2 cents.

closed 2016-12-22 by grandcat