Avnu/OpenAvnu #194: Question: What ethernet controller cards are supported by openAVB?

VinodTadvai, 2015-03-26:
Apart from I210 etherent controller card, what other ethernet cards are supported by openAVB?

I see that simple_talker is dependent on ethernet card driver (igb_avb). Is gPTP and mrpd implementation also dependent on the driver?

andrew-elder, 2015-03-26:
Only the Intel i210 at present. We are waiting for other vendors to step up.

mrpd and gPTP under linux are not dependent on the i210.

GuillaumeLeNost, 2015-03-26:
Are there any plans to support Intel i211 / i218 ?

intel-ethernet, 2015-03-26:
Unless you only want to use them as a listener, not a talker…. (all Intel adapters at this point have 802.1AS / 1588 support). But the I210 is the only available model with Qav and timed transmission required for a talker application.

ekm

closed 2015-04-17 by andrew-elder

MythiliR, 2015-07-22:
How to enable the Hardware Queueing (802.1 Q FQTSS)in linux kernel ? Kindly suggest me