Avnu/OpenAvnu #197: Setup with back-to-back I210 ethernet controller cards

VinodTadvai, 2015-03-30:
In first phase, we want to connect 2 desktop system with I210 ethernet controller cards back-to-back without a switch. We will utilize gPTP, SRP, simple talker and listener from openAVB. Talker will be running on one desktop and listener on other desktop. Any issues with such a setup where there is no switch between the endpoints?

At a later point, we will buy COTS AVB switch. What AVB swtiches are used in openAVB testing? Are there any known AVB switches that are known to have issues with openAVB running on endpoints?

intel-ethernet, 2015-03-30:
Back-to-back mode should work for a ‘simple’ configuration (like a demo) - although you won’t be able to test all permutations which can arise with a switch.

I believe any AVnu certified switch will work with the Open AVB stack.

Hope this helps.
ekm

closed 2015-04-17 by andrew-elder

marvinyane, 2015-12-30:
Hi VinodTadvai,
    Have you ever met any issues in this mode?

    I find that the MSRP do not work in this mode, the simple talker or listener cannot find the domain. 

Thanks.