Avnu/OpenAvnu #199: [Open-AVB] stream reservation in the network

swethachallawar, 2015-04-02:
in the simple talker and simple listener examples, I am trying to understand how the stream reservation is done in the network and what are the mrp calls made for achieving the stream reservation between talker and listener

i see that talker calls igb_set_class_bandwidth() for bandwidth calculation on the host. Apart from this, can someone please explain if there are other mrp calls to reserve stream in the network?

andrew-elder, 2015-04-02:
See also
https://github.com/AVnu/Open-AVB/blob/open-avb-next/examples/simple_talker/simple_talker.c#L757

swethachallawar, 2015-04-03:
Thanks for the response.
So I understand that once the talker has advertised the stream , it is the mrp daemon that reserves resources on the network.
Does the listener also need to invoke any calls for stream reservation before sending a ready message? Please let me know how listener is allocating necessary bandwidth before sending out a ready message.

intel-ethernet, 2015-04-03:
The flow is best described in the AVnu whitepaper which describes the back-and-forth between talker<->MRP<->listener.

http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf > Fig 16 & 17.

Hths.
ekm

swethachallawar, 2015-04-03:
Thank you. 

Can the simple talker and simple listener example here, be used for a use case where there are multiple talkers and a single listener/multiple listeners?

intel-ethernet, 2015-04-03:
Not sure what is being asked – can the ‘simple_talker’ and ‘simple_listener’ examples be extended to handle more than (1) stream?

If so – yes. You will need to figure out how to interleave transmits from multiple sources on the local endpoint onto the same hardware queue (potentially) – with appropriate synchronization (locking). Similar on the listener side.

Starts to become the not_so_simple_talker example.

ekm

swethachallawar, 2015-04-09:
Thanks for your response.

the spec states that "If there are sufficient resources and bandwidth available MAP will declare a Talker Advertise. Otherwise, MAP will declare a Talker Failed on the outbound port and add appropriate FailureInformation."

Can you please clarify whether changing from Talker Advertise to Talker Failed is being handled in the mrp code in Open AVB? if so, based on what checks are we changing from Advertise to Failed?

I can see that the below function is setting Talker Failed. 
msrp_emit_talkervectors(mrpdu_msg_ptr, mrpdu_msg_eof, &bytes, lva, MSRP_TALKER_FAILED_TYPE);
but I do not quite understand the criteria based on which we are sending out this failed. Can you please clarify?

andrew-elder, 2015-04-09:
The mrp daemon will not automatically change a TalkerAdvertise to a TalkerFailed. The mrp client is responsible for declaring a TalkerFailed attribute when required.

The msrp_emit_talkervectors() call scans the attribute database and emits vectors of the type specified. So if the type is TALKER_FAILED, only talker failed vectors are emitted by this call. The emit operation only scans the database, it does not make any declarations.

Hope this helps.

intel-ethernet, 2015-04-09:
I think the AVnu design guide - http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf - specifically section 5.2<http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf%20-%20specifically%20section%205.2> – describes talker failed behavior. In general, the switch(es) convert Talker Advertises into Talker Failed types depending on available bandwidth and number of concurrent reservations. Similarly, switches can modify Listener Ready types to reflect if all or only some listeners can receive the stream.

You can certainly adapt the Open AVB mrp daemon to emulate these behaviors - it would be a hardware-specific modification.

ekm

swethachallawar, 2015-04-10:
Thank you so much for the response.

please confirm when you say mrp client, you are referring to the talker mrp client and the listener mrp client modules? 

I that case, after bandwidth calculations (igb_set_class_bandwidth call), should the talker add failure information if any, before advertising the stream (mrp_advertise_stream)? that is add the info while publishing the S++ steam?

Avout the set class bandwidth call,  the Avnu design guide says that "This function could be called by an MRP service – as SRP has global knowledge of the various stream reservations on each of the SR classes (A & B). However – common implementations have a single Talker instance, which would also know about global bandwidth allocated per class as well"
I can see one call to igb_set_class_bandwidth  being made in simple talker. should be ideally called every time we check the bandwidth availability of each bridge/endpoint? Please clarify

Thanks
Swetha

swethachallawar, 2015-04-13:
I am looking for the above info. Can someone please help me with understanding the above design aspect of the mrp? that is the specific mrp calls which check the availability of bandwidth on the network and accordingly set TalkerAdvertise or TalkerFailed based on the availability

Thanks
Swetha

andrew-elder, 2015-04-13:
Existing mrpd does not make any decisions regarding TalkerAdvertise or 
TalkerFailed. The current design expects the logic for making that decision 
to exist in an external application/module.

swethachallawar, 2015-04-13:
Ok. Thank you so much for the clarification.

Please let me know in case there are any applications already available which have this logic in place. 

Thanks
Swetha

swethachallawar, 2015-04-15:
In the above mail, I understand that the mrp client referred to is talker application/listener application. Please confirm. 

I have one more query. Supposing my AVB set up has two PCs (with I210 cards) connected via AVB compliant switch and assuming that the openAVB stack is being run on a switch, I would like to know how would the availability of bandwidth be checked on the switch. Is there a logic already existing in openAVB for the same or do we need to develop an mrp client application which would perform bandwidth availability checks on the switch? Please clarify. 

Thanks
Swetha

jdkoftinoff, 2015-04-15:
Each switch in the network does the bandwidth checks because the available bandwidth can be different at each hop. 

If there is not enough bandwidth for the specific listener, the switch transforms the Talker Advertise attribute into a Talker Failed attribute and sends it to the listener. The talker would receive either a Listener Asking Failed attribute or a Listener Ready Failed attribute.

Since streams are multicast, it is possible for there to be enough bandwidth for a stream to one listener down one path in the network, but not enough for another stream.

If you are implementing an end station that does not have bridge functionality then you would not do bandwidth checks besides looking at LAF/LRF/TF

Regards,
Jeff

swethachallawar, 2015-04-16:
Thank you.

I have a very basic doubt here. If my set up is as shown below:
[i210NIC]<----------switch-------->[i210NIC]
   PC1              AVB compliant       PC2

Can I run the openAVB stack on the AVB switch also or does an avb compliant switch come with the necessary AVB stack, which would also include the bandwidth checking capability as explained above?

Thanks
Swetha

jdkoftinoff, 2015-04-16:
The AVB compliant switch will support SRP and it will do the bandwidth allocation and testing.  

Please see: http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf

And http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf

Regards
Jeff