Avnu/OpenAvnu #201: Question: Talker and listener application supporting video

VinodTadvai, 2015-04-03:
Do you have talker and listener application supporting live video (video camera)?

intel-ethernet, 2015-04-03:
I know of groups which have - I thought it was a prototype gstreamer plug-in - but I’m don’t recall whether these ever made it back into the Open AVB repositories.  I also remember a hack to plug into VNC to live-stream packets as well.

Often for simplicity, the designs often use the Qav traffic shaper WITHOUT timed transmission of the individual video frames (packets) - e.g. they use the I210’s Queue0 (Class A) for audio and Queue1 (Class B) for video – and just dump the encapsulated video packets onto the Qav-only transmit queue as fast as possible.

Often the video encoder selection – e.g. MPEG vs M-JPEG vs raw bitmaps(!) – can make the bandwidth calculation ‘hard’ – its hard to predict how big any individual packet is going to be, and necessarily when the packet will be available to send out of the encoders … so often the bandwidth is just ‘allocated’ but not always fully consumed.

Hope this helps.
Ekm

VinodTadvai, 2015-04-03:
I see gstreamer-avb-plugin in the repository, Is that what you are referring above?

Regards,
Vinod

jdkoftinoff, 2015-04-03:
For reference, while “dumping packets into the Qav-only transmit queue as fast as possible” is passable for a demonstration, people need to understand that it would not be sufficient for a shipping product.  Specifically, each bridge down the line has it’s own Qav shaper as well which is running off of a different clock domain and the media would be transmitted as FAST as possible within the shaper’s rules instead of the appropriate nominal frequency.  The Qav shaper is not sufficient as a media clock source.  At minimum for video, the software needs to at least trigger the sending of each video frame at the appropriate time.

Regards,

Jeff

closed 2015-04-17 by andrew-elder