Avnu/OpenAvnu #205: [Open-AVB] simple_listener

swethachallawar, 2015-04-13:
I am trying to understand the simple_listener.c. 
What I observe is, the simple listener is sending a ready(send_ready API) on seeing a talker. However, wouldn't the listener need to check the availability of bandwidth before sending a ready?
If so, is there any API/call in open-avb code that checks for bandwidth requirements at the listener side.
Also,  I would like to understand which APIs are taking care of validating and setting Listener Ready, Listener Ready Failed and Listener Asking Failed.

Thanks
Swetha

andrew-elder, 2015-04-13:
simple listener is exactly that - a simple listener example application. It 
does not attempt to handle error conditions. That said, if you wanted to 
add a module that that better tested for error conditions I'm sure that 
would be a welcome addition to the Open-AVB codebase.

swethachallawar, 2015-04-13:
sure, our project might need to do error checks too. Thank you so much for the response !

Thanks
Swetha