Avnu/OpenAvnu #207: Making asCapable state available to client.

terrybaumbach, 2015-04-24:
I am working on an older code base which contains an object from the IPC class. It utilized this ipc object's setSharedAsCapable method to make the currect state of asCapable available to remote clients. I am now bringing this application up to date with the current open-avb source code and see that there is only the one setAsCapable call, which is a member of the IEEE1588Port class. Unfortunately my mechanism to transfer to my client is driven through the setMasterOffset() method of the IEEE1588Clock class and the update() method of the IPC class. As such I don't have access to the Port object, and more importantly, the asCapable private member therein.

   Before I get into deriving my own Port class to hotwire this state to the code that needs this information I was wondering if there is a mechanism I have missed that is designed to get the asCapable state to the client.

   If not would it be possible to add a state/status parameter to setMasterOffset() and update() to make this state available outside of the Port object?

christopher-s-hall, 2015-05-04:
I exposed "pdelay_count" in the shared memory segment.  The intended behavior is that this should be set to 0 when asCapable is set to false and then incremented for each valid pdelay response received.

from gPtpTimeData:

uint32_t sync_count;
**uint32_t pdelay_count;**
PortState port_state;

andrew-elder, 2015-05-04:
Right, but asCapable does not have a one to one relationship with pdelay responses. What say the neighborDelayThreshold is exceeded?

I'll aim to post a patch for you to take a look at over the next couple of days.

christopher-s-hall, 2015-05-04:
I think it still works because if any threshold is exceeded (some of this is still unimplemented in the daemon) asCapable becomes false and the pdelay count is set to zero.  In this case pdelay count should always be less than three.  I'm not against adding an additional interface component I'm just not convinced that it's necessary.

andrew-elder, 2015-05-05:
Hmmmm. Interesting. So pdelay_count is really pdelay_valid_count? We may be able to get away with only pdelay_valid_count, but we need some more documentation around it. For example, is it incremented every valid pdelay? That can't be correct because it doesn't handle the wrap case. It is pdelay_count up to a maximum of X?

terrybaumbach, 2015-05-05:
As I may have multiple places in my code that needs the AsCapable state, that means that there are multiple locations which need to have the rules  to deduce AsCapable's state. As the code evolves I think this is a bad idea since we would need to keep these rules updated with each change. If I had a vote I would vote for a method of retrieving the actual AsCapable so that only one piece of code needs the knowledge of these rules.

christopher-s-hall, 2015-05-05:
Yes.  It is valid Pdelay count.  I think maximum X would be fine.  We can also go 64 bit.  I don’t see it as a problem either way.

andrew-elder, 2015-05-11:
Pull request has been submitted. Rather simple one liner!

christopher-s-hall, 2015-05-11:
I don't have any objection, but does this solve the original problem?

andrew-elder, 2015-05-11:
It does for the Windows build we are using in combination with updated Windows ipcdef. I imagine it won't help linux folks without a couple of additions to linux specific code.

davidcemin, 2015-07-27:
Judging by the comments on this issue, I understand that it is working on windows, and it has to be ported to linux. So, would be the same case as integrating ipcdef between linux and windows ?

davidcemin, 2015-11-09:
On OS_IPC I created a new parameter with type bool called 'asCapable'. On IEEE1588Clock::setMasterOffset I added a new parameter type bool called asCapable which is set through IEEE1588Port::getAsCapable method.

I tested on Linux and it is working. On windows I added the asCapable parameter to the virtual method update on WindowsNamedPipeIPC::update() just not to break anything. I did not test it on a windows platform. 

Would that solve this problem ? (pushing in code to my fork soon)

closed 2016-03-16 by davidcemin