Avnu/OpenAvnu #209: Strange dmatime ?

zhoupeng1982, 2015-05-05:
Simple_test.c is used to test dmatime, and result seems strange:
log(attime dmatime)=    104013778       887216480
log(attime dmatime)=    114013778       887216280
log(attime dmatime)=    124013778       887216080
log(attime dmatime)=    134013778       887215880
log(attime dmatime)=    144013778       887215680
log(attime dmatime)=    154013778       887215480
log(attime dmatime)=    164013778       887215280
log(attime dmatime)=    174013778       887215080
log(attime dmatime)=    184013778       887214880
log(attime dmatime)=    194013778       887214680
The attime increase every 10ms as I want, the dmatime seems incorrect and only change a little digit.

Simple_test.c is linked to Open-AVB-master.
///////////////////////Simple_test.c :
#include <unistd.h>
#include <fcntl.h>
#include <stdlib.h>
#include <stdio.h>
#include <string.h>
#include <syslog.h>
#include <signal.h>
#include <errno.h>

#include <sys/ioctl.h>
#include <sys/time.h>
#include <sys/resource.h>
#include <sys/mman.h>
#include <sys/user.h>

#include <stdio.h>
#include <pci/pci.h>
#include "igb.h"

#define IGB_BIND_NAMESZ 24

device_t 	igb_dev;
volatile int	halt_tx;

void
sigint_handler (int signum)
{
	printf("got SIGINT\n");
	halt_tx = signum;
}

int
connect()
{
	struct	pci_access	*pacc;
	struct	pci_dev	*dev;
	int		err;
	char	devpath[IGB_BIND_NAMESZ];

	memset(&igb_dev, 0, sizeof(device_t));

	pacc	=	pci_alloc();
	pci_init(pacc);
	pci_scan_bus(pacc);
	for	(dev=pacc->devices;	dev;	dev=dev->next)
	{
		pci_fill_info(dev,	PCI_FILL_IDENT	|	PCI_FILL_BASES	|	PCI_FILL_CLASS);

		igb_dev.pci_vendor_id = dev->vendor_id;
		igb_dev.pci_device_id = dev->device_id;
		igb_dev.domain = dev->domain;
		igb_dev.bus = dev->bus;
		igb_dev.dev = dev->dev;
		igb_dev.func = dev->func;

		snprintf(devpath, IGB_BIND_NAMESZ, "%04x:%02x:%02x.%d", \
			dev->domain, dev->bus, dev->dev, dev->func );
		printf("probing %s\n", devpath);

		err = igb_probe(&igb_dev);
		if (err) {
			continue;
		}

		printf ("attaching to %s\n", devpath);
		err = igb_attach(devpath, &igb_dev);

		if (err || igb_attach_tx( &igb_dev )) {
			printf ("attach failed! (%s)\n", strerror(errno));
			continue;
		}
		printf("attach successful to %s\n", devpath);
		goto out;
	}

	pci_cleanup(pacc);
	return	ENXIO;

out:
	pci_cleanup(pacc);
	return	0;

}

#define PACKET_IPG	(10000000) /* 10 msec */

int
main()
{
	unsigned	i;
	int	err;
	struct igb_dma_alloc	a_page;
	struct igb_packet	a_packet;
	struct igb_packet	*tmp_packet;
	struct igb_packet	*cleaned_packets;
	struct igb_packet	*free_packets;
	u_int64_t	last_time;
	u_int64_t	rdtsc0;

	err = connect();

	if (err) { printf("connect failed (%s)\n", strerror(errno)); return(errno); }

	err = igb_init(&igb_dev);

	if (err) { printf("init failed (%s)\n", strerror(errno)); return(errno); }

	err = igb_dma_malloc_page(&igb_dev, &a_page);

	if (err) { printf("malloc failed (%s)\n", strerror(errno)); return(errno); }

#define PKT_SZ	200
	a_packet.dmatime = a_packet.attime = a_packet.flags = 0;
	a_packet.map.paddr = a_page.dma_paddr;
	a_packet.map.mmap_size = a_page.mmap_size;

	a_packet.offset = 0;
	a_packet.vaddr = a_page.dma_vaddr + a_packet.offset;
	a_packet.len = PKT_SZ;

	free_packets = NULL;

	/* divide the dma page into buffers for packets */
	for (i = 1; i < ((a_page.mmap_size) / PKT_SZ); i++) {
		tmp_packet = malloc(sizeof(struct igb_packet));
		if (NULL == tmp_packet) { printf("failed to allocate igb_packet memory!\n"); return(errno); }

		*tmp_packet = a_packet;
		tmp_packet->offset = (i * PKT_SZ);
		tmp_packet->vaddr += tmp_packet->offset;
		tmp_packet->next = free_packets;
		memset(tmp_packet->vaddr, 0xffffffff, PKT_SZ); /* MAC header at least */
		memset(tmp_packet->vaddr+6, 0x4, PKT_SZ-6);
		((char *)tmp_packet->vaddr)[12] = 0x88;
		((char *)tmp_packet->vaddr)[13] = 0x88;
		free_packets = tmp_packet;
	}

	igb_set_class_bandwidth(&igb_dev, 0, 0, 0, 0); /* disable Qav */

	halt_tx = 0;
	signal(SIGINT, sigint_handler);

	err = igb_get_wallclock(&igb_dev, &last_time, &rdtsc0);

	while (!halt_tx) {
		tmp_packet = free_packets;
		if (NULL == tmp_packet) goto cleanup;

		free_packets = tmp_packet->next;
		tmp_packet->dmatime = 0;
		tmp_packet->attime = last_time + PACKET_IPG;
		*(u_int64_t *)(tmp_packet->vaddr + 32) = tmp_packet->attime;

		err = igb_xmit(&igb_dev, 0, tmp_packet);

		if (!err) { last_time += PACKET_IPG; continue; }

		if (ENOSPC == err) {
			/* put back for now */
			tmp_packet->next = free_packets;
			free_packets = tmp_packet;
		}
cleanup:
		igb_clean(&igb_dev, &cleaned_packets);
		i = 0;
		while (cleaned_packets) {
			i++;
			tmp_packet = cleaned_packets;
			cleaned_packets = cleaned_packets->next;

			/* remap attime to compare to dma time */
			tmp_packet->attime -= (tmp_packet->attime / 1000000000)*1000000000;
			/* doesn't handle wrap-around ! */
			printf("log(attime dmatime)=\t%d\t%d\n",
				(unsigned int)(tmp_packet->attime), (unsigned int)(tmp_packet->dmatime));

			tmp_packet->next = free_packets;
			free_packets = tmp_packet;
		}
	}

	igb_dma_free_page(&igb_dev, &a_page);
	err = igb_detach(&igb_dev);
	return(0);
}

////////////////////Makefile :
OPT=-O2
CFLAGS=$(OPT) -Wall -W -Wno-parentheses

CC=gcc
INCFLAGS=-I../../lib/igb
LDLIBS=-ligb -lpci -lrt -lm -pthread
LDFLAGS=-L../../lib/igb

all: simple_test

simple_test: simple_test.o

simple_test.o: simple_test.c
	gcc -c $(INCFLAGS) -I../../daemons/mrpd $(CFLAGS) simple_test.c

%: %.o
	$(CC) $(LDFLAGS) $^ $(LDLIBS) -o $@

clean:
	rm -f `find . -name "*~" -o -name "*.[oa]" -o -name "\#*\#" -o -name TAGS -o -name core -o -name "*.orig"` simple_test

zhoupeng1982, 2015-05-05:
This is a bug: if dmatime is wanted, 1588_STAT_EN must be set in TQAVCTRL, 
which is missing in kmod and lib.
When I add this flag 1588_STAT_EN, every thing is right:
log(attime dmatime)=    403080885       403073365
log(attime dmatime)=    413080885       413073221
log(attime dmatime)=    423080885       423073317
log(attime dmatime)=    433080885       433073317
log(attime dmatime)=    443080885       443073317
log(attime dmatime)=    453080885       453073237
log(attime dmatime)=    463080885       463073365
log(attime dmatime)=    473080885       473073189
log(attime dmatime)=    483080885       483073301
log(attime dmatime)=    493080885       493073189

zhoupeng1982, 2015-05-05:
This is a bug: if dmatime is wanted, 1588_STAT_EN must be set in TQAVCTRL, 
which is missing both in kmod and lib.
When I add this flag 1588_STAT_EN, every thing is right:
log(attime dmatime)= 403080885 403073365
log(attime dmatime)= 413080885 413073221
log(attime dmatime)= 423080885 423073317
log(attime dmatime)= 433080885 433073317
log(attime dmatime)= 443080885 443073317
log(attime dmatime)= 453080885 453073237
log(attime dmatime)= 463080885 463073365
log(attime dmatime)= 473080885 473073189
log(attime dmatime)= 483080885 483073301
log(attime dmatime)= 493080885 493073189

davidcemin, 2015-05-05:
Im not familiar with this code, but im marking it as a bug as per the comment above.

intel-ethernet, 2015-05-06:
Somehow this thread (and several other Open-AVB posts) got moved to my ‘junk’ box …

This history of the I210’s “DMA time” is complicated. As it was originally designed, the timestamp written back would be the 1588 PTP timestamp time. However, as implemented, the value instead is the timestamp when the packet descriptor is written back to host memory. If the tx queue writeback thresholds are set to batch tx descriptor writebacks (a performance/PCIe efficiency feature) then the timestamp reported is very far removed from the time the packet was DMAed from memory, and also potentially has little correlation to the time the packet is on the physical wire.

We continue to use the feature internally to measure low-level processor/chipset/PCIe memory access latencies by disabling transmit descriptor writeback batching, setting the pre-fetch threshold to -0-, sending a stream of timed transmit packets, looking at the attime vs dmatime (subtracting off 1.8 usec of tx descriptor writeback turnaround latency), and reporting memory access times.

“Simple talker” started out as this simple test routine. It obviously has been enhanced to make it a basic AVB protocol demonstration application. ‘dmatime’ is a vestige of the internal tool still used.

Caveat emptor.

ekm

pinealservo, 2015-09-04:
After @intel-ethernet's comment, I'm not sure this is even a bug, or what @zhoupeng1982's change was doing. I don't think we have any need for this dmatime measurement in simple-talker, but I'm going to leave it alone unless someone who knows more about the i210 wants to re-open this.

closed 2015-09-04 by pinealservo