Avnu/OpenAvnu #210: Issue with "Detected Tx Unit Hang" dropping network connections

chris-kuhr, 2015-05-06:
Hi *,

since yesterday my AVB setup suddenly has issues!
I have two NICs with subnet A and subnet B, I switched the VLANs between both switch ports. Which worked fine.
I am not in front of this machine, thus I will post Kernel log later. 
Kernel Version is 3.13.0-24-generic.

On boot up, the kernel tries 60 seconds to reset the network adapter.
And fails.
I have to rmmod and insmod igb_avb.ko again.

I found a solution on the web:
https://communities.intel.com/message/212871#212871

But why should I deactivate LRO, if it worked previously?

Perhapes someone knows this issue?

intel-ethernet, 2015-05-06:
Tx Hangs are indicative that the internal pointers used within the MAC have become corrupted (point into the “weeds”). This usually (on AVB) means that someone tried to transmit and some other thread re-configured the packet buffer memories (as done with AVB), but there are other ways as well - like transmitting 0-length with some offload (checksum?) enabled or oversized (13KB+) Ethernet frames.

I have no idea why “large receive offload” (LRO) would impact AVB on the 1G products – as LRO is a software feature. The 10G products (e.g. ixgbe) have HW-based LRO.

LSO on the other hand would impact AVB - as the HW arbiter treats the 64KB block as “one big packet” and doesn’t switch over to the AVB queues until it is done.

ekm

https://communities.intel.com/message/212871#212871

chris-kuhr, 2015-05-06:
I Have now seen, this happens to the igb, not the igb_avb module.

So perhapes the transmit queues are "locked" to the AVB queues from the last uptime?
How can I redirect them as they have been before?

chris-kuhr, 2015-05-06:
How can I tell the kernel to load the igb_avb module instead of the igb module?
Igb is blacklisted and igb_avb is registered with `depmod`. But the igb module is loaded at the first try.

Sorry, it's a bit long... `/var/log/kern.log`:

```
May  6 19:47:17 workstation kernel: [    1.866583] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.0.5-k
May  6 19:47:17 workstation kernel: [    1.866585] igb: Copyright (c) 2007-2013 Intel Corporation.
May  6 19:47:17 workstation kernel: [    1.867053] igb 0000:09:00.0: irq 99 for MSI/MSI-X
May  6 19:47:17 workstation kernel: [    1.867060] igb 0000:09:00.0: irq 100 for MSI/MSI-X
May  6 19:47:17 workstation kernel: [    1.867067] igb 0000:09:00.0: irq 101 for MSI/MSI-X
May  6 19:47:17 workstation kernel: [    1.867074] igb 0000:09:00.0: irq 102 for MSI/MSI-X
May  6 19:47:17 workstation kernel: [    1.867080] igb 0000:09:00.0: irq 103 for MSI/MSI-X
May  6 19:47:17 workstation kernel: [    1.917683] igb 0000:09:00.0: added PHC on eth0
May  6 19:47:17 workstation kernel: [    1.917686] igb 0000:09:00.0: Intel(R) Gigabit Ethernet Network Connection
May  6 19:47:17 workstation kernel: [    1.917688] igb 0000:09:00.0: eth0: (PCIe:2.5Gb/s:Width x1) a0:36:9f:3f:be:a3
May  6 19:47:17 workstation kernel: [    1.917875] igb 0000:09:00.0: eth0: PBA No: G69016-001
May  6 19:47:17 workstation kernel: [    1.917876] igb 0000:09:00.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
...

...
May  6 19:47:19 workstation kernel: [    8.202666] igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
May  6 19:47:19 workstation kernel: [    8.203016] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
May  6 19:47:21 workstation kernel: [    9.410215] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
May  6 19:47:21 workstation kernel: [    9.410389] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
May  6 19:47:21 workstation kernel: [   10.125989] igb 0000:09:00.0: Detected Tx Unit Hang
May  6 19:47:21 workstation kernel: [   10.125989]   Tx Queue             <0>
May  6 19:47:21 workstation kernel: [   10.125989]   TDH                  <0>
May  6 19:47:21 workstation kernel: [   10.125989]   TDT                  <1>
May  6 19:47:21 workstation kernel: [   10.125989]   next_to_use          <1>
May  6 19:47:21 workstation kernel: [   10.125989]   next_to_clean        <0>
May  6 19:47:21 workstation kernel: [   10.125989] buffer_info[next_to_clean]
May  6 19:47:21 workstation kernel: [   10.125989]   time_stamp           <fffee2ae>
May  6 19:47:21 workstation kernel: [   10.125989]   next_to_watch        <ffff8804080fd000>
May  6 19:47:21 workstation kernel: [   10.125989]   jiffies              <fffee48d>
May  6 19:47:21 workstation kernel: [   10.125989]   desc.status          <168000>
May  6 19:47:23 workstation kernel: [   12.124022] igb 0000:09:00.0: Detected Tx Unit Hang
May  6 19:47:23 workstation kernel: [   12.124022]   Tx Queue             <0>
May  6 19:47:23 workstation kernel: [   12.124022]   TDH                  <0>
May  6 19:47:23 workstation kernel: [   12.124022]   TDT                  <1>
May  6 19:47:23 workstation kernel: [   12.124022]   next_to_use          <1>
May  6 19:47:23 workstation kernel: [   12.124022]   next_to_clean        <0>
May  6 19:47:23 workstation kernel: [   12.124022] buffer_info[next_to_clean]
May  6 19:47:23 workstation kernel: [   12.124022]   time_stamp           <fffee2ae>
May  6 19:47:23 workstation kernel: [   12.124022]   next_to_watch        <ffff8804080fd000>
May  6 19:47:23 workstation kernel: [   12.124022]   jiffies              <fffee681>
May  6 19:47:23 workstation kernel: [   12.124022]   desc.status          <168000>
May  6 19:47:23 workstation kernel: [   12.124026] igb 0000:09:00.0: Detected Tx Unit Hang
May  6 19:47:23 workstation kernel: [   12.124026]   Tx Queue             <3>
May  6 19:47:23 workstation kernel: [   12.124026]   TDH                  <0>
May  6 19:47:23 workstation kernel: [   12.124026]   TDT                  <1>
May  6 19:47:23 workstation kernel: [   12.124026]   next_to_use          <1>
May  6 19:47:23 workstation kernel: [   12.124026]   next_to_clean        <0>
May  6 19:47:23 workstation kernel: [   12.124026] buffer_info[next_to_clean]
May  6 19:47:23 workstation kernel: [   12.124026]   time_stamp           <fffee42a>
May  6 19:47:23 workstation kernel: [   12.124026]   next_to_watch        <ffff8804081e5000>
May  6 19:47:23 workstation kernel: [   12.124026]   jiffies              <fffee681>
May  6 19:47:23 workstation kernel: [   12.124026]   desc.status          <628000>
May  6 19:47:23 workstation kernel: [   12.124029] igb 0000:09:00.0: Detected Tx Unit Hang
May  6 19:47:23 workstation kernel: [   12.124029]   Tx Queue             <1>
May  6 19:47:23 workstation kernel: [   12.124029]   TDH                  <0>
May  6 19:47:23 workstation kernel: [   12.124029]   TDT                  <1>
May  6 19:47:23 workstation kernel: [   12.124029]   next_to_use          <1>
May  6 19:47:23 workstation kernel: [   12.124029]   next_to_clean        <0>
May  6 19:47:23 workstation kernel: [   12.124029] buffer_info[next_to_clean]
May  6 19:47:23 workstation kernel: [   12.124029]   time_stamp           <fffee3b3>
May  6 19:47:23 workstation kernel: [   12.124029]   next_to_watch        <ffff880402c90000>
May  6 19:47:23 workstation kernel: [   12.124029]   jiffies              <fffee681>
[... cut ...]
May  6 19:49:31 workstation kernel: [  140.030350] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
May  6 19:49:31 workstation kernel: [  140.231730] userif-4: sent link down event.
May  6 19:49:31 workstation kernel: [  140.231734] userif-4: sent link up event.
May  6 19:49:56 workstation kernel: [  165.116975] igb_avb 0000:09:00.0: removed PHC on eth0
May  6 19:49:57 workstation kernel: [  165.365060] userif-4: sent link down event.
May  6 19:49:57 workstation kernel: [  165.365063] userif-4: sent link up event.
```

intel-ethernet, 2015-05-07:
Low-tech way is to edit ‘/etc/modprobe.d/blacklist.conf’ and add ‘blacklist igb’ to prevent the in-kernel module from loading. Then drop igb_avb.ko in /lib/modules/’uname –r’/updates, and run depmod –a’

Some folks however have multiple Intel LAN cards and want the others to work as well … which if I recall, you can skip the blacklist bit … but it it doesn’t work, let me know and I’ll add the full details to the readme.

ekm

pinealservo, 2015-08-11:
Closing due to inactivity.

closed 2015-08-11 by pinealservo