Avnu/OpenAvnu #211: AVB switch to be used with OpenAVB protocol stack

swethachallawar, 2015-05-06:
Hi,

We are planning to run a AVB set up with two endpoints and an switch in between using the OpenAVB protocol stack. We would highly appreciate any suggestions on which switch could be used.

I have read through the forums  that some are using the Marvell based dsp4u AVB switch. I am assuming that the same OpenAVB protocol stack can be used with dsp4u aslo. Please let me know if that's correct?

Thanks
Swetha

swethachallawar, 2015-05-07:
Any inputs on the above question would be of great help.

Thanks
Swetha

jdkoftinoff, 2015-05-07:
I have no information about DSP4U

I would recommend going with an AVnu certified switch with the appropriate options.

One switch is the 8 port switch from extreme networks: http://www.extremenetworks.com/product/summit-x440-series <http://www.extremenetworks.com/product/summit-x440-series>

Remember that with Extreme Networks switches, each option is performed with a license purchased and enabled. So you would have to purchase the AVB license.

Another option that some people have had good success with is the MOTU AVB Switch, however it is not AVnu compliant:

http://www.motu.com/products/avb/avb-switch

Regards,
Jeff

swethachallawar, 2015-05-07:
Ok. Thank you for your time and inputs.

I have another doubt. Can we use the OpenAVB protocol stack on this AVB switch. 
We plan on running the OpenAVB protocol stack on the endpoints.
So here I am wondering how would the switch understand the OpenAVB mrp messages(say S++ or S+L), if it has a different AVB stack than what the endpoint has.

Please help me understand this aspect.

Thanks
Swetha

jdkoftinoff, 2015-05-07:
The AVB switch will use its own stack which is compliant with the standard SRP and MVRP messages.

What are you trying to do exactly?

--jeffk++

andrew-elder, 2015-05-07:
S++ etc. are routed to the mrp daemon running on the end station. The mrp daemon declares attributes by sending ethernet packets to the switch. There is no direct communication with the switch using S++ type commands. The end station OpenAVB mrp daemon communicates with the switch according to IEEE802.1Q, so they can understand each other.
  
  

----------------------------------------

swethachallawar, 2015-05-11:
Thank you so much. That clarifies my doubt.

The simple listener example is expecting a single new stream from the talker. However, if my listener wants to receive multiple streams from different talkers, would the openAVB mrp support multiple streams or does the mrp support single stream as of now?

Thanks
Swetha

swethachallawar, 2015-05-12:
Can please some one comment on OpenAVB MRP support for multiple streams on a single listener?

I understand that the simple listener application as it is can handle only one stream as of now and needs modifications for multiple stream support, however does the mrp as it is support multiple streams?

andrew-elder, 2015-05-12:
Yes, OpenAVB MRP does correctly support multiple input streams to a 
listener capable end station.

swethachallawar, 2015-05-13:
Ok. Thank you. 

We are planning to develop a listener application which should be able to play four active streams at the same time. Any pointers please on what are the aspects that I should be taking care of with respect to mrp ?

Thanks
Swetha

davidcemin, 2015-05-13:
Hi, 

Your listener application should be able to subscribe then to four streams. The simple listener provides you an idea of how you can do it using one stream basically. 

Please take a look at this document, at section 5.2.5:
http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf

This section will provide you timing diagrams of how things should be configured from both talker and listener point of view.

Another very good document that summarizes the mrp behavior is the following one. Id advise you also to take a look at it. It will give you an idea of what is implemented on Open-AVB mrpd application.
http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf

closed 2015-08-11 by pinealservo