Avnu/OpenAvnu #213: Simple listener

swethachallawar, 2015-05-12:
I have read that "technically we don't need to run the listener on I210 HW". Can some one please explain why? 
is it because only talker needs to support traffic shaping and hence tightly tied to HW. More information on this would be of great help

Thanks
Swetha

jdkoftinoff, 2015-05-12:
Hello Swetha

Traffic shaping is only used on transmit. If the device is a listener only, then the device’s ethernet port doesn’t need to be able to do traffic shaping.

However, there is another important point - Whatever ethernet port is used must be capable of doing 802.1AS. This means it needs proper time stamp support on both the ingress time and egress time of specific 802.1AS messages. If it doesn’t do 802.1AS properly, then the ethernet switch that is connected to it will reject your device and will refuse to negotiate bandwidth with it.

You may find some of the questions in my AVB Developer FAQ relevant:

	https://avb.statusbar.com/page/developer-faq

Specifically:

1.7   What is the difference between PTP, PTPv2, gPTP, 802.1AS and IEEE 1588? <https://avb.statusbar.com/page/developer-faq/#id8>

1.8   Can an AVB device just use IEEE 1588v2 (PTP) instead of 802.1AS? <https://avb.statusbar.com/page/developer-faq/#id9>

1.9   My device's ethernet port says it supports IEEE 1588. Can it do AVB? <https://avb.statusbar.com/page/developer-faq/#id10>

1.10   Can FQTSS be implemented in pure software? <https://avb.statusbar.com/page/developer-faq/#id11>

Best Regards,
Jeff Koftinoff

swethachallawar, 2015-05-13:
Thank you so much for the clarification.

I have another doubt related to bandwidth reservation on endpoints. 
Do the endpoints need to evaluate the bandwidth availability too or is it only the bridges that perform bandwidth availability checks?
In simple talker, I can see that igb_set_class_bandwidth() is being called? however, is that sufficient for bandwidth checking? 
Also. the simple listener does not perform any bandwidth checks and directly sends out a ready message. Ideally, should the endpoints be doing bandwidth checks or not?

Thanks
Swetha

jdkoftinoff, 2015-05-13:
Hi Swetha

In general, no.

If the network link is not AVB capable, then the device should not be sending any stream reservation packets at all.

If the network link is 100baseT and the stream doesn’t fit, the switch will send Talker Failed attributes for the affected streams.

Jeff

swethachallawar, 2015-05-15:
Thanks a lot Jeff!

Curious about the value of PKT_SZ (100) in simple talker code. It is being passed as a parameter to igb_set_class_bandwidth as well as mrp_advertise_stream. How is the vale of PKT_SZ arrived at? Is it dependent on the type of media ? Here in simple talker and listener example it raw uncompressed data, but if my media format is any different (say mpeg,mjpeg or mp3), should the PKT_SZ be calculated differently? if so, please let me know how. 
Also, in general, does the format of media stream needs to be taken care of while configuring QAV shaper, ie in the igb_set_class_bandwidth call. I would like to understand the details please.

Thanks
Swetha

swethachallawar, 2015-05-15:
If I want to send across a video stream(mpeg or mjpeg), how would I arrive at PKT_SZ?
Also, would the per-frame-overhead remain the same as in OpenAVB code which is Ethernet IPG+Preamble+Start of Frame+Mac Header with VLAN+Etype+CRC
Or would I need to take care of anything more? Please let me know. 

Thanks
Swetha

swethachallawar, 2015-05-18:
Any suggestions on the query I have asked above would be of great help.

Thanks
Swetha

jdkoftinoff, 2015-05-18:
PKT_SZ is maximum size of the mpeg video packet that you expect to send within the 125 microsecond observation interval.

Jeff

closed 2015-08-11 by pinealservo