Avnu/OpenAvnu #215: Need to implement multiple stream audio and video

keyurparekh1, 2015-05-14:
Hi,

did any body try to send multiple stream from single talker? I need help to implement multiple audio stream to send from one talker to multiple listener.

intel-ethernet, 2015-05-14:
Maybe a dumb question – but do you mean sending one stream from a single talker to multiple listeners, or sending multiple streams from a single endpoint to multiple listeners?

If it is the latter, it’s a question of multiple audio feeds into multiple streams and muxing onto the same transmit queue (or using both SR queues if necessary/if it is easier). They are ultimately just packets to the LAN interface. How software decides what packet to transmit when is a higher level problem.

ekm

keyurparekh1, 2015-05-14:
Hi,

I mean to send multiple streams from one talker. And I have multiple listener. So I want all listener should play different streams.

pcub, 2015-08-05:
Hello , I also want to send more than one stream (for example two stream) from two talker to one listener.

As I knew simple_listener now only supports one stream.

How to implement it in simple_listener ?

jdkoftinoff, 2015-08-05:
Please clarify.

Do you mean two talkers with two stream sources each going to one listener receiving four streams?

Jeff

xcalger, 2015-08-05:
I just read through this thread. Thank you Jeff for answering this question. I agree with you in your settings for producing what I *think* pcub is trying to accomplish.

One stream of 4 channels from a talker on each XMOS DC going to a PC running simple_listener (I'm not familiar with that utility, so not sure what it is capable of).

-Chris
XMOS

pcub, 2015-08-06:
xcalger's right. yesterday I posted in https://github.com/AVnu/Open-AVB/issues/244.
Is there anyone who is familiar with simple_listener hacking?

pcub, 2015-08-10:
@xcalger 
Can you give some information about xmos side configuration.
I posted here: https://github.com/xcore/sc_ethernet/issues/60

I want to know whether there is problem or not in xmos daisy chain board configuration.

Here is the xmos guy demonstrating daisy chain: https://www.youtube.com/watch?v=DrG4oDxl78Q
As I saw from this demo two DC and audio slice's output  audio channels are connected to speakers and acting as listener. Is it right?
If it is right, then there is no demo that shows all stream are transmitted via ethernet slice kit. is not it?

Thank you.

pinealservo, 2015-08-11:
@pcub This isn't where you go to get answers about XMOS daisy chain board configuration. Please don't hijack other people's issues with unrelated questions.

@keyurparekh1 Having a talker send multiple streams is essentially the same as having it send one stream, except you do it once per unique Stream ID you wish to send.  The only thing special about talker AVB streams as opposed to, for example, http streams, is that they have to be sent at a specific rate and you can't oversubscribe your bandwidth.

Our issue system here isn't meant as a long-term training support dialog system; we need specific questions or problems with the software pointed out that we can give specific answers to. If you make some attempt at writing code to send multiple streams from a talker, or if you have a specific question about how an API should be used, please open a new issue regarding that.

closed 2015-08-11 by pinealservo

keyurparekh1, 2015-08-11:
Hi Pinealservo,

I am currently modifying talker code to send multiple streams. Once I have stable implementation I will paste code for further clarification in new issue.

Thanks & Regards,
Keyur Parekh

xcalger, 2015-08-12:
@pcub: Levi made an excellent description of how AVB works in issue #264 . I hope it is clear on the responsibility of each device and part of the firmware stack. Please make sure you have this in mind on what you are trying to do first before asking about configuring each device a certain way. This means what streams and channels are sent where ... this also includes understandingthe media clocking for each stream.

As for XMOS support, I see your request in sc_ethernet. You should enter AVB and specifically firmware questions about the DC in http://github.xcore.com/sw_avb.

If you think there is a bug in firmware you can also open a ticket on xmos.com by logging in and reporting a bug.

For how to set up the DC cards to do what you want using the sw_avb, I would also try the xcore.com Q&A section.

From what I think you are trying to do the XMOS DC kits should be able to each send a stream of 4 channels. It's the pc that has to listen to two different streams and then do something with the audio data in those streams that has the tough task of what to do with two separate audio streams.

-Chris