Avnu/OpenAvnu #216: Talker frame size

swethachallawar, 2015-05-18:
Hi,

I am trying to understand how to arrive at the value of packet size (which is to be passed as argument to mrp_advertise_stream as well as igb_set_class_bandwidth)

in simple talker and kackd talker PKT_SIZE is 100 whereas in live stream, we are getting this value from user (command line) and then adding six1883_header, seventeen22_header, sizeof(eth_header) to it.
packet_size += sizeof(six1883_header) + sizeof(seventeen22_header) + sizeof(eth_header);

May I know why is this being done, Also I would like to understand why adding the headers size is not need in case of simple talker and jackd talker.

Thanks
Swetha

swethachallawar, 2015-05-18:
Any comments on the above query would be of great help

Thanks
Swetha

intel-ethernet, 2015-05-18:
The frame size is media dependent. On an 802.3 (wired) connection it typically is:

“Maximum Frame Size: The largest possible Talker AVB frame transmitted as part of the stream. Together with Frames Per Interval, this specifies the stream bandwidth requirements. The application should allocate the maximum possible data payload of an individual packet (the PDU). This excludes media-specific Layer-2 information, such as, using LAN as an example, the 6 byte source address, 6 byte destination address, 4 byte VLAN/User Priority tag and 4 byte CRC applied to an Ethernet frame.”

Note that more can be included – you can put any number of headers on an L2 Ethernet frame – like MACsec, double-VLAN, GRE, etc. I believe WiFi networks have different packet size calculations as well.

This variability – particularly the potential for different frame sizes on a per-hop basis – makes the switch implementations of SRP complex. At least I hope so - I believe bridges adapt the reservations such that an 802.3 talker doesn’t need to know the PDU framing requirements of an 802.11 listener if they are different.

You can read more in http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf, Section 5.2.1.2.

ekm

swethachallawar, 2015-05-19:
Thanks for the response.

So the max frame size is to be determined by the application based on the type of media. 

We are trying to send across a MJPEG video. I am trying to understand the modifications to be done in igb_set_class_bandwidth based on media type. 
Once the packet_size is arrived at, the bandwidth calculations, in igb_set_class_bandwidth don't remain the same. be it any type of audio/video, we should be adding the per frame media overhead to the max payload for calculating the actual bandwidth. Please confirm if am right?

Thanks
Swetha

swethachallawar, 2015-05-19:
All am trying to confirm is that igb_set_class_bandwidth() function is independent of media type. Please confirm my understanding.

Thanks
Swetha

swethachallawar, 2015-05-20:
Can someone please comment on the above query?
Thanks in advance

Thanks
Swetha

jdkoftinoff, 2015-05-20:
By media type do you mean copper vs fibre?

Or audio vs video?

--jeffk++

swethachallawar, 2015-05-20:
Hi Jeff, 

I mean audio vs video. To be specific, I wanted to understand handling of different formats like raw uncompressed audio and video, compressed audio and compressed video formats like mjpeg

thanks
Swetha

jdkoftinoff, 2015-05-20:
Hello Swetha

Do you have a copy of IEEE 1722-2011?

--jeffk++

swethachallawar, 2015-05-20:
Hi Jeff,

We have a draft version p1722a_D8.

Basically, I understand that MaxFrameSize is to be determined by the application developer based on the type of media to be streamed(audio/video) and other parameters like bit rate and frames per second. 
Since MaxFrameSize and MaxIntervalFrames are passed as parameters to igb_set_class_bandwidth() fucntion, this function would not to handle the type of media (audio/video) and hence I thought the  igb_set_class_bandwidth() function is indepent of type of media.

Also I had a query regarding the max interval frames:
for classA (125 us interval), the MaxIntervalFrames is 1 and hence 8000 frames per sec.
if I want to stream 400 frames per second, meaning .05 frames in 125 us, can I goahead and define max interval frames as .05?

Wondering if MaxIntervalFrames can be a fractional value?

Thank you so much for your responses.
Swetha

jdkoftinoff, 2015-05-20:
Hi Swetha

igb_set_class_bandwidth() is absolutely independent of the type of media inside an ethernet frame.

With AVB you must allocate the bandwidth for the largest ethernet frame and the highest count of Ethernet frames that may be sent within one class interval. If you are using Class A, that interval is 125 microseconds.

These parameters are based directly on the media encoding and packetization format as defined in IEEE 1722 and other relevant standards (iec61883, MPEG, JPEG, SDI, etc).

No, you can not define max interval frames as 0.05.  The minimum value is 1. Bandwidth is allocated as a percentage of the class interval only.  This is required in order for the latencies of all streams to be deterministic.

Regards, 
Jeff Koftinoff

swethachallawar, 2015-05-20:
Hi Jeff, 

That clarifies a lot. Thank you.

However, is there an upper limit on max interval frames? Is there a limitation from I210 HW perspective on max interval frames?

Thanks
Swetha

jdkoftinoff, 2015-05-20:
Hi Swetha

The upper limit on max interval frames is dependent on the type of ethernet port, the bandwidth allocated for AVB, and the capabilities of the ethernet chip.

I do not know the i201 HW well enough to tell you more information.

Remember that there is a class traffic shaper and a stream shaper, and you have to make sure the traffic is shaped properly. i.e. multiple ethernet frames for one stream or class are not allowed to be sent back-to-back.

Regards,

Jeff

intel-ethernet, 2015-05-20:
If I understand the question, you can send as fast or as slowly as you want with the I210 up to 1Gbps @ any frame size. The timed transmission and shaper are all in RTL in the primary datapath, not moderated by internal firmware, if that was the concern.

ekm

swethachallawar, 2015-05-20:
Thank you.

As an example,I was trying to check if I can send 8 packets, each of 1200 bytes in one interval or 125us. i.e if there is any limitation on I210 hw, with respect to no of frames being sent in each interval (per 125 us).

So i understand from the above response there is no such limitation on no of frames per interval as long as we do not exceed 1Gbps.

Please correct if my understanding is incorrect.

Thanks
Swetha

swethachallawar, 2015-05-21:
Can someone please confirm the above understanding. 

Also it would be great if you can point to any datasheets/specs that talk about the criteria for transmitting the streams.

Thanks
Swetha

swethachallawar, 2015-05-25:
Can someone please confirm the below understanding?
"As an example,I was trying to check if I can send 8 packets, each of 1200 bytes in one interval or 125us. i.e if there is any limitation on I210 hw, with respect to no of frames being sent in each interval (per 125 us).

So i understand from the above response there is no such limitation on no of frames per interval as long as we do not exceed 1Gbps."

Thanks
Swetha

intel-ethernet, 2015-05-25:
Yes it should work.  I will point out at 614Mbps you are getting close to the default 75% bandwidth limit of AVB.

swethachallawar, 2015-05-26:
Thank you so much for the response.

As I understand that minimum value of MAxIntervalFrames is 1. But I would like to understand potential impact if we have frames per interval set to 1 but actual rate is less than 1. 
that is if we are sending one frame in say 30 milli sec but the frames per interval is still set to 1. Please let me know if this can be achieved?

Thanks
Swetha

intel-ethernet, 2015-05-26:
You can send as few (or no) frames as you want … you just have reserved more AVB bandwidth with SRP through the network than you will actually use. So long as you don’t have 1000’s of such streams to artificially exhaust the bandwidth, it should “just work”.

Hths.
ekm

swethachallawar, 2015-05-27:
Thanks Again !

closed 2015-08-11 by pinealservo