Avnu/OpenAvnu #218: jackd-talker issue with wallclock time (Several Problems?)

chris-kuhr, 2015-05-25:
Hi *,

I am experiencing an issue with jackd-talker.
gptpd, maapd and mrpd are running as expected... well as I expect...
The ethernet port is asCapable and stream reservation for jackd-talker also works, as my switch shows the reserved stream.

Now if I connect a hardcoded XMOS listener to the stream, jackd-talker fails with the following message:

_______Error_______
```
   EVENT on STREAM ID=00187d8cc2bc0000 with state ready
   got a new/join indication
   got a listener ...
   ml_phoffset = -28395260977624, ls_phoffset = -5323597465048600730
   ml_freqffset = 1.000005, ls_freqoffset = 1.000000
   Failed to get wallclock time
```

_______PTP Group_______
I found the thread about the ptp group, which I checked. It exists and the users are members.

_______CPU Arch_______
I read the thread about measuring time with pcie cycles and perhapes my CPU arch plays a role?
4 Cores of:
```
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 21
model		: 16
model name	: AMD R-464L APU with Radeon(tm) HD Graphics
stepping	: 1
microcode	: 0x600110f
cpu MHz		: 2295.681
cache size	: 2048 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 16
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb arat cpb hw_pstate npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold bmi1
bogomips	: 4591.36
TLB size	: 1536 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro
```

_______Tx Hangs_______
Also worth mentioning, at boot time I am still having issues with TxHangs of the I210... Which disappears after about 5min uptime.

_______User RAW Socket_______
I also did some magic with:

```
    sudo setcap cap_net_raw+ep jackd_talker
    sudo chmod -R 777 /dev/igb_avb
```
Thus a normal user can use the raw socket and connect to jackd.

_______Modification to jackd-talker_______
I use :
```
    #define SRC_CHANNELS (8) //2
    #define SAMPLES_PER_SECOND (48000)
    #define FREQUENCY (480)
    #define SAMPLES_PER_CYCLE (SAMPLES_PER_SECOND/FREQUENCY)
    #define GAIN (.9) //.5

```
Because I got an error at compile time I modified the following line of code  (l.773) in `jackd_talker.c`:

```
   err = igb_attach_tx(/*devpath, */&igb_dev);
```

Any Ideas why the wallclock time fails?
Hardware issues?
Most likley, a permission issue? 
Any hint is appreciated!

intel-ethernet, 2015-05-25:
I suspect you are getting hit by a known issue (which I need to apply a patch to fix). Occasionally the clocks in the chip do not startup correctly, leading to all sorts of side-effects. (See issue#25 in the latest spec update - https://www-ssl.intel.com/content/www/us/en/embedded/products/networking/i210-ethernet-controller-spec-update.html).

There is a firmware update, but also a SW workaround which will be included in a future igb_avb update.

ekm

andrew-elder, 2015-05-26:
I don't see issue#25 in Table 1-5 (pg 10) of the linked document. Am I 
missing something?

- Andrew

intel-ethernet, 2015-05-26:
It is on page 30 … I have no idea what algorithm was used to sort or select the items in Table 1-5.

ekm

chris-kuhr, 2015-05-28:
Hi,

I had a look at it and decided to wait for the patch :-)
Do you, when you are going to to this?

BR,
Ck

pinealservo, 2015-08-11:
The software workaround is probably in the driver now, or perhaps there's been a firmware update? @intel-ethernet ?

closed 2018-10-02 by chris-kuhr