Avnu/OpenAvnu #219: Connection issue of avdecc

pcub, 2015-05-26:
Hello Andrew-elder. 
I installed avdecc-c under ubuntu. I have two xmos daisy chain board connected with non-avb compatible switch and one PC. I run the command ./avdecccmdline to enumerate the devices but I cannot get the eth1(connected with i210), only get the eth0 and lo.
avdecccmdline works If I first run ./avdecccmdline and plug the ethernet cable (eth1:which is from switch connected with xmos avb daisy chain) into jack of eth0.
What is the problem?

I also compiled openavb-next under ubuntu 12.04. There is a information about connecting two PC through avb. I used daemon_cl, mrpd, and simple_listener for connecting daisy chain and PC. But it seems does not work. I need a audio sample through avb. Daisy chain can get the audio input from autoside. Please give me a information about how to do it.
Thank you.

andrew-elder, 2015-05-26:
I'm sorry, I haven't tried your combination of devices, operating systems and AVB modules.
  
 - Andrew
  
  
  

----------------------------------------

pcub, 2015-06-18:
Hello Andrew,
Can you give me some information. Are you using two computer and avb switch inbetween two computer?
AS mentioned above issue, my computer should be the listener and xmos should be the master.
So, computer provides time information right?
But When I use ./daemon_cl -i eth1 -L
it immediately switches to master. ("Switching to Master").
What criteria must have in order to make computer as a listener?

andrew-elder, 2015-06-18:
I use mrpd on an ARM under linux in one configuration and then gPTP and mrpw under windows in another configuration. I don't use simple_talker or simple_listener at all. I'm not sure that I can help you.

pcub, 2015-06-18:
Thanks for your response, I want to contact someone who can help me

davidcemin, 2015-06-18:
I am not sure about the current status of simple_talker and simple_listener but hopefully i will have some time to give it a try soon.
Regarding the gPTP, in the file gptp/linux/src/linux_hal_generic.cpp there are defines for TX / RX phy delay. Make sure you have the right ones for i210, i.e. the following values should be defined:  
#define TX_PHY_TIME 184
#define RX_PHY_TIME 382

They are not the default values, so you might have to change it and recompile gPTP.

pcub, 2015-06-18:
I defined as above:
#define TX_PHY_TIME 184
#define RX_PHY_TIME 382
who wrote the openavb code?

pinealservo, 2015-08-11:
Closing; original question was too vague to answer and the PHY_TIME issue has been addressed.

closed 2015-08-11 by pinealservo