Avnu/OpenAvnu #220: Gstreamer plugin build issue

bharaths16, 2015-05-27:
Hi,
I am working on "open-avb-next" branch code and trying to compile the gstreamer plugin in it.
I am trying to build for TALKER side and following the procedure in the README file of "next-Open-avb/examples/gstreamer-avb-plugins".
Please find below error logs:bharath@EMMYS1124:/media/bharath/d_drive/Work/AVB/next-Open-avb/examples/gstreamer-avb-plugins/gst-plugin$ LDFLAGS='-L/media/bharath/d_drive/Work/AVB/next-Open-avb/lib/igb' CPPFLAGS='-I/media/bharath/d_drive/Work/AVB/next-Open-avb/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
[... 149 lines cut ...]
libtool: link: rm -fr  .libs/libgstplugin.exp .libs/libgstplugin.ver
libtool: link: /usr/bin/nm -B  .libs/libgstplugin_la-gstavbsink.o .libs/libgstplugin_la-avb.o   | sed -n -e 's/^.*[	 ]\([ABCDGIRSTW][ABCDGIRSTW]*\)[	 ][	 ]*\([_A-Za-z][_A-Za-z0-9]*\)$/\1 \2 \2/p' | sed '/ __gnu_lto/d' | /bin/sed 's/.* //' | sort | uniq > .libs/libgstplugin.exp
libtool: link: /bin/grep -E -e "_*(gst_|Gst|GST_).*" ".libs/libgstplugin.exp" > ".libs/libgstplugin.expT"
libtool: link: mv -f ".libs/libgstplugin.expT" ".libs/libgstplugin.exp"
libtool: link: echo "{ global:" > .libs/libgstplugin.ver
libtool: link:  cat .libs/libgstplugin.exp | sed -e "s/\(.*\)/\1;/" >> .libs/libgstplugin.ver
libtool: link:  echo "local: *; };" >> .libs/libgstplugin.ver
libtool: link:  gcc -shared  -fPIC -DPIC  .libs/libgstplugin_la-gstavbsink.o .libs/libgstplugin_la-avb.o   -L/media/bharath/d_drive/Work/AVB/next-Open-avb/lib/igb -lgstbase-1.0 -lgstcontroller-1.0 -lgstreamer-1.0 -lgobject-2.0 -lglib-2.0 -ligb -lpci -lrt -lpthread -lm  -pthread -O2   -pthread -Wl,-soname -Wl,libgstplugin.so -Wl,-version-script -Wl,.libs/libgstplugin.ver -o .libs/libgstplugin.so
/usr/bin/ld: /media/bharath/d_drive/Work/AVB/next-Open-avb/lib/igb/libigb.a(igb.o): relocation R_X86_64_32S against `.data' can not be used when making a shared object; recompile with -fPIC
/media/bharath/d_drive/Work/AVB/next-Open-avb/lib/igb/libigb.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [libgstplugin.la] Error 1
make[2]: Leaving directory `/media/bharath/d_drive/Work/AVB/next-Open-avb/examples/gstreamer-avb-plugins/gst-plugin/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/media/bharath/d_drive/Work/AVB/next-Open-avb/examples/gstreamer-avb-plugins/gst-plugin'
make: *** [all] Error 2
autoreconf: make failed with exit status: 2
autogen.sh failed

./configure is running properly but having issue with autogen.sh.
Any help would be appreciated.
Thanks in advance.

Regards,
Bharath S

swethachallawar, 2015-06-03:
Hi Bharath,

Please recompile your libigb with -fPIC option. It should resolve the error.
Check the error messages, that is what it says.

Thanks
Swetha

bharaths16, 2015-06-03:
Hi Shwetha,

Thanks very much.
Added -fPIC in next-Open-avb/lib/igb/Makefile and compilation completed.

I am getting another error when I am running talker side gstreamer plugin.
I have one Linux PC with Intel i210 card inserted, which is inturn connected to a ethernet switch.
After compiling the Gstreamer plugin for talker side successfully (commands as mentioned in the README file of "next-Open-avb/examples/gstreamer-avb-plugins"), I run the following below commands:
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/video$ gst-inspect-1.0 avbsinkFactory Details:
  Rank                     none (0)
  Long-name                AVB packet sender
  Klass                    Sink/Network
  Description              Send data over the network via AVB
  Author                   Symphony-Teleca

Plugin Details:
  Name                     avbsink
  Description              Ethernet AVB Sink Element
  Filename                 /home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src/.libs/libgstplugin.so
  Version                  1.0.0
  License                  LGPL
  Source module            my-plugin-package
  Binary package           GStreamer
  Origin URL               http://gstreamer.net/

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstBaseSink
                         +----GstAvbSink

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      ANY

Element Flags:
  no flags set

Element Implementation:
  Has change_state() function: gst_base_sink_change_state

Element has no clocking capabilities.
Element has no indexing capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Implementation:
      Has chainfunc(): gst_base_sink_chain
      Has custom eventfunc(): gst_base_sink_event
      Has custom queryfunc(): gst_base_sink_sink_query
      Has custom iterintlinkfunc(): gst_pad_iterate_internal_links_default
    Pad Template: 'sink'

Element Properties:
  name                : The name of the object
                        flags: readable, writable
                        String. Default: "avbsink0"
  parent              : The parent of the object
                        flags: readable, writable
                        Object of type "GstObject"
  sync                : Sync on the clock
                        flags: readable, writable
                        Boolean. Default: true
  max-lateness        : Maximum number of nanoseconds that a buffer can be late before it is dropped (-1 unlimited)
                        flags: readable, writable
                        Integer64. Range: -1 - 9223372036854775807 Default: -1 
[... 57 lines cut ...]
          RX bytes:734570604 (734.5 MB)  TX bytes:5236328 (5.2 MB)
          Interrupt:20 Memory:f7f00000-f7f20000 

eth1      Link encap:Ethernet  HWaddr a0:36:9f:4c:ab:2b  
          inet6 addr: fe80::a236:9fff:fe4c:ab2b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2142333 errors:0 dropped:20386 overruns:0 frame:0
          TX packets:20326 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:195945460 (195.9 MB)  TX bytes:2179803 (2.1 MB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8126 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8126 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:770671 (770.6 KB)  TX bytes:770671 (770.6 KB)

Please help.
Thanks in advance.

Regards,
Bharath S

bharaths16, 2015-06-05:
Hi,

Please help.
Thanks in advance.

Regards,
Bharath S

bharaths16, 2015-06-08:
Hi,

Now I am facing an issue when running "gst-inspect-1.0".
I compiled the gstreamer plugin as mentioned in README. 
Please find the log below for talker side:
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin$ LDFLAGS='-L/home/barath/work/avb/Open-AVB-open-avb-next/lib/igb' CPPFLAGS='-I/home/barath/work/avb/Open-AVB-open-avb-next/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
[... 250 lines cut ...]
make  all-recursive
make[1]: Entering directory `/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin'
Making all in src
make[2]: Entering directory `/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src'
make[2]: Entering directory `/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin'
make[2]: Leaving directory `/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin'
make[1]: Leaving directory `/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin'
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin$ export GST_PLUGIN_PATH=/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src/.libs
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin$ gst-inspect-1.0 avbsink

(gst-plugin-scanner:14590): GStreamer-WARNING **: Failed to load plugin '/home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src/.libs/libgstplugin.so': /home/barath/work/avb/Open-AVB-open-avb-next/examples/gstreamer-avb-plugins/gst-plugin/src/.libs/libgstplugin.so: undefined symbol: gst_app_src_end_of_stream
No such element or plugin 'avbsink'

As mentioned earlier, compilation is successful but "gst-inspect-1.0" is failing with above error. Even on the listener side, I am getting the same error message.
What am I missing now?

P.S: Last week, even gst-inspect-1.0 was successful on the same machine, as mentioned in my previous post. 

Any comments would be of great help.
Thanks in advance.

Regards,
Bharath S

pcub, 2015-06-18:
undefined symbol: gst_app_src_end_of_stream No such element or plugin 'avbsink' -means some elements are missing gstreamer. You can install gstreamer1.0 related packeges using synaptic,search gstreamer1.0 and install related packages.

pinealservo, 2015-08-11:
The gstreamer example never had a completely implemented build system and is currently unmaintained. I don't think anyone can offer any more help with this, but if you do decide to dig in and get a working build for it, we'd appreciate a pull request!

closed 2015-08-11 by pinealservo

rohan-lt, 2015-08-12:
We are working on Open-AVB-open-avb-next source code for a gstreamer plugins.
We have a two different pc with ubuntu 14.04 and NIC card(Intel i210) in both the system.

LISTENER side (WITHOUT SUDO)

1)Go to folder: cd /Open-AVB-Open-AVB-Next/examples/gstreamer-avb-plugins/gst-plugin/src 

and in Makefile.am file set
libgstplugin_la_SOURCES = gst_avb_playbin.c gst_avb_playbin.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h gst_app_src.c

2) LDFLAGS='-L/home/pcadmin/Downloads/AVB/lib/igb' CPPFLAGS='-I/home/pcadmin/Downloads/AVB/lib/igb' LIBS='-ligb -lpci -lrt -lpthread   -lm' ./configure

pcadmin@EMMYS1776:~/Downloads/AVB_fresh/examples/gstreamer-avb-plugins/gst-plugin$ LDFLAGS='-L/home/pcadmin/Downloads/AVB_fresh/lib/igb' CPPFLAGS='-I/home/pcadmin/Downloads/AVB_fresh/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm -lgstapp-1.0' ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert i686-pc-linux-gnu file names to i686-pc-linux-gnu format... func_convert_file_noop
checking how to convert i686-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
[... 888 lines cut ...]

1)Go to folder: cd /home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin/src
and in Makefile.am file set
libgstplugin_la_SOURCES = gstavbsink.c gstavbsink.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h

2)sudo LDFLAGS='-L/home/pcadmin/avb/avb_codeflow/lib/igb' CPPFLAGS='-I/home/pcadmin/avb/avb_codeflow/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' ./configure

3)sudo LDFLAGS='-L/home/pcadmin/avb/avb_codeflow/lib/igb' CPPFLAGS='-I/home/pcadmin/avb/avb_codeflow/lib/igb' LIBS='-ligb -lpci -lrt -lpthread -lm' ./autogen.sh

Now type 'make' to compile this module.

4)sudo make

5) export GST_PLUGIN_PATH=/home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin/src/.libs

6)Check weather gstreamer has installed the plugin or not:
gst-inspect-1.0 avbsink

root@pcadmin-OptiPlex-7010:/home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin# sudo gst-inspect-1.0 avbsinkNo such element or plugin 'avbsink'
root@pcadmin-OptiPlex-7010:/home/pcadmin/avb/avb_codeflow/examples/gstreamer-avb-plugins/gst-plugin# 

kindly help us what is missing or what need to be fixed in it. 

Thanks and Regards
Rohan

rohan-lt, 2015-08-13:
Hi all

the above mentioned problem is resolved using sudo -i.
talker is able to transmit the data using avbsink.

but on listener side we are using avbplaybin and it is not recieving any file, AS code is running but it is stuck at set state returned 2.

root@EMMYS1776:/home/pcadmin/Downloads/AVB_fresh/examples/gstreamer-avb-plugins/gst-plugin# gst-launch-1.0 avbplaybin talkerip=192.168.35.48 port=7600 interface=eth2
Setting pipeline to PAUSED ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
attaching to 0000:03:00.0
set state returned 2

kindly help us what is missing or what need to be fixed in it.

Thnaks & Regards
Rohan Gupta

andrew-elder, 2015-08-13:
@rohan-lt, please don't cross post the same question to three different github issues

davidcemin, 2015-08-13:
Also, this issue has already been closed.

rohan-lt, 2015-08-14:
sorry 
as the topic was same for all three thats why all posted

Please kindly help in resolving the matter

sachindes, 2016-09-30:
@rohan-lt, 

Is problem mentioned by you is resolved? If yes please share.
Also is there any dependency on GCC versions. 
For 4.9.2 GCC, we are getting below issue:

configure:2827: checking for gcc
configure:2843: found /usr/bin/gcc
configure:2854: result: gcc
configure:3083: checking for C compiler version
configure:3092: gcc --version >&5
gcc (Ubuntu 4.9.2-0ubuntu1~12.04) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
configure:3103: $? = 0
configure:3092: gcc -v >&5
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.9/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.9.2-0ubuntu1~12.04' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.9.2 (Ubuntu 4.9.2-0ubuntu1~12.04) 
configure:3103: $? = 0
configure:3092: gcc -V >&5
gcc: error: unrecognized command line option '-V'
gcc: fatal error: no input files
compilation terminated.
configure:3103: $? = 4
configure:3092: gcc -qversion >&5
gcc: error: unrecognized command line option '-qversion'
gcc: fatal error: no input files
compilation terminated.

Neha-Khan, 2016-11-25:
Hello,

I'm currently trying to explore on AVB in automotive, and i'm trying to build the simple talker and listener gstreamer plugin,
i have installed plugins successfully,
 currently my pc configuration is UBUNTU 16.04
and ethernet adapter details are as follows,
enp2s0    Link encap:Ethernet  HWaddr 64:00:6a:45:bd:22  
          inet addr:192.168.7.157  Bcast:192.168.7.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1373306 errors:0 dropped:0 overruns:0 frame:0
          TX packets:297978 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:305475845 (305.4 MB)  TX bytes:44299019 (44.2 MB)
but when i run the talker pipeline i'm getting the following issue from the pipeline, pls help me in understanding what is the issue, is there a problem regarding the configuration

 gst-launch-1.0 filesrc location="test.mp4" ! avbsink interface=enp2s0
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
connect failed (No such file or directory) - are you running as root?

I'm new to the gstreamer and also to AVB please help me in resolving these issue.

intel-ethernet, 2016-11-28:
Just a guess, as I haven’t run the avtp app myself … but :

“connect failed (No such file or directory) - are you running as root?”

… suggests you may not have installed the “avb” specific igb_avb.ko driver for the I210 (?).