Avnu/OpenAvnu #222: Issue on SYNC interval !

Hrusikesh4302, 2015-06-01:
Hi,
I am trying to measure the SYNC interval from the software and I am getting it around  129-131 msec. I tried to understand the snippet which starts the event timer on SYNC_INTERVAL_TIMEOUT_EXPIRES

				  wait_time =
					  ((long long)
					   (pow((double)2,getSyncInterval())*1000000000.0)) -
					  wait_time;
				  wait_time = wait_time > EVENT_TIMER_GRANULARITY ? wait_time :
					  EVENT_TIMER_GRANULARITY;
				  clock->addEventTimer
					  ( this, SYNC_INTERVAL_TIMEOUT_EXPIRES, wait_time );
   I am always getting the wait_time = 124000000, but when I measure the interval using gettimeofday() I am getting ~129 msec. I tried to narrow down the interval by adjusting the wait_time but it is not working expectedly. 

Thanks
Hrusikesh

Hrusikesh4302, 2015-06-03:
HI @davidcemin,
          Can you please help me in understanding the what the following function(line no 92) do. In my system this adds delay in each sync packet.

#ifndef ARCH_INTELCE	   
		net_lock.lock();
#endif
               
Thanks
Hrusikesh

davidcemin, 2015-06-03:
Can you please provide me the file and line where this line is defined please? 

thanks

davidcemin, 2015-06-03:
would it be this line ? 
https://github.com/AVnu/Open-AVB/blob/open-avb-next/daemons/gptp/linux/src/linux_hal_common.cpp#L93

Hrusikesh4302, 2015-06-03:
Hi,
Sorry I forget to mention the file, you hit the right one. I am getting a delay in msec right there, can you help me with that. It increases the sync interval from 125 --> 129 msec. 

Thanks
Hrusikesh

davidcemin, 2015-06-03:
This is the mutual exclusion lock (it calls pthread_mutex_lock on its very end). Im not sure where this lock is released though. 

What is the environment where you are running gptp ?

Hrusikesh4302, 2015-06-03:
Running on x86 platform, using I210 nic to send gptp packets. My understanding is, it calls 
bool TicketingLock::lock( bool *got ), line no 361 in the same file. But not sure why it takes 
such long delay ???

Thanks

davidcemin, 2015-06-03:
You might be running into scheduling problems and the amount of time you are seeing on that is your gptp hanging on a busy wait. What happens if you renice gptp and increase its priority ?

Hrusikesh4302, 2015-06-03:
I think renice will not help, there is a conditional wait in TicketingLock::lock() which should signalled by TicketingLock::unlock(). This unlock called from the followup message preparation, but there is some issue with it may be.

davidcemin, 2015-06-03:
I think it wont help, but it is very simple to verify, thats why i asked,. 
The conditional wait is due to the ticketing lock mechanism. 

If you suspect that this might be the issue, you could just replace the lock/unlock calls by the pthread_mutex_lock and unlock respectively. 

How did you measure the difference between 125 and 129 msec? Only by removing the lock  call ?

Hrusikesh4302, 2015-06-03:
I calculate the diff between two sync packets using gettimeofday(). In previous version of AVB stack I was getting 125 msec, after the implementation of the hw time stamping method I am facing the problem.

davidcemin, 2015-06-03:
Do you happen have the git hash where you did the first measurements?

jdkoftinoff, 2015-06-03:
Keep in mind that using gettimeofday() is not guaranteed to be accurate enough always. In addition, gettimeofday() is not a monotonic clock. See http://linux.die.net/man/3/clock_gettime <http://linux.die.net/man/3/clock_gettime>

Regards,

Jeff

davidcemin, 2015-06-03:
That is exactly what i thought jeff when I asked how that was measured. It can be misleading.

Hrusikesh4302, 2015-06-03:
Hi Jeff,
        The time we are getting in gettimeofday() may not be accurate enough, but when we calculate the diff between two sync packets it will always give us the interval i.e. ~125 msec, It will never jump to 129 msec. Also in before they have implementing the hw timestamping method, we were getting around 125 msec.

thanks

keyurparekh1, 2015-06-03:
Hi Jeff,

In later version of GPTP we are getting 129ms difference between two sync packet. We measure using wireshark. We run gptp on one machine which has Intel I210 card and capture sync messages on other machine using wireshark. 

In previous version of GPTP this gap was 125 ms. But that has increase. So we are trying to figure out why this gap increase. Can you give some hint about this issue?

jdkoftinoff, 2015-06-03:
Hello!

Can you give me more information? is it consistently 129 ms between sync ethernet frames?  i.e. are you saying that the time between the first SYNC and the eleventh SYNC frame is 1,290 ms and not 1,250 ms?

When wireshark measures the timing, how accurate is wiresharks timestamping? Additional processes on the wireshark computer can easily cause a 4 ms timestamp inaccuracy.

Jeff

keyurparekh1, 2015-06-03:
Hi Jeff,

You are correct. We are getting diff of 129ms between two consecutive sync message. And wireshark gives accuracy in us level. Right now we care about interval between two sync message. It should be 125ms. So we are trying to understand what could be the reason for extra 4 to 5 ms delay. And 129ms delay is pretty much constant.

jdkoftinoff, 2015-06-03:
So you are saying that the time difference between eleven SYNC messages is 1,290 milliseconds?

Jeff

keyurparekh1, 2015-06-03:
Hi Jeff,

In GPTP, when we start it, master will send sync packet first and than send follow up message. And after that again master will send sync packet. So diff between these two sync msg is 129ms.

jdkoftinoff, 2015-06-03:
Hi, 

What I am specifically asking for is the time between ten SYNC packets.

Jeff

keyurparekh1, 2015-06-03:
Hi Jeff,

Yes between ten sync packets time gap between each of them is 129ms. so its 1290ms.

jdkoftinoff, 2015-06-03:
Thank you!

The timing for the Sync packets is most definitely wrong! this is very important for us to diagnose the actual reason why the scheduling is wrong.  Did the system running wireshark change in anyway (kernel version/hardware?) 

Regards,
Jeff

keyurparekh1, 2015-06-04:
Hi Jeff,

Sorry for late reply. Can you tell me what exactly you want to know? what do you mean by system running wireshark change in anyway? 

We are running wireshark on Fedora machine.

keyurparekh1, 2015-06-08:
Hi Jeff,

So are you going to check GPTP daemon again for this 129ms delay?

jdkoftinoff, 2015-06-08:
No, I’m sorry, I am not available for this now.

Regards,
Jeff

keyurparekh1, 2015-06-08:
okie. So Can you tell, who can help me in this problem?

jdkoftinoff, 2015-06-08:
All I can say is try determine how the 125ms is being scheduled and find out what could be causing that timer to be consistently wrong

Regards,
Jeff

andrew-elder, 2015-06-25:
Can I close this one out? Is it solved?

keyurparekh1, 2015-06-26:
Hi Andrew,

Still in GPTP we are getting delay of 129ms between sync packets. But we haven't try to download latest one. So is it going to solve or we need to take care at our side? Let me know the same.

andrew-elder, 2015-06-26:
Andrew Elder is out of the office until July 6th. If you have an urgent technical issue, please contact support@audioscience.com.

kbstanton, 2015-06-29:
Does 129ms cause a problem?

Hrusikesh4302, 2015-06-29:
Yes, again it is not 129msec all the times but fluctuates in between 129-131msec which is undesirable.

intel-ethernet, 2015-06-29:
I’ll point out that if you are using a ‘regular’ system timer (e.g. a POSIX timer a’la timer_create()), you will likely be driven off of some variant of the system timer tick (10 msec? 12.5msec? 15 msec? ‘tickless’ – depends on your kernel configuration). So the system may round your timer to the ‘nearest’ interval. Also – a lot of software registers ‘maintenance’ tasks to run on the same timer – like polling and saving system statistics, running garbage collection, etc – your ‘timer’ routine may run after a large number of other routines have fired as well.

Net effect – these timers are usually not precise on their execution.

ekm

Hrusikesh4302, 2015-06-29:
Hi @intel-ethernet ,
  
       We also initially suspected the linux timer_create(). To confirm this we were checking how frequently the calling sequence hits the SYNC_INTERVAL_TIMEOUT_EXPIRES label in processEvent() timer handler routine, with calling the below line at the start of this label
				  clock->addEventTimer
					  ( this, SYNC_INTERVAL_TIMEOUT_EXPIRES, wait_time );
We found the this label is called in every 125msec with ~60 usec resolution which is acceptable. Also we found that the delay added by the below snippet,
					while (ts_good != 0 && iter-- != 0) {
						timer->sleep(req);
						wait_time += req;

						if (ts_good != -72 && iter < 1)
							XPTPD_ERROR(
								"Error (TX) timestamping Sync (Retrying), "
								"error=%d", ts_good);
						ts_good =
							getTxTimestamp
							(sync, sync_timestamp, 
							 sync_timestamp_counter_value, iter == 0);
						req *= 2;
					}
where getTxTimestamp() gets the hw TS to be filled in the followup packet. Somehow Even after we subtract wait time before next sync scheduling it does not work properly. I mostly suspect the following lock we are using is causing the issue, 
net_lock.lock(); 
We are using x86 platform and fedora-20 OS. 

Thanks
Hrusikesh

intel-ethernet, 2015-06-29:
If it’s the lock you suspect, have you tried instrumenting the lock and unlock calls with timestamps to get an idea of any threads holding onto the lock for 4msec?

Atul140, 2017-11-03:
Hi,

I am also facing the same Issue for time stamps.
Is there any solution that any one have tried which is working for this Issue.