Avnu/OpenAvnu #224: Issue while running simple_talker

bharaths16, 2015-06-02:
Hi,

I have PC with Intel i210 Card inserted.
I have downloaded Open-Avb-Next branch source code and successfully built all the necessary binaries (Gptp, mrpd, kmod, simple talker, Simple listener) using travis.sh script for running simple_talker and simple_listener.

Please find below the logs when I run different binaries:

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/kmod/igb$ sudo modprobe i2c_algo_bit
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/kmod/igb$ sudo modprobe dca
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/kmod/igb$ sudo modprobe ptp
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/kmod/igb$ sudo insmod ./igb_avb.ko
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/kmod/igb$ dmesg
[  351.788304] igb_avb: module verification failed: signature and/or  required key missing - tainting kernel
[  351.789438] Intel(R) Gigabit Ethernet Network Driver - version 5.2.9.4_AVB
[  351.789439] Copyright (c) 2007-2014 Intel Corporation.
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/kmod/igb$

Does this mean that the driver is not inserted?
But in /dev and lsmod, i see the driver to be inserted.

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ ls -l /dev/igb_avb
crw------- 1 root root 10, 56 Jun  2 13:22 /dev/igb_avb

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ lsmod | grep igb
igb_avb               194168  0
igb                   179302  0
i2c_algo_bit           13413  2 igb,radeon
dca                    15130  2 igb,igb_avb
ptp                    18933  3 igb,e1000e,igb_avb

After this, I am running different dameons.

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/daemons/gptp/linux/build/obj$ sudo ./daemon_cl eth1
[sudo] password for barath: 
ERROR at 634 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay
*** Announce Timeout Expired - Becoming Master
New Grandmaster "A0:36:9F:FF:FE:4C:AB:2B" (previous "00:00:00:00:00:00:00:00")

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/daemons/mrpd$ sudo ./mrpd -mvs -i eth1
[sudo] password for barath: 
process_events()

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_talker$ sudo ./simple_talker -t 2 -i eth1
[sudo] password for barath: 
attaching to 0000:03:00.0
attach failed! (No such file or directory)
connect failed (No such file or directory) - are you running as root?
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_talker$ 

Why this error? Do I need to do anything else?
Please let me know.

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_listener$ sudo ./simple_listener -i eth1 -f ss.wav
[sudo] password for barath: 
Waiting for talker...
Msg: SJO D:C=6,P=3,V=0002,N=3 R=000000000000 VP/IN

Msg: VNE 0002 R=000000000000 VN/MT

Msg: SLE D:C=6,P=3,V=0002,N=3 R=000000000000 QA/MT

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_talker$ ifconfig
eth0      Link encap:Ethernet  HWaddr f8:b1:56:bf:ba:ae  
          inet addr:192.168.34.193  Bcast:192.168.39.255  Mask:255.255.248.0
          inet6 addr: fe80::fab1:56ff:febf:baae/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:93222 errors:0 dropped:1351 overruns:0 frame:0
          TX packets:1509 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:9846326 (9.8 MB)  TX bytes:206437 (206.4 KB)
          Interrupt:20 Memory:f7f00000-f7f20000 

eth1      Link encap:Ethernet  HWaddr a0:36:9f:4c:ab:2b  
          inet6 addr: fe80::a236:9fff:fe4c:ab2b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:91370 errors:0 dropped:895 overruns:0 frame:0
          TX packets:1174 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7815128 (7.8 MB)  TX bytes:114086 (114.0 KB)
          Memory:f7c00000-f7d00000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:523 errors:0 dropped:0 overruns:0 frame:0
          TX packets:523 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:55355 (55.3 KB)  TX bytes:55355 (55.3 KB)

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_talker$ uname -a
Linux EMMYS1775 3.13.0-32-generic #57-Ubuntu SMP Tue Jul 15 03:51:08 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

Let me know if you need anymore info.
Thanks in advance.
Regards,
Bharath S

intel-ethernet, 2015-06-02:
You probably want to ‘rmmod igb’ before ‘insmod igb_avb’.

ekm

bharaths16, 2015-06-03:
Hi,

Thanks very much.
I did "rmmod igb" and then "insmod igb_avb.ko".
Now facing another error.
I have one Linux PC with Intel i210 card inserted, which is inturn connected to a ethernet switch.
On this, I run the below commands:

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ sudo modprobe i2c_algo_bit
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ sudo modprobe dca
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ sudo modprobe ptp
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ sudo insmod kmod/igb/igb_avb.ko

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ dmesg
[99638.047589] Intel(R) Gigabit Ethernet Network Driver - version 5.2.9.4_AVB
[99638.047593] Copyright (c) 2007-2014 Intel Corporation.
[99638.048421] igb_avb 0000:03:00.0: irq 44 for MSI/MSI-X
[99638.048432] igb_avb 0000:03:00.0: irq 45 for MSI/MSI-X
[99638.048440] igb_avb 0000:03:00.0: irq 46 for MSI/MSI-X
[99638.048449] igb_avb 0000:03:00.0: irq 47 for MSI/MSI-X
[99638.048457] igb_avb 0000:03:00.0: irq 48 for MSI/MSI-X
[99638.126811] igb_avb 0000:03:00.0: added PHC on eth1
[99638.126816] igb_avb 0000:03:00.0: Intel(R) Gigabit Ethernet Network Connection
[99638.126819] igb_avb 0000:03:00.0: eth1: (PCIe:2.5GT/s:Width x1) 
[99638.126822] igb_avb 0000:03:00.0 eth1: MAC: a0:36:9f:4c:ab:2b
[99638.127131] igb_avb 0000:03:00.0: eth1: PBA No: H30936-003
[99638.133928] igb_avb 0000:03:00.0: LRO is disabled
[99638.133934] igb_avb 0000:03:00.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[99638.975963] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[99638.976481] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[99640.789105] igb_avb 0000:03:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[99640.789325] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ lsmod | grep igb
igb_avb               194168  0 
dca                    15130  1 igb_avb
ptp                    18933  2 e1000e,igb_avb

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ ls -l /dev/igb_avb 
crw------- 1 root root 10, 56 Jun  3 16:54 /dev/igb_avb

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next$ ifconfig -a
eth0      Link encap:Ethernet  HWaddr f8:b1:56:bf:ba:ae  
          inet addr:192.168.34.193  Bcast:192.168.39.255  Mask:255.255.248.0
          inet6 addr: fe80::fab1:56ff:febf:baae/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7147568 errors:0 dropped:137622 overruns:0 frame:0
          TX packets:68874 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:760843064 (760.8 MB)  TX bytes:5327323 (5.3 MB)
          Interrupt:20 Memory:f7f00000-f7f20000 

eth1      Link encap:Ethernet  HWaddr a0:36:9f:4c:ab:2b  
          inet6 addr: fe80::a236:9fff:fe4c:ab2b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4450 errors:0 dropped:0 overruns:0 frame:0
          TX packets:34 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:389557 (389.5 KB)  TX bytes:6400 (6.4 KB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9038 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9038 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:848771 (848.7 KB)  TX bytes:848771 (848.7 KB)

Do I need to assign ip address to eth1?
		  
barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/daemons/gptp/linux/build/obj$ sudo ./daemon_cl eth1
[sudo] password for barath: 
ERROR at 634 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay
*** Announce Timeout Expired - Becoming Master
New Grandmaster "A0:36:9F:FF:FE:4C:AB:2B" (previous "00:00:00:00:00:00:00:00")

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/daemons/mrpd$ sudo ./mrpd -mvs -i eth1
[sudo] password for barath: 
process_events()

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_talker$ sudo ./simple_talker -t 2 -i eth1
[sudo] password for barath: 
attaching to 0000:03:00.0
detected domain Class A PRIO=3 VID=0002...
advertising stream ...
awaiting a listener ...
VNE 0002 R=000000000000 VN/MT
 unhandled from mrpd
VNE 0002 R=000000000000 VN/MT
 unhandled from mrpd

"unhandled from mrpd" - what is this error? Am I missing anything?

barath@EMMYS1775:~/work/avb/Open-AVB-open-avb-next/examples/simple_listener$ sudo ./simple_listener -i eth1 -f ss.wav
[sudo] password for barath: 
Waiting for talker...
Msg: SJO D:C=6,P=3,V=0002,N=3 R=000000000000 QA/IN

Msg: VNE 0002 R=000000000000 VN/MT

Msg: SLE D:C=6,P=3,V=0002,N=3 R=000000000000 QA/MT

I am new to AVB protocol stack.
Where can I find good guide for understanding AVB stack and source code flow for avb-open-next branch?
Any documentation available?
Also, in wireshark, what is filter rule to be applied for filtering out AVB packets?

Please let me know if anymore details are required.
Thanks in advance.

Regards,
Bharath S

eric-mann, 2015-06-04:
maybe I missed a question/issue in the log ... but 
(a) you need to make sure the interface is 'up' - it doesn't need an IP address per-se, but it does need to be up. 
(b) it is advised to stop any pesky 'network manager' from trying to assign a DHCP address to the interface ... it frequently downs/ups the interface which disrupts simple talker ...
(c) some of the messages out of mrpd are 'benign' like the 'unhandled' ones ...

pinealservo, 2015-08-11:
Closed due to inactivity.  Please reopen if more support on this is needed.

closed 2015-08-11 by pinealservo