Avnu/OpenAvnu #225: ways to separate avb traffic at the receiver side

virupaxss, 2015-06-02:
Hi
in the simple_listener/simple_listener.c libpcap is used to receive the packets that are destined to stream id the listener is interested in, is there no other way to receive the packets other than using the libpcap. like using the ports in TCP connection ?

intel-ethernet, 2015-06-02:
There are many ways. It just depends on how much code you want to write.

The biggest issue is the common 1722 encapsulated frames are Layer-2 (L2) – Ethertype-based frames, not IP/UDP or IP/TCP. As a result, you can’t just open up a socket on a UDP or TCP port and receive frames.

You could however register to receive specific Ethernet multicast addresses – as each AVB/TSN stream has a unique multicast address – and receive frames via that method. This only really works on Unixes (Linux) – Windows has a variety of restrictions what you can receive over a socket which usually prevents this (if that is a concern to you). On windows you need a kernel-mode protocol driver usually … or some packet capture library/driver …

Hope this helps
ekm

virupaxss, 2015-06-02:
The reason I wanted to know is , all the application in a node have to look into all the packets to check if there is a matching stream id so as to come to know the packet is for me. this seems like more work taken if there are many streams running to a node. 

is creating virtual nodes to a avb ethernet interface based on the vlan id's and use those vlan id's with the stream id's to differentiate the avb packets a good idea to use ? 

like, application ABC  will have vlan id A and will listen to a virtual ethernet interface created using vconfig with vlan id A. This will not make the application  ABC too grep for all the incoming packets to get a matching stream id packets.

eric-mann, 2015-06-04:
Current* common practice is to keep all streams of a given AVB Class (user priority) on the same VLAN. This is what is communicated with the 'Domain' attributes which you can see reference to in simple_talker. 

I think the easiest method is to listen to a specific ethernet multicast address (on the VLAN which corresponds to the SR class you want to receive - Class A or Class B). So on Linux, create a vlan virtual node, bind to the interface in your app, then register the multicast address of the stream you want to receive.

*IEEE 802.1Qca suggests using multiple VLANs - possibly one per TSN stream - to implement path reliability ... I personally think the protocol overall has some scaling issues (as SRP did - now we are talking about pruning...) as ISIS PDUs are typically limited to 1492 bytes and the protocol looks like it uses a lot of bytes to encode the paths ... but I digress ...

pcub, 2015-06-18:
https://github.com/AVnu/Open-AVB/issues/244

pinealservo, 2015-08-11:
Closing, as the asker hasn't asked for additional clarification in a long time.

closed 2015-08-11 by pinealservo