Avnu/OpenAvnu #230: consolidate IPC definition across operating systems

andrew-elder, 2015-06-04:
Currently linux and Windoze have separate definitions for what information gets shared with a client application. It would be best for maintenance purposes if a single definition existed. After all, client applications in both Windows and Linux need to share use the same information in the same way.

davidcemin, 2015-10-28:
Currently we have two files called ipcdef.hpp, and they are located at:
- linux/src/ipcdef.hpp
- windows/daemon_cl/ipcdef.hpp

First thing im gonna do is merge these two files into a common/ipcdef.hpp and separate windows/linux implementations with ifdefs.

andrew-elder, 2015-10-28:
Sounds like a most excellent plan!

davidcemin, 2015-10-28:
There are two types of IPCs on these files. Windows uses named pipes and linux uses shared memory. Do we want to integrate that as well?

andrew-elder, 2015-10-28:
I think we should try to keep the data separate from the transport/plumbing.

davidcemin, 2015-10-28:
I wrote a similar application to the named_pipe_test to test the shared memory interface. Still need some work, but I can read the data from the shared memory. On the example below, I started reading the values right after starting daemon_cl (you can see it by the pdelay incrementing value count)

--------------------------------------------
Frequency 800000000 Hz
ml phoffset -141710
ml freq offset 1.000000
ls phoffset -24758142253839
ls freq offset 1.000000
local time 1446087063753931143
sync count 1
pdelay count 3
Port State 9
--------------------------------------------
Frequency 800000000 Hz
ml phoffset -3206
ml freq offset 0.999989
ls phoffset -24758142384916
ls freq offset 1.000003
local time 1446087064756849142
sync count 8
pdelay count 4
Port State 9
--------------------------------------------
Frequency 800000000 Hz
ml phoffset -194
ml freq offset 0.999999
ls phoffset -24758142380466
ls freq offset 0.999993
local time 1446087065760508166
sync count 16
pdelay count 5
Port State 9
--------------------------------------------
Frequency 800000000 Hz
ml phoffset -4
ml freq offset 1.000000
ls phoffset -24758142373180
ls freq offset 0.999992
local time 1446087066763777846
sync count 24
pdelay count 6
Port State 9

davidcemin, 2015-10-28:
I did not test it on a windows platform (i do not have any windows machine ready to test that). I really would appreciate if someone that normally uses it in windows could compile it from my branch(https://github.com/davidcemin/Open-AVB/tree/task-gptpV2) an give it a try on the named_pipe_test application (which I never used before, but I assume it works)

andrew-elder, 2015-11-04:
Is the a reason we wouldn't/shouldn't move the Windows interface to be shared memory as well? I don't like all the Windows specific code in a shared header file. My 2c at least.

davidcemin, 2015-11-05:
Thats a good point. I thought about it when I was migrating it. Well, one thing I could do is to keep the ipc definition on the common header and move back windows specific stuff to the windows directory.

davidcemin, 2015-11-09:
I removed windows specific implementation and added it to windows sub-directory. Also, I created a file for Linux specific defines to be consistent. We do have two different implementations, and what can be consolidated is the IPC format itself, which is done. Please let me know your thoughts, but I think this is the best we could get without changing the IPC itself.

closed 2016-03-16 by davidcemin