Avnu/OpenAvnu #237: HW queue priorities

keyurparekh1, 2015-06-08:
Hi,

I found one write up which says queue priority should be as per below.
https://avb.statusbar.com/page/developer-faq/

Highest Priority: #1 802.1AS messages
Next Priortity : #2 IEEE 1722 media packet messages
Next Priority : #3 IEEE 802.1Qat (SRP) stream reservation messages
Lowest priority : #4 all other control messages (IEEE 1722.1, MAAP) and IPv4 and IPv6.

But I check intel code and in that GPTP traffic is routing in Q3, which is lowest priority. So can you tell me, why it is going to lowest priority? As per above explanation, it should be in highest priority.

Let me know if my understanding is wrong.

Thanks & Regards,
Keyur Parekh

intel-ethernet, 2015-06-09:
I think it depends on the ‘customer’ which has traffic has higher priority over another, and furthermore the priority discussed here here isn’t entirely consistent IMO.

There is 802.1p priority  (a PCP field in the packet), and then there is queue processing priority between the host and the LAN (which may or may not be linked to the 802.1p priority). As an example,  the industrial segment would state the TSN isochronous real time frames (which loosely ~ 1722 media frames in AVB) are higher priority than 802.1AS or anything else for that matter, given their sensitivity to transmission-time jitter.

AVB (media) usages on the other hand are more relaxed with transmission time and can tolerate jitter from pre-emption – they transmit with a periodic presentation  time embedded in the frames which handles path delay variation between listeners.

With the exception of the IEEE 1722 Class A/Class B frames - many of these protocols – SRP, MAAP, etc - are not tagged with a user-priority, which defaults to best-effort.

In short, I’m not certain that the I210 queue priority implementation leads to any specification non-conformance issues, although the implementation may not be acceptable to all depending on  usage.

Sorry for the long winded answer.
ekm

https://avb.statusbar.com/page/developer-faq/

jdkoftinoff, 2015-06-09:
My opinion is that this is a “Quality of Implementation Issue”

AVB was carefully designed so that media stream interruption would not occur due to network congestion.

At the first level, this is achieved by the stream reservation protocol protecting a specific VLAN Tag/Priority combination.

But media streaming can not occur if time synchronization is not achieved.  Time synchronization is of primary importance.  3 lost pDelay messages or responses in a row cause asCapable to transition to false. This causes all streams to or from that port to stop and this is disastrous to installations. lost sync messages will cause syncs to be synthesized down the line via the 802.1AS PortSyncSyncSend state machine, and this can degrade synchronization quality down the line as the correction field increases by sync timeout at every hop.

Media streaming relies on the stream reservation protocol. Even though SRP is a relatively slow protocol, it still is a real time protocol.

Typically, media streaming also relies on the MAAP protocol.  An undefended Multicast MAC Address range because of lost packets can cause stream interruption.

While media transport theoretically has a lot of ‘slack’, they still need to be launched within the appropriate observation interval otherwise the msrp_accumulated_latency value calculated by the network via SRP becomes irrelevant or unusable.

While typical designs implementing the  802.1AS daemon, SRP daemon, MAAP daemon, and media streaming daemon all as non-real-time user level tasks can be made to work in normal circumstances, without real time guarantees in the O/S scheduler and without priority queues for this time sensitive network management traffic (gPTP,SRP,MAAP)

In a perfect world, the operating system and network driver would allow us to have these 4 priority queues for packet transmission and these would be fed by a real time operating system.

Since we do not have this with Linux, RTLinux, or Xenomai based linux yet, many designers implement this functionality within an FPGA or ASICS. 

Without these protocols having their own queues, you can be at a risk for stream interruption simply due to the following quite normal situations:

* ping flood

* cpu usage on linux hits 100% for too long, delaying SYNC FUP / PDELAY FUP / MAAP Defend / SRP LeaveAll Defend

* a funny one which I ran into: A large daemon which had to run with mlockall() (xenomai real time thread), runs fork() in order to exec a subtask. fork() then has to copy all memory pages instead of relying on COW; during this time all daemons paused for almost 1 second (2.1 GHz Xeon) - this kills asCapable and all streams to/from the device, even though all media packets were handled by an FPGA.

At this point, I’d like to point out the Linux Foundation’s RT Linux initiative. Please support it:

http://www.linuxfoundation.org/programs/rt-linux-initiative <http://www.linuxfoundation.org/programs/rt-linux-initiative>

We want to be able to take it further - Once rt linux becomes mainstream in the linux kernel proper, we want to be able to add socket options to the network stack to allow for real time threads to send/receive real time protocols via real time priority queues. Until then, it will remain really hard to have any sort of low latency reliable media transport over ethernet without leveraging custom FPGA or ASICs.

Best Regards,
Jeff Koftinoff

intel-ethernet, 2015-06-09:
Agree 100%. More queues are better … to a point. And use the knobs on the queue priorities consistent with your architecture.

ekm

http://www.linuxfoundation.org/programs/rt-linux-initiative <http://www.linuxfoundation.org/programs/rt-linux-initiative>

MythiliR, 2015-07-23:
how to configure the hardware queueing in kernel? Kindly suggest the solution

pinealservo, 2015-08-11:
@MythiliR See the readme here: https://github.com/AVnu/Open-AVB/tree/open-avb-next/kmod/igb

I think this question has been answered; closing the issue.

closed 2015-08-11 by pinealservo