Avnu/OpenAvnu #239: gstreamer listener and talker

swethachallawar, 2015-06-11:
Hi,

I am able to play a mjpeg video successfully using gstreamer plug in of Open AVB.
However, the listener plug in hangs even after talker has finished sending data. ctrl+c does not help.
Also, on ctrl+c, the signal handler of plugin does not get invoked. looks like the sigint signal handler of gstlaunch is getting invoked. Any suggestions on how to ensure the plugin signal handler gets invoked.

Also, on the talker plugin, the gstlauch on seeing EOS exits the plugin. However, i want the plugin also to see EOS and handle the talker exit inside plugin.
Please let me know your suggestions.

Thanks
Swetha

swethachallawar, 2015-06-11:
Any comments on why the avbplaybin plugin hangs after avbsink stops sending data?
Also, how I can I handle SIGINT(ctrl+c) in avbplaybin code?

I am trying to integrate mrp with open avb gstreamer. Hence I need the clean up (realted to mrp, igb etc) to happen before the plugins exit.

Any comments would be of great help.

Thanks
Swetha

pinealservo, 2015-08-11:
If you are running gstlaunch as your wrapper for your gstreamer pipeline, you're stuck with gstlaunch's signal handlers. I believe gstlaunch is more of a tool for prototyping gstreamer pipelines than for implementing full-fledged applications. It's a fairly thin wrapper around gstreamer's pipeline composition libraries, so you should be able to just wrap the pipeline you want in your own program.

Beyond that, I don't think anyone here can help much. Whoever contributed the OpenAVB gstreamer example doesn't seem to be active in the project anymore and so it's essentially unmaintained.

closed 2015-08-11 by pinealservo