Avnu/OpenAvnu #243: correctionField assumes Little-endian Machine.

davidcemin, 2015-06-17:
We should be able to detect endianness set this value accordingly.
Code is here:
https://github.com/AVnu/Open-AVB/blob/open-avb-next/daemons/gptp/common/ptp_message.cpp#L522

davidcemin, 2016-03-28:
I confess that im not sure what I wanted when I opened this issue. I think that the idea is to discuss the way we are storing information within ptp (host/network order). Currently, several internal variables are stored in network order, and changing the whole thing would have a significant cost (code/time). 

Is that what we want to do or can we accept the things the way they are today ?

andrew-elder, 2016-03-31:
Just fix referenced code, not the whole project.

closed 2016-04-01 by davidcemin