Avnu/OpenAvnu #244: Simple_listener ethernet header does not match

pcub, 2015-06-18:
Hello all, Since I am working on one xmos board with one pc.
I am using simple_listener example, to capture audio samples via ethernet. However I am stuck in it.
The result of  
    fprintf(stdout,"eth type: 0x%02x\t", eth_header->type[0]);
    fprintf(stdout,"eth type: 0x%02x\n", eth_header->type[1]);
give 0x2c, 0x4c,0x00,0x36 etc.
but in 
   u_char glob_ether_type[] = { 0x22, 0xf0 };
...
   if (0 == memcmp(glob_ether_type,eth_header->type,sizeof(eth_header->type)))
	{		
		test_stream_id = (unsigned char*)(packet + ETHERNET_HEADER_SIZE +    SEVENTEEN22_HEADER_PART1_SIZE);
...

my understanding is if glob_ether_type and eth_header->type are equal, then audio samples are captures.
Unfortunately, eth_header type result never shows the 0x22 and 0xf0 values.
In wireshark packet content (which is sets in bottom), there is not 0x22 and 0xf0 values. It only shows 0x2c, 0x4c,0x00,0x36.

In xmos side, they defined ethernet type as:
for avb ethernet type :
#define AVB_1722_ETHERTYPE          (0x22f0)
and for srp ethernet type :
#define AVB_SRP_ETHERTYPE (0x22ea)
I attached the result of wireshark: www.xcore.com/forum/download/file.php?id=981
What is the problem?

davidcemin, 2015-06-18:
Without even reading the code, this looks suspicious to me:
if (0 == memcmp(glob_ether_type,eth_header->type,sizeof(eth_header->type)))

eth_header->type is a pointer apparently and that comparison is not be comparing the right size.

Please, post a link with the correct line you are debugging so it helps understanding the context.

pcub, 2015-06-18:
https://github.com/AVnu/Open-AVB/blob/open-avb-next/examples/simple_listener/simple_listener.c
above comparison starts from line 101.

pinealservo, 2015-06-20:
@davidcemin , this is actually one of the few places in C where an lvalue of array type is not converted to a pointer type. The expression `sizeof (eth_header->type)` evaluates to 2, not the size of the platform's pointer, because the type of the object `eth_header->type` is an array of two unsigned characters. This 

@pcub , there aren't any 1722 audio packets in that capture, but you've got a bit of most of the other AVB protocols. You've got a lot of PTP Sync/Follow_Up frames, a couple of MSRP frames, and then two frames with the ethertype you're looking for, 0x22F0. But 1722 has Control as well as Data frames; these are both Control frames. One is MAAP, the other is a 1722.1 avdecc entity announcement thing. But there's no audio at all for simple_listener to capture.

I actually just ran into this today; if you're using the latest XMOS code, it expects a 1722.1 controller in the system to direct things. In its default state, it won't stream until you've got PTP and MSRP satisfied that a reservation is in place, and it won't set up MSRP reservations without a 1722.1 controller telling it to. You can flip a #define in one of the config files in the top-level project to tell your XMOS device to act as a controller as well, and that will at least get two of them to talk back-to-back when you hook them together, but I'm not sure how it'll react with a PC running OpenAVB code. You might need to hack the XMOS a bit to make it a bit more permissive.

pcub, 2015-06-22:
Thank you pinealservo,
Regarding to your comment, I want to know specifically for some point.
1. How did you know there is two frames with 0x22F0?
2. I only tested with xmos code from xcom.com (AVB-Daisy-Chain-Software_1.0.6, not from github)
   I enabled: avb_conf.h
/** Enable 1722.1 Controller functionality on the entity. */
#define AVB_1722_1_CONTROLLER_ENABLED 0 //zero to one

3 .I was actually upload the wrong wireshark capture. (sorry for this).
Now I attached files:
1. w1 for original xmos source code AVB-Daisy-Chain-Software_1.0.6
https://www.dropbox.com/s/k0q665108a2ntox/w1.pcapng?dl=0
2. w2 for AVB_1722_1_CONTROLLER_ENABLED 1 enabled captures.
https://www.dropbox.com/s/gpz441oxt7o44z0/w2.pcapng?dl=0
3. avdecc and openavb running then captures wireshark.
https://www.dropbox.com/s/otyhyfahpmfivuk/w3.pcapng?dl=0

Even I enable avb1722_1 controller it pc seems like not capturing audio samples.

pinealservo, 2015-06-22:
1. I dumped the packets with tshark (the terminal-only version of wireshark) and searched for 0x22F0. You can also filter based on ethertype in wireshark.

2. That version seems to correspond with what's current on github: https://github.com/xcore/sw_avb_dc

3. Those wireshark captures look pretty similar to the previous one. There's a lot of PTP traffic and a bit of MVRP and MSRP. One thing that sticks out to me is that your PC is declaring an SRP Domain attribute, but I only see LeaveAlls from the XMOS. Both sides must declare the same SRP Domain attribute before SRP stream reservations can be made, and the XMOS talker won't talk without a reservation.

Do you have the serial console output from the XMOS turned on? The XTag needs to be configured to XScope mode, and there may be some switch on the SliceKit hardware to tell it to send the signals to the XTag rather than the other boards. The console messages should tell you the state changes that occur with the various protocols and give you a good idea of why it's behaving like it is.

pcub, 2015-06-22:
Thank you pinealservo. :D
I will try with serial console xmos (it is great idea what is going on inside xmos).
By the way, the setup that I want to make is PC as listener and xmos daisy chain as talker.
When I run ./daemon_cl eth1 -L      then it switches to master immediately.
1.Do you have comment for this?
2. In openavb-next/gptp source code there are several declaration of mach addresses.
   for example, in mrpd.c and mrpw.c there are unsigned chan STATION_ADDR[] = {}.
I set above address as mac address of eth1 where i210 is connected with xmos.
   in simple_listener.c there is char filter_exp[] ="ether dst ..."
I set it above mac address.
Whole source codes are poor comments and description.

Is my assigned values right? What should those address be?
I tried also with xmos's mac address last time.
I set listener_mrp_client.c volatile int talker as 1, then running simple_listener. If it is 0, then I am not able to run simple_listener.
I will try listener.c soon, Also there is unsigned char glob_dest_addr[]= {};
I will set it as mac address of eth1.

Thank you.

pcub, 2015-06-25:
Here is the serial console output of xmos side.
first message is as following:
   PTP Port 0 Role: Master
   PTP Port 1 Role: Master
   Setting clock source: INPUT_STREAM_DERIVED
   MAAP reserved Talker stream #0 address: 91:E0:F0:0:28:9C
//after running gptp (daemon_cl eth1 -L ) on computer side
   PTP Port 1 Role: Slave
   PTP Port 0 Role: Master
   PTP sync locked
// after mrpd
// after running simple_listener on computer side:
Added stream:
 ID: 00
AP: MSRP_LISTENER 0:1:0:0	 MRP_EVENT_RECEIVE_LEAVE_ALL: MRP_LO -> MRP_UNUSED 
1722 router: Removed entry for stream 00
// after stop running simple_listener on computer:
Removed stream:
 ID: 00
 PTP threw away Sync outlier (master_diff 158142560)

Thank you.

pinealservo, 2015-06-28:
I suspect that the XMOS talker won't actually advertise a stream unless you
modify the code so that it always does it or unless you use a 1722.1
controller to tell it to. Once the XMOS declares a Talker Advertise
attribute and your PC listener declares a matching Listener Ready
attribute, the XMOS device should start sending its stream.

pcub, 2015-06-29:
Now I am stuck in 1722.1. Can you guide me for modifying the code.
Which part should be changed?
I added line in avb_1722_1.xc 
#if (AVB_1722_1_CONTROLLER_ENABLED)
    avb_1722_1_acmp_controller_init();
#endif
#if (AVB_1722_1_TALKER_ENABLED)
    // Talker state machine is initialised once MAAP has finished
    avb_1722_1_acmp_talker_init(); // added this line of code
#endif
#if (AVB_1722_1_LISTENER_ENABLED)
    avb_1722_1_acmp_listener_init();
#endif

but it does not make sense.
Have you tested xmos as talket and pc as listener?
Thank you.

pcub, 2015-07-01:
Can anyone see the file and check whether are ok or not. (I am not expert so...)
there is wireshark capture: https://www.dropbox.com/s/axv7fj0jil24f6b/w0.pcapng?dl=0
simple_listener generated wave file: https://www.dropbox.com/s/tdq6tmgo9s788cs/ww18.wav?dl=0
But this file size is only 44 bytes.

pcub, 2015-07-01:
My problem is. talker side does not work in xmos side properly.
in avb_1722_1_acmp.c
there is:
void avb_1722_1_acmp_talker_init()
{
acmp_talker_state = ACMP_TALKER_WAITING:
}

i tracked the console result and came here that causes the problem.
when code runs above definition:
serial console result gives following:
PTP Port 0 Role: Master
PTP Port 1 Role: Master
Setting clock source: INPUT_STREAM_DERIVED
MAAP reserved Talker stream #0 address: 91:E0:F0:0:CD:DC
PTP Port 1 Role: Slave
PTP Port 0 Role: Master
PTP sync locked

when I change acmp_talker_state = ACMP_TALKER_CONNECT; (inside above function)
then serial console output is:

PTP Port 0 Role: Master
PTP Port 1 Role: Master
Setting clock source: INPUT_STREAM_DERIVED
MAAP reserved Talker stream #0 address: 91:E0:F0:0:1E:A1
came here t connect else
CONNECTING Talker stream #0 (229700617C0000) -> Listener 0:0:0:0:0:0:0:0
entering talker handle cmd
MSRP: Register stream request 229700:617C0000
Added stream 
ID: 229700617C0000
DA:91:E0:F0:0:1E:A1:
max size: 224
interval: 1
Talker stream #0 ready
PTP Port 1 Role: Slave
PTP Port 0 Role: Master
PTP sync locked
Added stream
ID: 229700617C0000
Increasing port 1 shaper bandwidth to 17024000
Setting stream 0 1722 TX port to 1
Talker stream #0 on
PTP Port 1 Role: Master
Decreasing port 1 shaper bandwidth to 0
Talker stream #0 off
AP: MSRP_LISTENER 617C0000:1:0:0 MRP_EVENT_RECEIVE_LEAVE_ALL: MRP_LO -> MRP_UNUSED

here is the wireshark capture for above test: https://www.dropbox.com/s/0a723fq7hsigbfn/w00.pcapng?dl=0

but it seems no audio samples are received from xmos.

What is the problem with acmp_talker_state = ACMP_TALKER_WAITING ?
Maybe I am finding wrong problem. If so, Please check this out.

Thank you.

jdkoftinoff, 2015-07-01:
Do you have two xmos devices? They transport audio streams between themselves just fine - they also stream to and from Mac OSX Yosemite version with the built in avb support of the Mac.  It would be good for you to test with one of those known working configurations. 

--jeffk++

pcub, 2015-07-01:
I tested against two daisy chain last time. 
Now I am using one xmos daisy chain as talker and one linux machine as listener.

xcalger, 2015-07-01:
Hi,

I am the XMOS FAE in Austin, TX. I just added myself to this thread a couple of days ago, and I am new to the Avnu group. I would like to help. Can you summarize what you are trying to do?

Thank you.

-Chris

pcub, 2015-07-01:
Hello @xcalger, Thank you :D
My setup is one xmos daisy chain as talker and one linux machine as listener. So I want to process audio samples in pc. Where linux pc (with openavb and i210 NIC) is listener and one xmos daisy chain is as talker which records audio signal and send via ethernet to pc.

I am using AVB-Daisy-Chain-Software_1.0.6 version of xmos source code with latest openavb.
But in xmos side there is some problem. In normal xmos source code, talker does not work. You can see the <comment a day ago>.

xcalger, 2015-07-01:
@pcub

Do you have an XMOS endpoint available to try? The reason I ask is because some updates to the AVB firmware were made that came after that release of the AVB DC kit firmware that you are using. I would like to make sure that we are not chasing something that was fixed in the Endpoint firmware v6.1.1.

-Chris

pcub, 2015-07-01:
@Chris, Yes I have

jdkoftinoff, 2015-07-01:
I’m still confused as to your test setup.  The XMOS code uses the ACMP connection management protocol. Your PC needs to as well.

Regards,
Jeff

pcub, 2015-07-01:
@ Jeff, I will try.
But I have another issue that, when I use ./daemon_cl eth1 -L,
then it immediately switches to master in pc side, but in xmos side;xmos console shows xmos DC as master. How to make it work?
Cheers

jdkoftinoff, 2015-07-02:
Are the devices directly connected with each other?

--jeffk++

pcub, 2015-07-02:
@Jeff, yes one xmos and one pc are connected directly. when I use ./daemon_cl -T 255
then it becomes slave.

pcub, 2015-07-02:
here is the openavb result on pc:
root@b-desktop:~/Open-AVB# sudo clear && ./daemons/gptp/linux/build/obj/daemon_cl eth1 -R 255
Using clock device: /dev/ptp0
Starting PDelay
Switching to Slave
AsCapable: Enabled

 there is also mrpd.

root@b-desktop:~/Open-AVB/examples/simple_listener# sudo ./simple_listener -i eth1 -f e2.wav
Waiting for talker...
Msg: SJO D:C=6,P=3,V=0002,N=3 R=000000000000 QA/IN

Msg: VNE 0002 R=000000000000 VN/MT

Msg: VJO 0002 R=00229700617c QA/IN

Msg: SJO T:S=00229700617c0000,A=91e0f000c4d4,V=0002,Z=224,I=1,P=112,L=150000 R=00229700617c VO/IN

Send ready-msg...
Created file called e2.wav
Got session pcap handler.
Compiled and applied filter.

simple_listener just stuck like this.

jdkoftinoff, 2015-07-02:
So it is important to see on both ends that each end station determines the asCapable state of the other as true

The next step you are missing is ACMP

--jeffk++

pcub, 2015-07-02:
I do not see any ACMP in openavb

jdkoftinoff, 2015-07-02:
Since Open AVB is incomplete, in the meantime you could use my avdecc-cmd tool to fabricate the ACMP message (CONNECT_TX_COMMAND ) that it should have sent to the talker.

See https://github.com/jdkoftinoff/avdecc-cmd

--jeffk++

pcub, 2015-07-02:
I am sorry, avdecc of in github the descriptions are not clear to me. 
I used sudo ./avdecccmdline -l 5 and it shows log messages.
But I am not clear how to use acmp

jdkoftinoff, 2015-07-02:
You will need to refer to IEEE Std 1722.1-2013 

--jeffk++

jdkoftinoff, 2015-07-02:
One option you have to learn how AVB works is to connect up a combination of AVB devices that are complete and correct; For instance the XMOS solution with another XMOS board, or an AudioScience HONO module, or a Meyer Sound CAL speaker, or a Macintosh computer, and capture the packets that occur between the end stations and the controller with Wireshark. Wireshark will show you all of the AVB related conversations - To zoom in on the ACMP protocol, filter on the string ‘ieee1722.subtype == 0x7c’.  To filter on the media data use ‘ieee1722.subtype == 0x00’

Good luck,
Jeff

pcub, 2015-07-02:
So does it really resolve my problem?

jdkoftinoff, 2015-07-02:
Sounds like you need to do more learning about AVB.

Please review my AVB website at https://avb.statusbar.com/ <https://avb.statusbar.com/>

Specifically: https://avb.statusbar.com/page/developer-faq/ <https://avb.statusbar.com/page/developer-faq/>

and: AVDECC AES NY 2013 Presentation <https://avb.statusbar.com/page/presentations/avdecc-aes-ny-2013-presentation/>

and Best practices for AVB <https://avb.statusbar.com/page/presentations/best-practices-for-avb/>

Regards,
Jeff

pcub, 2015-07-02:
Thanks for ur effort. 
Now I am using avdecc-cmd.
To use acmp, I need to know the talker_entity_id listener_entity_id.
I am using  avdecc [verbosity] [timeout_in_ms] [network_port] acmp [message_type] [sequence_id] [talker_entity_id] [talker_unique_id] [listener_entity_id] [listener_unique_id]

sudo ./avdecc 2 250 eth1 acmp CONNECT_TX_COMMAND 0x0000
how to see the talker_entity_id and listener_entity_id?

pcub, 2015-07-02:
I realized the listener id is zero.
als0 in xmos serial console:

CONNECTING Talker stream #0 (229700617C0000) -> Listener 0:0:0:0:0:0:0:0
 here is the listener id is all zero!

pcub, 2015-07-03:
Hello @jdkoftinoff ,
I tested avdecc-cmd command with xmos serial console, it works for adp and aecp.
sudo ./avdecc 1 500 eth1 aecp AEM_COMMAND 0x0017 A0:36:9F:FF:FE:4A:52:20 00:22:97:FF:FE:00:61:7C
then it works fine.

But to use acmp I used following command:
sudo ./avdecc 1 500 eth1 acmp CONNECT_TX_COMMAND 0x0000 00:22:97:FF:FE:00:61:7C A0:36:9F:FF:FE:4A:52:20
acmp: invalid listener_entity_id: ''

1. is it enough for avdecc-cmd one xmos as talker and one PC as listener? 
(also PC is controller).

Thank you.

jdkoftinoff, 2015-07-03:
Hi

You are missing the talker unique id and the listener unique id. The usage is:

avdecc [verbosity] [timeout_in_ms] [network_port] acmp [message_type] [sequence_id] [talker_entity_id] [talker_unique_id] [listener_entity_id] [listener_unique_id] 
--jeffk++

pcub, 2015-07-03:
@jdkoftinoff 
Here is the result. Can you check please.

sudo ./avdecc 1 250 eth1 acmp CONNECT_TX_COMMAND 0x0006 00:22:97:FF:FE:00:61:7C 1 A0:36:9F:FF:FE:4A:52:20 2
Sent ACMPDU:
                                      DA:91-E0-F0-01-00-00
                                      SA:A0-36-9F-4A-52-20
                               EtherType:0x22F0
                          Payload Length:0x0038
                            message_type:CONNECT_TX_COMMAND
                                  status:SUCCESS
                     control_data_length:0x002C
                               stream_id:00-00-00-00-00-00:00-00
                    controller_entity_id:A0:36:9F:FF:FE:4A:52:20
                        talker_entity_id:00:22:97:FF:FE:00:61:7C
                      listener_entity_id:A0:36:9F:FF:FE:4A:52:20
                        talker_unique_id:0x0001
                      listener_unique_id:0x0002
                         stream_dest_mac:00-00-00-00-00-00
                        connection_count:0x0000
                             sequence_id:0x0006
                                   flags:[ ]
                          stream_vlan_id:0x0000
As I see stream dest mac is all zero.
Still same problem. no audio is receiving.

pcub, 2015-07-03:
Hello all;@pinealservo , @davidcemin, @intel-ethernet ,@jdkoftinoff, @Chris
Here is the avb command, daemon_cl, mrpd,simple_listener, acmp that I executed and
xmos serial console result:

PTP Port 0 Role: Master
PTP Port 1 Role: Master
Setting clock source: INPUT_STREAM_DERIVED
MAAP reserved Talker stream #0 address: 91:E0:F0:0:DD:64
PTP Port 0 Role: Slave
PTP Port 1 Role: Master
PTP sync locked
PTP Port 0 Role: Master
PTP threw away Sync outlier (master_diff 1391654448)
PTP Port 0 Role: Slave
PTP Port 1 Role: Master
PTP Port 0 Role: Master
PTP threw away Sync outlier (master_diff 1753844159)
PTP Port 0 Role: Slave
PTP Port 1 Role: Master
PTP sync locked
<<< 1722_1 acmp subtype >>> 
CONNECTING Talker stream #0 (229700617C0000) -> Listener A0:36:9F:FF:FE:4A:52:20
MSRP: Register stream request 229700:617C0000
Added stream:
 ID: 229700617C0000
 DA:91:E0:F0:0:DD:64:
 max size: 224
 interval: 1
Talker stream #0 ready
Added stream:
 ID: 229700617C0000
Increasing port 0 shaper bandwidth to 17024000
Setting stream 0 1722 TX port to 0
Talker stream #0 on

But in simple_listener just stucks:

Waiting for talker...
Msg: SJO D:C=6,P=3,V=0002,N=3 R=000000000000 VP/IN
Msg: VNE 0002 R=000000000000 VN/MT
Msg: SJO D:C=6,P=3,V=0002,N=3 R=00229700617c AA/IN
Msg: SJO D:C=6,P=3,V=0002,N=3 R=00229700617c AA/IN
Msg: VNE 0002 R=00229700617c QA/IN
Msg: SNE T:S=00229700617c0000,A=91e0f000dd64,V=0002,Z=224,I=1,P=112,L=150000 R=00229700617c VO/IN
Send ready-msg...
Created file called w0.wav
Got session pcap handler.
Compiled and applied filter.

wav file is generated but only 47 bytes. In simple_listener code program has to run in pcap_callback, but it does not reach there. 

What is problem?.
I read some post of simple_listener about stopping network manager,vlan and something about tcpdump: (http://sourceforge.net/p/open-avb/mailman/message/31069469/)

here is the dmesg of igb_avb:

root@b-desktop:~/Open-AVB-master/kmod/igb# dmesg |grep igb
[    2.917139] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.0.5-k
[    2.917142] igb: Copyright (c) 2007-2014 Intel Corporation.
[    2.917724] igb 0000:04:00.0: irq 45 for MSI/MSI-X
[    2.917730] igb 0000:04:00.0: irq 46 for MSI/MSI-X
[    2.917735] igb 0000:04:00.0: irq 47 for MSI/MSI-X
[    2.917739] igb 0000:04:00.0: irq 48 for MSI/MSI-X
[    2.917744] igb 0000:04:00.0: irq 49 for MSI/MSI-X
[    2.948380] igb 0000:04:00.0: added PHC on eth1
[    2.948382] igb 0000:04:00.0: Intel(R) Gigabit Ethernet Network Connection
[    2.948384] igb 0000:04:00.0: eth1: (PCIe:2.5Gb/s:Width x1) a0:36:9f:4a:52:20
[    2.948628] igb 0000:04:00.0: eth1: PBA No: G69016-002
[    2.948629] igb 0000:04:00.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[  251.771996] igb 0000:04:00.0: removed PHC on eth1
[... 39 lines cut ...]
[ 4510.596656] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[ 4544.700428] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Down
[ 4547.680463] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[ 5570.889934] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Down
[ 5574.101890] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[ 5647.888625] igb_avb 0000:04:00.0: removed PHC on eth1
[ 5649.777535] igb_avb 0000:04:00.0: irq 45 for MSI/MSI-X
[ 5649.777539] igb_avb 0000:04:00.0: irq 46 for MSI/MSI-X
[ 5649.777541] igb_avb 0000:04:00.0: irq 47 for MSI/MSI-X
[ 5649.777544] igb_avb 0000:04:00.0: irq 48 for MSI/MSI-X
[ 5649.777547] igb_avb 0000:04:00.0: irq 49 for MSI/MSI-X
[ 5649.861193] igb_avb 0000:04:00.0: added PHC on eth1
[ 5649.861197] igb_avb 0000:04:00.0: Intel(R) Gigabit Ethernet Network Connection
[ 5649.861200] igb_avb 0000:04:00.0: eth1: (PCIe:2.5GT/s:Width x1) 
[ 5649.861202] igb_avb 0000:04:00.0 eth1: MAC: a0:36:9f:4a:52:20
[ 5649.861665] igb_avb 0000:04:00.0: eth1: PBA No: G69016-002
[ 5649.868304] igb_avb 0000:04:00.0: LRO is disabled
[ 5649.868309] igb_avb 0000:04:00.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[ 5652.388619] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[ 6103.045161] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[ 6256.462609] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Down
[ 6259.718566] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[ 6415.744131] igb_avb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None

Thank you. Any suggestion will appreciated.

jdkoftinoff, 2015-07-03:
Why did you choose 1 and 2 for those parameters?

--jeffk++

pcub, 2015-07-03:
Now I chose those parameters as 0 and 1. It works well. Finally I got the audio file by commenting pcap_setfilter part inside the simple_listener. THe Audio file is somehow not perfect, there is mismatch of FS and amplitude is only between -0.2 to 0.2.

Thank you. Jeff

jdkoftinoff, 2015-07-03:
You are welcome, but why is your listener_unique_id set to 1? How many listener stream sinks do you have on the listener? 0 is the first one.

Regards,
Jeff

pcub, 2015-07-03:
My setup is xmos-DC as talker and PC as listener. Maybe I was wrong, But it worked. What does those unique id do? I am new to your tool.
Cheers.

jdkoftinoff, 2015-07-03:
But to my question, how many streams is your listener set to receive ?

--jeffk++

pcub, 2015-07-03:
oh, Now just 4 channels

jdkoftinoff, 2015-07-03:
But how many streams?

--jeffk++

pcub, 2015-07-03:
I am sorry, I am confusing channels and stream.

jdkoftinoff, 2015-07-03:
So how many streams?

--jeffk++

pcub, 2015-07-03:
4

jdkoftinoff, 2015-07-03:
And how many channels in each stream?

--jeffk++

pcub, 2015-07-03:
I am sorry, my xmos board now have 4 audio channels (microphone). So I thought stream as just channel. I hope i have one stream in which 4 channels are used.

jdkoftinoff, 2015-07-03:
That is the xmos talker, how about your listener?

--jeffk++

pcub, 2015-07-03:
same, in simple_listener, there is a macro I defined as 4 channels. So, what is your main point?

jdkoftinoff, 2015-07-03:
So I am assuming you are saying that the listener receives one four channel stream?

If so then your listener_unique_id value needs to be 0

--jeffk++

pcub, 2015-07-04:
OK. I used zeros for talker and listener uniq id. But extracted audio sampling rate are not correct.

jdkoftinoff, 2015-07-04:
What are the sampling rates?

Jeff

pcub, 2015-07-04:
it works fine when FS = 96000 in simple listener,but at 48000 it hears strange.
 in xmos max fs=96000, working fs=48000

xcalger, 2015-07-06:
> <reply+006d5445d1573890c2214d6f34177a9370997dd8fad41be592cf0000000111af12f9
> 92a169ce0550d420@reply.github.com>

@pcub,

can you confirm your last statement? 96k works, but 48k does not.
Also you are trying to set up 1 stream with 4 channels right?

further to my earlier question, do you have an endpoint board? XMOS had
two AVB kits. The DC(daisy-chain) which you mentioned, and the endpoint or
low-cost board. The firmware for the endpoint is further along than the
daisy-chain, FYI.

Can you successfully connect two XMOS boards with the same AVB stream /
channel / audio rate configurations as you are trying to do between XMOS
and your other platform?

-Chris Alger

Cc:  Chris Alger <chris@xmos.com>

view it on GitHub

pcub, 2015-07-07:
It worked at 48k. in I did not change the numberOfChannels inside the for loop for 4 channels. That is why output was not right, So after I changed the numberOfChannels to 4 inside the for loop, then it worked. I am using one DC.
I am little confusing in xmos stream and channels. How to set the stream and channels?

andrew-elder, 2015-07-07:
I think you set XMOS stream count and channels per stream via #defines and then re-build the XMOS firmware. I don't think it is changeable at run time.

xcalger, 2015-07-09:
> <reply+006d54458b9d0b35dc693b6d86b8f1433713b4be535659c792cf0000000111b3b054

/** The total number of AVB sinks (incoming streams that can be listened
to) */
#define AVB_NUM_SINKS 1
/** The total number of media outputs (typically the number of I2S output
channels). */
#define AVB_NUM_MEDIA_OUTPUTS 8

So AVB_NUM_SINKS are the number of streams,

and the number of channels per stream is AVB_NUM_MEDIA_OUTPUTS /
AVB_NUM_SINKS. It just splits the number of outputs.

Same for AVB_NUM_SOURCES and AVB_NUM_MEDIA_INPUTS

The unique_ids you were using before start at 0 for the first sink or
source stream and then 1 would be for the next sink or source stream
 

-Chris

Date:  Tuesday, July 7, 2015 at 10:20 AM

view it on GitHub

pcub, 2015-08-03:
Hello all, and @xcalger 

Now I am connecting two daisy chain board each other without AVB switch and (known as daisy chain topology).
Then I modified:
line 39: #define AVB_NUM_SOURCES 2
line 61: #define AVB_NUM_SINKS 2

and avdecc-lib shows two devices.

So I recorded using simple_listener at first just 4 channels, and 8 channels.
1. But recorded signal is not look like right.

in acmp,I used: 
sudo ./avdecc 1 250 eth1 acmp CONNECT_TX_COMMAND 0x0006 00:22:97:FF:FE:00:61:7C 1 A0:36:9F:FF:FE:4A:52:20 0

1 indicates that there is two streams.

2. How you ever worked on daisy chain topology?

Thank you.

jdkoftinoff, 2015-08-03:
Can you describe the network a bit more? Are any streams coming from the middle XMOS device? Is there a PC connected to one end as well?

Do you have one XMOS device providing 2 streams? What is your value of AVB_NUM_MEDIA_OUTPUTS?

The ACMP connect message is only for connecting one stream. You need to send two connection commands to connect two streams.

The simple_listener program only receives one stream (but may be multiple channels).

Can you tell us exactly what you are trying to accomplish first so we can make sure your configuration is correct?

Jeff

[18 more replies]

closed 2015-08-11 by pinealservo