Avnu/OpenAvnu #249: Cyclic timer event from i210?

ginkgoz, 2015-06-22:
Dear Open-AVB experts,

Maybe it is slightly off-topic, but I have got a question regarding the internal i210 timer:

We would like to use the internal timer directly and no synchronized one in order to reduce jitter for an application that needs a high-accuracy cyclic event on a Linux PREEMPT_RT system. Is it possible to somehow create a cyclic timer event based on the internal timer of the i210 in user mode using the igb_avb driver? Or would a kernel module be needed (which we would like to avoid)?

Ideally would be something like a signal, a semaphore or a callback mechanism. Or could functions such as clock_nanosleep() or timer_create() somehow mapped to the internal i210 timer?

Thanks in advance for any help!

intel-ethernet, 2015-06-22:
There are any number of methods – it primarily depends how much CPU you want to burn stalling for the event to occur vs how much latency for any of the other methods to trigger and fire.

For example, you could setup a shared memory location between the user-mode app and the driver, and setup (from user-space or the driver) a PTP TIMER interrupt (synced to the I210 timer) to fire an interrupt. The user—mode app would then go into a spin-loop waiting for the event to occur. Assuming the interrupt isn’t masked/blocked on the CPU, about 800 CPU clock cycles later you can enter into the driver interrupt service routine and atomically set the variable from 0->1. The user-mode app see this transition and exit the loop. Also helps to isolate the process onto its own CPU (see Linux ‘taskset’), steer interrupts away from the core, etc. Pros: relatively low latency – potentially < 1 usec. Cons: Burns a thread on the CPU blocking on the event.

The other methods – generating a signal (which only is delivered to the app when the thread is scheduled to execute by the OS scheduler, not injected dynamically like an ‘interrupt’ into the app), semaphore or callback all have higher latency (typically the latency of the OS scheduler).

The above methods can be extended to many other all HW designs, not just the I210 design.

Hope this helps
ekm

christopher-s-hall, 2015-06-22:
I'll add a few points/suggestions:

The user space interface to PTP clocks is defined in drivers/ptp/ptp_clock.c (kernel code).  There aren't  any timer function defined here.  It would be a significant amount of kernel and driver plumbing to implement this.

The other, purely userspace, possibility would be to use the system clock to trigger events and make adjustments based on the output of gPTP.  gPTP output clock phase and frequency outputs which could be used to discipline a system timer.

I don't claim to be an expert in the posix timer APIs but if I, for example, wanted a timer to go off every 10 milliseconds, I would:

* Calculate the first timer in terms of the system clock using the master to system phase offsets from gPTP and set the timer (timer_settime) to go off periodically based on the frequency offset (10 ms +/- small amount of compensation)
* Monitor the accumulated offset between the master and system clock.  Periodically restart the timer as above to compensate for this

This approach might be difficult to implement for very fast timers.

Eric's above discussion of signalling latency would apply here.

ginkgoz, 2015-06-24:
Thank you for your hints! We will try some of the proposed options.

pinealservo, 2015-08-11:
Closing due to no requests for clarification from the asker in a long while.

closed 2015-08-11 by pinealservo