Avnu/OpenAvnu #253: gPTP PHYdelay compensation

andrew-elder, 2015-07-06:
Need to fix delay compensation so that gPTP can correctly run on multiple platforms (not just i210) and support 100 Mbps.

wgoossens, 2015-07-06:
Hi Andrew,

I recently got my hands on a pair of i210 cards and a pair of minidsp (xmos) boards and a minidsp switch (marvell 88e7251) and I can't seem to get the i210's to talk 100mbit. I can sync two i210's with a cross-cable, I can sync a "normal" e1000 (onboard of notebook) to i210 using a cross, and the xmos devices and the switch all seem to like eachother as well, but there's no way to connect the two together...

I've also succesfully connected a e1000 (not the i210's) to an extreme switch (can't remember the exact type but it's a loud 52port 10G switch, nothing to be messing with at home ;)) at work, all without problems.

Long story short: I really like to get 100M support in, and if you can give me a place to start then I'm more then willing to take this on.

Walter

intel-ethernet, 2015-07-06:
Maybe because it is Monday after a long weekend, … but can you describe what “get the i210's to talk 100mbit” means? Doesn’t link? Links at 1Gbps? Establishes link but doesn’t send network packets.

ekm

wgoossens, 2015-07-06:
Hmm yes, I might have been a bit vague on that. "Normal" traffic works fine at both 100M and 1G, it's just the gPTP I can't get to lock at 100M. I consistently get:
*** Announce Timeout Expired - Becoming Master
For both the i210 and other e1000e when trying to communicate with either an xmos endpoint or the 100M switch.I mentioned.
Since Andrew wrote "support 100 Mbps." I thought this would be a nice time to chime in.

Walter

andrew-elder, 2015-07-07:
I may have been a bit vague too. I meant to say "add the correct PHY delay compensation for 100M operation". I did not have information that 100M gPTP did not work.

Walter, I'm not sure what to try next. The base code should work @ 100M. Maybe the PHY compensation is causing -ve delays? Perhaps we should start a different issue related to gPTP not working at 100M..... I'd like this thread to discuss the correct way to implement PHY delay compensation and that may or may not overlap with getting 100M working.

pinealservo, 2015-07-08:
@wgoossens If you add a debug line like this one to your PTP daemon, it'll print the calculated path delay every time it's updated: https://github.com/AVnu/Open-AVB/commit/5668e632c2063510b7f35e8a326c1622eab86325#diff-10349ad6d30a5fdb0a463bb547d953d0R1426

If your problem is related to this one, you'll see numbers that are out-of-spec. If the numbers look reasonable (i.e. positive and under 800) then the problem is elsewhere. Either way, you should probably open a separate issue for it so it's generally known that there's a problem and you're working on it.

pinealservo, 2015-07-08:
Regarding moving the PHY compensation: Previous discussion suggests that the right place for this is in the i210 driver, but I have a couple of questions on how it would be implemented in the network drivers more generally:

The kernel API for [timestamping](http://lxr.free-electrons.com/source/Documentation/networking/timestamping.txt) doesn't seem to say anything about the reference plane for hardware timestamps, so I'm not entirely sure all the driver authors will be on the same page as to whose job it is to compensate for different reference planes. Furthermore, in implementations where a timestamping MAC may be paired with various PHY implementations, there may not be enough information available to the MAC driver programmer to effectively calculate the delay just by link speed.

Based on evidence I've seen, it doesn't seem to be as simple as a 100 vs 1000 link. We had one showing a consistently high link delay value, which turned out to be caused by a marginal cable (had a hand-crimped end). It still said it was a gigabit link, but there was enough extra latency introduced in the PHY layer to throw the timestamp calculation out of spec.  If a hand-crimped end that was visually OK threw off the timestamp by ~1300ns or so, I have to wonder if there's some sort of feedback mechanism (other than abnormally high link delay measurements) that the PHY is having to make up for a marginal link. Even if such a mechanism is available, it seems likely that it will vary from PHY to PHY.

It seems to me that a more comprehensive solution would be to integrate PHY delay compensation into the timestamping layer itself rather than into individual drivers. HW timestamping implementations would need to differentiate between PHY-based timestamps and MAC-based timestamps, and an API to add Tx and Rx PHY delays would be available for MAC-based timestampers. The consumers of MAC-based timestamps would need to be able to receive link-state change events and recalculate PHY path delays in a PHY-dependent manner. Some API changes to the [PHY abstraction layer](http://lxr.free-electrons.com/source/Documentation/networking/phy.txt) to this effect may be necessary.

Possibly this could all be taken care of in the kernel by collaboration between the mac driver code, phy driver code, and the kernel timestamping layer. I'm not sure how reasonable that is, though, given how much variation can occur when pairing up MACs, PHYs, etc. In any case, it seems that a good solution   would require a lot of networking infrastructure development.

I think that we'll need to, in any case, continue to provide a timestamp adjustment feature in the gPTP daemon for those situations where the kernel infrastructure isn't in place to get fully-adjusted timestamps directly from the network drivers. A couple of command line parameters and config file entries seems like a reasonable solution.

jdkoftinoff, 2015-07-08:
So the command line parameters would specify TX and RX PHY offsets for both 100baseTX an 1000BaseT?

andrew-elder, 2015-07-09:
To my way of thinking, we are always going to need a way to adjust timestamps within gPTP because who knows what/when network drivers will support timestamp adjustments. Along with command line parameters and config file entries we also have the option of using the IPC interface that gPTP implements as a way to set the delays. Has anyone worked with automotive requirements? Are the starting rate ratios passed on the command line, or stored in a config file?

andrew-elder, 2015-07-09:
See also #115

wgoossens, 2015-07-09:
Hi all,

thanks for thinking with me on this. I've did a few more runs with PTP_DEBUG defined and browsed a bit through wireshark, but I'm now not at all convinced my problem has something to do with phy_delay settings. 
I Seem to be sending about ~8x as many packets as I receive from the xmos or switch. When connecting two 1g cards with a crosscable, it's ok.
I can attach some logs if anyone's interested, but I think I'll open up a separate issue when I got some more info on this (still not sure if it's the OpenAVB gPTP daemon or the 100Mb stuff I connected to it).

Walter

davidcemin, 2015-11-10:
Config file was fixed on #115

andrew-elder, 2015-11-10:
Do we need to add a "fixed" label for issues so that we can click on the milestone and see what is fixed?

davidcemin, 2015-11-10:
I think we should close them as soon as the pull requested is merged in. We could as well mark them as fixed and reference the branch in a comment. 

The pull request should come this week still.

closed 2016-03-16 by davidcemin