Avnu/OpenAvnu #254: gPTP needs to handle link state changes

andrew-elder, 2015-07-09:
Should handle link up/down and link speed change.

davidcemin, 2015-10-21:
What is the desirable behaviour for link state changes? For example on linuxptp when the link state changes, all the FSMs are re-started. Should we do something similar for gPTP?

andrew-elder, 2015-10-21:
Yes, I think so.

davidcemin, 2015-11-10:
There must be other things to do on the driver level to restart things in case of a fault. With gptp running, i brought the ethernet down (ifconfig ethx down) and I started seeing some error messages. 

After killing gPTP and restarting it, I still get errors like:
Invalid TX
ERROR at 682 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
ERROR at 1094 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response (Retrying-0), error=-72
ERROR at 1106 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response, error=-72

And these errors only go away if I remove the igb_avb driver and reinsert it again. Im not quite familiar with the igb driver, but is there a way of emptying any FIFOs we have and restarting clean the timestamping side after a fault ? Judging by the error, (-72 is set when reading the tx timestamp returns EAGAIN )

davidcemin, 2015-11-10:
same happens by changing the linkspeed.

intel-ethernet, 2015-11-10:
When you ‘down’ then ‘up’ an interface, the gPTP daemon needs to do a series of ioctl calls on the interface to re-enable the desired timestamp settings.

davidcemin, 2015-11-10:
Could you please list the ioctl calls that need to be done in this case? We would most likely need to call these calls in case of recovering automatically from a failure... (which is the main goal of this issue)

intel-ethernet, 2015-11-10:
I believe it is basically calling LinuxTimestamperGeneric::HWTimestamper_init() again, but Chris can confirm …

davidcemin, 2016-03-16:
@christopher-s-hall Can you please confirm latest comment where it says that we could solve this problem by calling  LinuxTimestamperGeneric::HWTimestamper_init() ?

davidcemin, 2016-03-17:
@kencarlino Did you guys fix this on the automotive branch already ?

kencarlino, 2016-03-17:
@davidcemin Detecting link state changes was added to the automotive profile branch. The implementation of what happens when a state changes was only focused on the automotive profile requirements. If gPTP in general has additional requirements for what to do on a link up or link down that would need to be implemented as well.

One of the known issues mentioned in the readme is that on some platforms a phantom link up or link down event can occur at a regular interval (fairly long interval of perhaps 30 seconds). This still needs to be investigated.

closed 2016-03-31 by davidcemin