Avnu/OpenAvnu #260: error in executing gptp daemon

rohan-lt, 2015-07-28:
admin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/daemons/gptp/linux/build/obj$ sudo ./daemon_cl eth2
ERROR at 634 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72

rohan-lt, 2015-07-31:
i2c_algo_bit           13197  2 igb,radeon
dca                    14642  1 igb
ptp                    18861  2 igb,e1000e
pcadmin@pcadmin-OptiPlex-7010:~$ cd avb/Open-AVB-open-avb-next/kmod/igb/
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/kmod/igb$ ls
COPYING          e1000_hw.h      e1000_manage.h  e1000_osdep.h  igb_avb.mod.o  igb_hwmon.o   igb_ptp.o          kcompat.o
e1000_82575.c    e1000_i210.c    e1000_manage.o  e1000_phy.c    igb_avb.o      igb_main.c    igb_regtest.h      LICENSE
e1000_82575.h    e1000_i210.h    e1000_mbx.c     e1000_phy.h    igb_debugfs.c  igb_main.o    igb_vmdq.c         Makefile
e1000_82575.o    e1000_i210.o    e1000_mbx.h     e1000_phy.o    igb_debugfs.o  igb_param.c   igb_vmdq.h         modules.order
e1000_api.c      e1000_mac.c     e1000_mbx.o     e1000_regs.h   igb_ethtool.c  igb_param.o   igb_vmdq.o         Module.symvers
e1000_api.h      e1000_mac.h     e1000_nvm.c     igb_avb.7      igb_ethtool.o  igb_procfs.c  kcompat.c          pci.updates
e1000_api.o      e1000_mac.o     e1000_nvm.h     igb_avb.ko     igb.h          igb_procfs.o  kcompat_ethtool.c  README
e1000_defines.h  e1000_manage.c  e1000_nvm.o     igb_avb.mod.c  igb_hwmon.c    igb_ptp.c     kcompat.h          startup.sh
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/kmod/igb$ sudo rmmod igb 
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/kmod/igb$ sudo insmod kmod/igb/igb_avb.ko
insmod: ERROR: could not load module kmod/igb/igb_avb.ko: No such file or directory
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/kmod/igb$ cd ../..
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next$ sudo insmod kmod/igb/igb_avb.ko
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next$ sudo modprobe i2c_algo_bit
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next$ sudo modprobe dca
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next$ sudo modprobe ptp
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next$ lsmod | grep igb
igb_avb               176531  0 
dca                    14642  1 igb_avb
ptp                    18861  2 e1000e,igb_avb
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next$ cd daemons/gptp/linux/build/obj/
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/daemons/gptp/linux/build/obj$ sudo ifconfig eth2 up
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/daemons/gptp/linux/build/obj$ sudo ./daemon_cl eth2
ERROR at 634 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay
ERROR at 1332 in ../../common/ptp_message.cpp: >>> Received PDelay followup but no RESPONSE exists
ERROR at 1092 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response (Retrying-0), error=-72
ERROR at 1104 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response, error=-72	
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
*** Announce Timeout Expired - Becoming Master
New Grandmaster "A0:36:9F:FF:FE:4C:AB:2B" (previous "00:00:00:00:00:00:00:00")
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
AsCapable: Enabled
ERROR at 1092 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response (Retrying-0), error=-72
ERROR at 1104 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response, error=-72	
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
Error (TX) timestamping Sync, error=-72
^CExiting on 2
pcadmin@pcadmin-OptiPlex-7010:~/avb/Open-AVB-open-avb-next/daemons/gptp/linux/build/obj$

pinealservo, 2015-08-11:
I believe this is the same issue as #265, hopefully we'll be able to help you there. I'm closing this as a duplicate.

closed 2015-08-11 by pinealservo