Avnu/OpenAvnu #261: gPTP - documentation/clarification of "rm" flag usage in LinuxTimerQueueActionArg

andrew-elder, 2015-07-29:
Clearly indicate the role of the ->rm flag in deciding whether to delete inner_arg object.

christopher-s-hall, 2015-07-29:
It might be worth investigating whether this flag is needed anymore.  I added it to avoid a signal handler deadlock problem.  I'm not sure whether it's still relevant, though.  The use of the flag would need to be checked on *both* Windows and Linux.  I wouldn't have time to do any testing at the moment, but I think it should be done (before making any big changes).

andrew-elder, 2015-07-29:
I was thinking of just adding a little documentation. That wouldn't be classified as a "big" change would it? I wasn't going to move code like Levi suggested.