Avnu/OpenAvnu #264: Simple_listener more than one stream issue

pcub, 2015-08-10:
Two xmos daisy chain (DC) boards are connected to each other (daisy chain topology) without AVB switch and one DC is connected to PC. I want this:
Two DC as talkers and PC as listener.
Listener should now record from 8 audio channels that came from two talkers.
However, the signal recorded using simple_listener is not as expected.

I changed the following part in simple_listener.c:

#define CHANNELS (8) //8 audio channels
void pcap_callback(u_char* args, const struct pcap_pkthdr* packet_header, const u_char* packet)
{
unsigned char* test_stream_id;
struct ethernet_header* eth_header;
uint32_t buf;
uint32_t frame[2] = { 0 , 0 };
int i;
(void) args; / unused /
(void) packet_header; / unused */

#if DEBUG
fprintf(stdout,"Got packet.\n");
#endif /* DEBUG*/

eth_header = (struct ethernet_header*)(packet);
#if DEBUG
fprintf(stdout,"Ether Type: 0x%02x%02x\n", eth_header->type[0], eth_header->type[1]);
#endif /* DEBUG*/

if (0 == memcmp(glob_ether_type,eth_header->type,sizeof(eth_header->type)))
{       
test_stream_id = (unsigned char*)(packet + ETHERNET_HEADER_SIZE + SEVENTEEN22_HEADER_PART1_SIZE);

#if DEBUG
fprintf(stderr, "Received stream id: %02x%02x%02x%02x%02x%02x%02x%02x\n ",
test_stream_id[0], test_stream_id[1],
test_stream_id[2], test_stream_id[3],
test_stream_id[4], test_stream_id[5],
test_stream_id[6], test_stream_id[7]);
#endif /* DEBUG*/

    if (0 == memcmp(test_stream_id, stream_id, sizeof(STREAM_ID_SIZE)))
    {
#if DEBUG
fprintf(stdout,"Stream ids matched.\n");
#endif /* DEBUG/
buf = (uint32_t) (packet + HEADER_SIZE);
for(i = 0; i < SAMPLES_PER_FRAME * CHANNELS; i += CHANNELS)
{

memcpy(&frame[0], &buf[i], sizeof(frame));
for(i=0;i<CHANNELS;i++)
{
            frame[i] = ntohl(frame[i]); 
            frame[i] &= 0x00ffffff;
            frame[i] <<= 8;    
            sf_writef_int(glob_snd_file, (const int *)frame, 1);
        }
}
    }   
}
}

1. How to implement this setup?
2. Is there any document?

Thank you.

jdkoftinoff, 2015-08-10:
Your configuration of simple listener is expecting one stream of 8 channels.

You need to configure simple listener to expect two streams of 4 channels each.

I don't think simple listener can do this.

--jeffk++

pcub, 2015-08-10:
Is there any limitation of simple_listener in theoretically?

You mean there is now way?

jdkoftinoff, 2015-08-10:
I don't know.  It would have to receive two packets per observation interval

--jeffk++

pinealservo, 2015-08-11:
@pcub simple_listener isn't meant to be a full-fledged 1722 listener. It's an example program showing you how to use the libraries provided by OpenAVB to listen to a stream. OpenAVB is not yet a turn-key solution for AVB endpoints; there's definitely some assembly and development required, and probably some digging into source code at this point too.

AVB is based on streams that are forwarded between endpoints through switches. Each XMOS daisy-chain unit actually *is* a switch as well as an endpoint, which is why your i210-based listener can potentially see streams from both devices. AVB switches do not tamper with the contents of streams, so you should *not* expect two XMOS devices sending 4-channel streams to somehow create an 8-channel stream.

To listen to a stream, a number of things must happen:

1. Devices in the network (those that will be participating in the streaming as well as any switches in-between) must join the AVB domain via gPTP and SRP protocol negotiations.

2. The stream source devices (talkers) must be told to advertise the streams that they will be sending via SRP.

3. Each switch between the source and potential listeners must verify that there are sufficient resources available to reserve bandwidth internally for each port that leads to a potential listener. When it verifies this, it forwards the talker advertisement.

4. Any device that you wish to listen to a stream needs to receive the talker's advertisement *for that stream*, as determined by its Stream ID, and make a SRP declaration that it's ready to listen.

5. Each switch between that listener back to the source needs to establish the reservation based on the talker's advertisement and the listener's declaration of readiness.

6. When the listener's declaration of readiness is received by the talker, it knows the reservation has been established and may then start to send 1722 packets with audio samples. It must pace them according to the rules established by the SRP domain it's using and the reservation it established. The rules involve a "credit-based" traffic shaper with a fixed observation interval; it may only send as much data as it reserved during each observation interval.

7. Each switch, as it receives a 1722 data packet from the direction of the talker, forwards it unmodified to each port that leads to a listener that has declared readiness and has an active reservation. It also obeys the same traffic shaping rules.

8. The listener, as it receives the data, extracts timing information from the packet as well as the sample data and uses it to aid in synchronizing the media clock (playback rate, essentially) with the talker and scheduling when the first sample should play so that all listeners play the same sample at the same time.

So you see, there's no mechanism for two talkers to have their streams merge. Each stream is an independent entity, and various streams can have different media clocks or even different media formats. Any particular endpoint *could* be programmed to listen to some number of streams and then rebroadcast them as a talker, but that is *not* what the XMOS daisy-chain devices do.

Because Ethernet (and by extension AVB) doesn't require its endpoints to know much about the overall network topology, your daisy-chained network is not fundamentally different from a typical star-topology network. Just imagine your devices are all directly connected to one another in some abstract "cloud" drawing and work from there.

I hope it is clear now that in order to get your PC listener to listen to both XMOS devices, it is going to have to interact with SRP to register for 2 streams and then set up traffic filters for each Stream ID. Because SRP runs in a separate process, you *might* be able to get this to work by running two copies of simple_listener, each configured to listen to one XMOS device's stream.  I am not certain this will work correctly, but it's closer to the approach you'll need to take than the one you have been taking.

pcub, 2015-08-11:
Thank you for your great explanation.
I watched the video that  demonstrated avb Daisy Chain topology. https://www.youtube.com/watch?v=dQtnMbM67_0 
It seems that those endpoint are only communicating inside the xmos DC's. first DC's audio can be listened to second DC's audio output channels then they connected to speaker as an output. Is not it?
Not sending whole audio channels via ethernet. 

I am not clear in this part. Two DC's audio channels are coming together (because of daisy chain topology) into only one NIC. Then how two simple_listener know which audio channels coming from which talker? using traffic filter? If so, what values to be set those filters? 
Now I am digging msrp.c.
Can you give me a guideline more precise?

Thank you.

jdkoftinoff, 2015-08-11:
Hello

The XMOS devices are operating as 3 port switches. Two ethernet ports and one internal port connected to the AVB talker and listener functionality.

All audio channels from both devices are being sent to the listener, if the listener were properly listening.  

The audio channels would be presented to the listener in separate streams.

Each stream has a unique StreamID and also has a unique Ethernet Destination Address (multicast).

Jeff

pcub, 2015-08-11:
I am sorry I linked wrong video.
this is https://www.youtube.com/watch?v=DrG4oDxl78Q

pinealservo, 2015-08-11:
@pcub I don't know how I could possibly be more precise without writing way more than one ought to write in github issue comments.

The resolution to this issue is that, no, simple listener doesn't listen to more than one stream. It probably shouldn't; it's intended to be a simple example. Use multiple instances of simple listener with different Stream IDs programmed in; if it doesn't work, we welcome pull requests with bugfixes or enhancements.

closed 2015-08-11 by pinealservo

pcub, 2015-08-11:
Thank you Pinealservo,
It works as individual streamID filter.

pcub, 2015-09-15:
Hello all,
I know that this issue is [closed].
However, for the multiple talkers (xmos dc boards), simple_listener does not receive multiple streams from multiple talkers. Even I used two different stream id on different simple_listeners running on ubuntu machine with openavb, they cannot receive streams from multiple talkers.
I also tried using qt thread to run two simple_listener simultaneously. In this case only one listener can receive and write audio file while other one cannot receive streams.