Avnu/OpenAvnu #265: Queries on talker and listener in open_avb_next

rohan-lt, 2015-08-10:
Hi all,

I have a two AVB network card (Intel i210) which is inserted into two PC`s: one for a talker and one for a listener.

So below is my hardware setup:

[PC-i210] <--------------------> [PC-i210]
 Talker							  Listener

NOTE: Talker and listener are linux machines with Ubuntu 14.04 running connected directly via Ethernet cable.

I have downloaded and built the open-avb-next branch source code obtained from 
https://github.com/AVnu/Open-AVB/tree/open-avb-next

In order to run talker and listener, I use the following below commands:

On listener machine:

sudo rmmod igb 
sudo modprobe i2c_algo_bit
sudo modprobe dca
sudo modprobe ptp
sudo insmod kmod/igb/igb_avb.ko

pcadmin@EMMYS1776:~/Downloads/AVB/daemons/gptp/linux/build/obj$ ifconfig
eth0      Link encap:Ethernet  HWaddr f8:b1:56:bf:c2:ca  
          inet6 addr: fe80::fab1:56ff:febf:c2ca/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:151194 errors:0 dropped:1300 overruns:0 frame:0
          TX packets:2381 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:17617780 (17.6 MB)  TX bytes:280057 (280.0 KB)
          Interrupt:20 Memory:f7f00000-f7f20000 

eth2      Link encap:Ethernet  HWaddr a0:36:9f:4c:ab:80  
          inet addr:192.168.34.48  Bcast:192.168.34.255  Mask:255.255.255.0
          inet6 addr: fe80::a236:9fff:fe4c:ab80/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:126683 errors:0 dropped:123588 overruns:0 frame:0
          TX packets:702 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11852126 (11.8 MB)  TX bytes:46140 (46.1 KB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:444 errors:0 dropped:0 overruns:0 frame:0
          TX packets:444 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:126434 (126.4 KB)  TX bytes:126434 (126.4 KB)

NOTE: eth2 is the AVB interface.

Run different daemons:
1. gptp daemons:
pcadmin@EMMYS1776:~/Downloads/AVB/daemons/gptp/linux/build/obj$ sudo ./daemon_cl eth2
[sudo] password for pcadmin: 
ERROR at 634 in ../src/linux_hal_common.cpp: Group ptp not found, will try root (0) instead
Using clock device: /dev/ptp0
Starting PDelay
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
ERROR at 1092 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response (Retrying-0), error=-72
ERROR at 1104 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response, error=-72	
Invalid TX
ERROR at 677 in ../../common/ieee1588port.cpp: Error (TX) timestamping PDelay request, error=-72	
ERROR at 1092 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response (Retrying-0), error=-72
ERROR at 1104 in ../../common/ptp_message.cpp: Error (TX) timestamping PDelay Response, error=-72	
Invalid TX
[... 195 lines cut ...]
listener left
listener left ...
pcadmin@pcadmin-OptiPlex-7010:~/avb/avb_codeflow/examples/simple_talker$

After this, ss.wav is created on the listener machine, it is playing properly but inbetween there are some glitches.
pcadmin@EMMYS1776:~/Downloads/AVB/examples/simple_listener$ ls -lh ss.wav 
-rw-r--r-- 1 root root 4.3M Aug 10 10:49 ss.wav

Please find below my queries:
1) Everytime I run listener and talker, the size of the output file(ss.wav) created is varying from ~2MB to ~13MB.
Why this variation in the file size? 

2)Even though listener is not stopped, talker is getting "listener leave message" and stops streaming the data.
Why is this so?

3) What are the error messages seen while running GPTP daemons on both listener and talker sides?

4) Also I see that GPTP daemons on both listener and talker are becoming Master.
"*** Announce Timeout Expired - Becoming Master"
Why is this so?If the Master is already present, then the other has to become slave. Please correct me if my understanding is wrong.
Am I missing anything here? Do I need to do any modifications in the current code?
Please let me know.

Regards 
Rohan Gupta

pinealservo, 2015-08-11:
You appear to have some sort of problem with gPTP timestamping. Error -72 seems to be an arbitrary code chosen to indicate that when asked for a timestamp, the driver told the userspace program to try asking again. After getting the same "try again" message some number of times in a row, it gives up and passes on the error code. This suggests to me that the i210 is probably not doing its AVB functionality properly, and the rest of the errors could well be due to the same root cause. The problem of both nodes thinking they're grand master is certainly due to the timestamp failure; they don't think the other one is as_capable anymore, so they both become grand master of their own isolated domains.

I've had all this working recently (last week), so I suspect it's a configuration problem. Note that the i210 retains power when your computer is in standby mode, and after getting it in a bad state I've had it stay that way through a soft reboot. I'd suggest doing a hard power cycle (pull the power plug and wait for all the lights to go out if you don't have a switch on the power supply unit) and then, instead of manually removing and reloading modules, run `./startup.sh eth2` in the kmod source directory. Then try starting the daemons again.

bharaths16, 2015-08-12:
Hi Pinealservo,

Thanks for your reply.
I am working with Rohan on the same issue.

As suggested by you, we ran startup.sh script instead of manually removing and reloading modules. Now gPTP is working fine and also one is becoming master and other is switched to slave. Also there is no loss packet in the output file(ss.wav)

The additional thing that is part of this script is the configuration the TX queues. 
Can you please give some info this configuration?

Also it would be helpful for us if you can tell us how the Grandmaster and slave are selected? We see that always a particular machine is becoming as Grandmaster and other machine as slave in our setup.

Regards,
Bharath S

pinealservo, 2015-08-12:
I am not very familiar with how the i210 queue configuration works; maybe @intel-ethernet can offer some insight. I looked briefly for some general Linux documentation on it last night, but was unsuccessful at finding anything. It hasn't come up in my desktop and embedded Linux use before. Anyway, I'm glad you were able to get things working!

It is normal for the same node in a fixed set of gPTP nodes to always be selected as grandmaster. Selection is based on a set of criteria (both fixed/intrinsic to the device and configurable ones) that provide a deterministic priority ordering. Wikipedia's article has a brief summary: https://en.wikipedia.org/wiki/Precision_Time_Protocol#Best_master_clock_algorithm

intel-ethernet, 2015-08-12:
The queue configuration via XPS isn’t required as the driver internally maps all non-AVB traffic (packets from the host OS stack) to the lowest priority queue. Prior to this internal change however, the recommended method was to remap all traffic to queue 3 (which the script demonstrates).

ekm

closed 2015-09-04 by pinealservo