Avnu/OpenAvnu #272: question regarding simple listener / talker with one I210 NIC

giladargus, 2015-08-23:
Hi all,

I am trying to set up the listener / talker demo, but I only have one I210 NIC.

I am running mrpd, gptp, and the talker and listener seems to connect (the talker prompts "got a listener...", and the listener created the wave file) but no AVB packets are running on the VLAN (I enabled VLAN 2 on the interface), and the listener file remains stuck at 44 bytes.

Is this setup supported?
Thanks in advance.

andrew-elder, 2015-08-24:
I don't think this would be expected to work. See Figure 16 on page 60 of <http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf>

giladargus, 2015-08-25:
Thanks, I'll just grab another NIC then.

andrew-elder, 2015-08-25:
Question answered

closed 2015-08-25 by andrew-elder