Avnu/OpenAvnu #276: Problems with receiving ethernet frames longer than 256 bytes

MarcinMiklasSTC, 2015-08-28:
During developing feature-avtp-pipeline branch I noticed that igb_avb has problems when receiving ethernet frames longer than 256 bytes. In this case the whole ethernet header is skipped.

See:
https://gist.github.com/MarcinMiklasSTC/85d4033e97bd9e0b6f2d

In this example you will see only packets longer than 1400 octets  (broken) and packets 64 bytes long which are not broken on i210. I have checked in other experiments, that 256 octets is the limit. 

If I remember correctly the issue is not related to AVTP traffic, it also affects best effort traffic.

intel-ethernet, 2015-08-28:
You may want to check whether “header split” is enabled (although it should only affect “IP” packets – e.g. ethertype 0800).

ekm

https://gist.github.com/MarcinMiklasSTC/85d4033e97bd9e0b6f2d

MarcinMiklasSTC, 2015-08-28:
But where/how to check it? It is some parameter of igb_avb module or #define in the code?

I'm not quite familiar with the igb_avb driver yet.

eric-mann, 2015-08-28:
Look for CONFIG_IGB_DISABLE_PACKET_SPLIT.

I'm a little confused however - on receive you are going over the standard socket API ... can you point out where in the packet trace you think there is an error? And how was the packet trace obtained - tcpdump? We have options to strip VLAN tags out of packets, but dropping ethernet headers altogether would cause serious problems (which we haven't seen).

MarcinMiklasSTC, 2015-08-28:
Compare frame 2 in packets received on i210 and packets sent in https://gist.github.com/MarcinMiklasSTC/85d4033e97bd9e0b6f2d.

I used wireshark for collecting these logs but using sockets gives the same results. Packets received on i210 has around 20 first octets missing which can be observed as "random" source Mac address in each packet. In reality all packets are sent from Texas instruments board.

It doesn't happen for packets shorter than 256 octets.

I'm seeing this issue on Ubuntu 14.04 32 bit, link speed is 100 Mbits. Maybe the issue is related to my setup. I'll try on 64 bit Ubuntu next week.

eric-mann, 2015-08-28:
Decidedly odd.

What I see is:
0000  91 e0 f0 00 fe 55 84 7e 40 2b 63 f4 81 00 00 02   .....U.~@+c.....
0010  22 f0 03 81 6b 00 ba bc 1a ba bc 1a 00 37 06 50   "...k........7.P
0020  45 7c 02 01 00 00 05 78 00 00 1c 81 9e 04 04 b7   E|.....x........

out of the TI peer becomes ...
0000  02 01 00 00 05 78 00 00 1c 81 9e 04 81 00 00 02   .....x..........
0010  04 b7 27 ff 00 36 00 00 00 08 00 00 00 00 00 00   ..'..6..........

in SW through the I210 ... about the only thing that looks valid on the inbound side is the VLAN tag 81-00 00-02. Even the trailing 1722 ethertype is messed up. 

Running with wireshark puts the I210 in promiscuous mode, which means *any* destination mac address will be received. If you don't run wireshark/tcpdump, do you still see the corrupted frames delivered to the sockets?

MarcinMiklasSTC, 2015-08-31:
Without wireshark I don't see corrupted packets. They are ignored. So instead of receiving 4000 packets per second I receive only around 300 (which are shorter than 256 octets).

Adding in `kmod/igb/Makefile`

    CFLAGS_EXTRA += -DDISABLE_PACKET_SPLIT

or

    CFLAGS_EXTRA += -DCONFIG_IGB_DISABLE_PACKET_SPLIT

doesn't help. It makes things even worse, gptp stops is no longer working (not able to get `AsCapable: enabled` state), and I do not receive any AVTP packets.

MarcinMiklasSTC, 2015-08-31:
I tested on another computer with 64 bit ubuntu and the issue is also present. Link speed doesn't matter, issue is present both on 100MB/s and 1000MB/s link speeds.

MarcinMiklasSTC, 2015-08-31:
01:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
            Subsystem: Intel Corporation Ethernet Server Adapter I210-T1
            Flags: bus master, fast devsel, latency 0, IRQ 16
            Memory at fe700000 (32-bit, non-prefetchable) [size=1M]
            Memory at fe6fc000 (32-bit, non-prefetchable) [size=16K]
            Expansion ROM at fe800000 [disabled] [size=1M]
            Capabilities: [40] Power Management version 3
            Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
            Capabilities: [70] MSI-X: Enable+ Count=5 Masked-
            Capabilities: [a0] Express Endpoint, MSI 00
            Capabilities: [100] Advanced Error Reporting
            Capabilities: [140] Device Serial Number a0-36-9f-ff-ff-2d-01-ba
            Capabilities: [1a0] Transaction Processing Hints
            Kernel driver in use: igb_avb

intel-ethernet, 2015-08-31:
I suspect the packets are corrupted before they even get to the PHY of the I210 (the HW normally filters out unrecognized destination MAC addresses unless you are in promiscuous mode). Maybe mirror a port on your switch, or put a tap in place, to look at the packets going into the I210?

MarcinMiklasSTC, 2015-08-31:
I am already using network tap. Also when I use "normal" igb driver instead of igb_avb I don't see that problem. So apparently packets get broken in igb_avb driver.

MarcinMiklasSTC, 2015-08-31:
I found IGB_RX_HDR_LEN define in igb.h which is 256  originally. If you change it to bigger value then packets of that size are received without damage. 

I am far from claiming that changing that to 1522 will fix the issue I am observing (I was looking into igb_avb driver for about an hour), but it is a tip where the problem really is.

MarcinMiklasSTC, 2015-09-01:
Using git bisect I found offending commit, before that commit issue is not visible.

commit 2fdb56a391e9e09ff22837e56d594507fbaf9250
Author: Todd Fujinaka <todd.fujinaka@intel.com>
Date:   Fri Jun 19 11:17:56 2015 -0700

    igb-avb: sync to igb-5.3.2

:040000 040000 ad9201c367750fc695d1636ef7c73320faf43da5 11a68c927b8bf3028c50da776e46a9860d97d1e6 M       kmod

htfujina, 2015-09-02:
I'm looking, but I'll need exact repro steps. Has this been a problem since the commit or is this a recent issue?

MarcinMiklasSTC, 2015-09-03:
This is how I reproduce it:
AVB talker sending big packets (bigger than 256) connected to extreme networks bridge. Connect I210 card to the same bridge. Run tcpdump or thark and observe received packets.

Issue is unrelated to the type of packets, so it doesn't need to be AVTP traffic. That's why the bridge is not needed. So you can just connect two PC with a cable and even use ping command with bigger packet size specified. That should be enough to see an issue.

pinealservo, 2015-09-04:
I have seen this problem as well, though I didn't have time to dig into it. I will try to verify the fix.

commit 4cf1c3eb (2015-09-10): Fixes: #276 - Problems with receiving ethernet frames longer than 256 bytes.

Details: Function igb_add_rx_frag taken from before commit:

AVNu/Open-AVB/2fdb56a391e9e09ff22837e56d594507fbaf9250
commit ab4e2c73 (2015-09-10): Merge pull request #278 from MarcinMiklasSTC/open-avb-next

Fixes: #276 - Problems with receiving frames longer than 256 bytes.
closed 2015-11-04 by andrew-elder