Avnu/OpenAvnu #277: Is there a way to change size of memory returnet by igb_dma_malloc_page?

MarcinMiklasSTC, 2015-08-28:
It only returns 4096 bytes of memory, which is not enough when I try to use it for use by multiple talkers.

It is enough for audio talkers, but h264 talker would like to use full ethernet frames, so in 4KB of memory I can only fit 2 of them, so that would limit the number of talkers to two.

I tried also to call igb_dma_malloc_page many times so that each talker has its own shared mem, but it turns out that it only works for 2 talkers. For 3 or more talkers I get bus errors, memory errors and whole machine crashes.

So I tried another approach to just call igb_dma_malloc_page twice so I get separate page for class A and class B streams.

And it happens to work quite well, allowing to run 7 class A streams or 14 class B streams without any crashes and moderate stream quality. With 4 class A streams quality is ok, that is that 8000 packets are sent per second (for 7 class A stream about 7950 packets are sent). All this tests were done with AAF streams, where packet size is 66 bytes.

So I would like to know if there is a way to increase number of bytes returned by igb_dma_malloc_page?

pinealservo, 2015-09-04:
Not without modifying both the igb library and the kernel module. The 'page' part of the name refers to the low-level pages of the virtual memory system, which have a fixed size. It's important to allocate at the page granularity because the memory for a buffer must be *physically* contiguous for DMA purposes rather than simply contiguous in a virtual address space, and the physical-to-virtual mappings happen at a page granularity.

It is possible in the kernel to allocate multiple contiguous pages, but it would require modifying the implementation of the IGB_MAPBUF ioctl in the kernel and the implementation of igb_dma_malloc_page(). The igb_mapbuf function in the driver would have to call alloc_pages() instead of alloc_page().

MarcinMiklasSTC, 2015-09-07:
Here is an attempt to fix that issue: #286

Now even 14 class A streams sending full frame works.

MarcinMiklasSTC, 2015-09-23:
Here is even better fix #295

closed 2015-12-10 by pinealservo