Avnu/OpenAvnu #279: igb_get_wallclock() call duration

MarcinMiklasSTC, 2015-09-03:
Mean duration of call is about 90 microseconds. Is there a way to speed this up? 

In the Open-AVB examples I see that this function is called only once, so the duration of the call is not important. In feature-avtp-pipeline we need to call it more frequently, ideally for each packet and we support many streams.

We should call it for each received packet to check its presentation time. Current Open-AVB examples doesn't care for presentation time. They just ignore it.

pinealservo, 2015-09-04:
The igb_get_wallclock() function samples the NIC's PTP timer in between two samples of the processor's own TimeStamp Counter.  Probably due to varying latency in the sampling process, it repeats the process until the two TSC values are below some minimum window size.  This is why it is a potentially time-consuming operation.

However, this is not something that you should have to do frequently, and certainly not for each received packet.  Because the result returns both a PTP timestamp and a TSC timestamp, you can use periodic calls to this function to calculate the offset and rate ratio between the TSC and PTP timer. Then, when you need the PTP time, you simply sample the TSC and apply the rate ratio and offset to convert to a PTP time.

Ideally, the PTP daemon would be providing this service for you, but at the moment it only appears to be providing the phase and frequency correction values you'll need. You should be able to grab these periodically from the PTP daemon and then use the latest values along with a normal system timestamp to calculate times on a per-packet basis, and you should never need to call igb_get_wallclock() yourself at all.

MarcinMiklasSTC, 2015-09-15:
I added gptplocaltime() funciton in  #291. It is supper fast comparing to igb_get_wallclock() and pretty accurate.

closed 2015-12-10 by pinealservo