Avnu/OpenAvnu #280: Only one process can access i210 pci card through libigb.

MarcinMiklasSTC, 2015-09-03:
Second process gets following error:

	attaching to 0000:01:00.0
	attach failed! (Device or resource busy)
	connect failed (Device or resource busy) - are you running as root?

Is it by design, or something can be done about it?

christopher-s-hall, 2015-09-03:
This is by design.  I can't remember the details, but I did extend igblib allowing multiple processes to access the register set.  Are you attempting to transmit frames from multiple processes?  I think that's still unimplemented.  I *think* it can be done however.  How many processes do you need?  I'm not signing up for the work, but I think more than 2 processes would require extensive re-architecture of the code.

MarcinMiklasSTC, 2015-09-03:
I learnt to live with that :) I run multiple threads from single process. Nevertheless the error message could me more precise. 

See also issue #284, what for we use system global semaphore when only one process can access the libigb? That suggest that multiprocess access was planned at some point.

closed 2015-12-10 by pinealservo