Avnu/OpenAvnu #281: simple_talker example uses 100% of CPU

MarcinMiklasSTC, 2015-09-03:
This is probably because there is no call to any sleep() kind of function. The pacing of the packets is done by setting proper at_time in packets and igb_xmit() (or other libigb function) function blocks until wallclock_time == at_time.

pinealservo, 2015-09-04:
I don't think there's any blocking going on in the igb functions, it's just busy-looping when resources are exhausted waiting for the next buffer to free up. I suspect it was easier this way than to try to schedule an appropriate amount of sleep time.

grandcat, 2015-09-29:
I also wondered about this issue some time ago.
My issue contains an interesting approach to reduce CPU usage: #192