Avnu/OpenAvnu #319: Setting Stream VLAN ID in the AVTP pipeline

jda72, 2015-11-09:
Hi everyone,

I am currently looking for a method to set the VLAN ID of the stream when using the AVTP pipeline. This is for testing purposes for the UNH-IOL, we are using the openavb_harness to set the parameters for a stream and we were wondering if there was some way to set the vlan id in the ini file , via the command line, or if there is a simple code addition to add support for configuration of vlan id?

Thanks,
Jackson Corson
UNH-IOL GRA

kencarlino, 2015-11-09:
Hi Jackson,

I don’t have a direct answer for you but perhaps I can point you in the right direction and ask a question myself.

Typically the vlan ID is obtained from MPR (SRP) for the SR Class being used. Marcin Miklas (copied on this email)  did the MRP integration with the AVTP Pipeline so he may have a direct answer for you.

Do you know how to set the vlan ID for other usages within OpenAVB?

Regards,
Ken

Kenneth Carlino
Principal Software Engineer
CONNECTED SERVICES

55 Cambridge St. Suite 101 Burlington, MA 01803
Office: +17812700422
www.harman.com<http://www.harman.com>
[cid:105180.png@84d027c8.4492a980]

Confidentiality Notice: This e-mail message, including any attachments, is for the sole use of the intended recipient(s) and may contain confidential and privileged information. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender by return e-mail and destroy all copies of the original message.

jda72, 2015-11-09:
Hi Ken,

We currently are running the avtp-pipeline without SRP so we can control SRP using violett. Previously we created an implementation by using libigb. So our previous implementations set the vlan id when creating the packets. So to answer your question, I am not familiar with setting the vlan id for other usages within open avb, other than the creation of packets.

Thanks,
Jackson Corson
UNH-IOL GRA

kencarlino, 2015-11-09:
Hi Jackson,

I took a quick look at the code and if SRP isn’t being used it might be that the vlan ID is only set via a define in the file:

openavb_talker_no_endpoint.c

#define SR_CLASS_A_DEFAULT_VID 2
#define SR_CLASS_B_DEFAULT_VID 2

If you are seeing a vlan ID of 2 then these defines may be being used. If you don’t see a vlan ID of 2 there may be something that isn’t as integrated/exposed as it should be.

Regards,
Ken

[cid:83f6ab.png@8b344ef1.429a1aa7]

MarcinMiklas, 2015-12-08:
See pull request #328 for the fix.

pinealservo, 2015-12-10:
The pull request was merged.

closed 2015-12-10 by pinealservo