Avnu/OpenAvnu #335: How SRP domain detected?

marvinyane, 2015-12-30:
Hi,
  I have some doubts in SRP domain detection.

  In **IEEE Std 802.1BA-2011  6.4 Detection of AVB domains**

> *The operation of MSRP allows a Port to make a declaration on its attached LAN of the SR classes that it supports, and for each supported SR class, the priority code point value that it associates with that SR class.By comparing its own supported SR classes and priority values with those received from a neighboring device attached to the Port, a determination can be made as to whether the Port is at the boundary of the SRP domain for a given SR class or not.*

And in **IEEE 802.1Q-2014 35.2.4 MSRP Attribute Propagation**

>*There is no MSRP MAP function for Domain attributes. MSRP simply declares the characteristics of the SR classes that are supported on the Bridge Port regardless of what has been learned from Domain registrations on other Bridge Ports.*

If the Domain attribute cannot propagate,  how the end point **comparing its own supported SR classes and priority values with those received from a neighboring device attached to the Port**?

And I dont find the source code of declare the Domain attribute for endpoint, but only receive the attributes, and if the domain attribute cannot propagate, why the endpoint receive these messages?

If any wrong please tell me, thanks very much.

marvinyane, 2015-12-31:
According to Spec IEEE802.1Q-2014 34.2 Detection of SRP domains

>Bridges detect the edge of an SRP domain by observing SRP behavior. If a Bridge receives SRP
registrations using a particular priority, then it is reasonable to believe that they are being received from an SRP capable device; the SRP engine can therefore signal which Ports of a Bridge are at the boundary of an SRP domain. The per-port, per-SR class, SRPdomainBoundaryPort parameter determines whether a Port is considered to be at the edge of an SRP domain or within the core of the domain, as defined in 35.1.4. This parameter is controlled by the operation of SRP.

__"If a Bridge receives SRP registrations using a particular priority“__  does it mean endpoints should register some attribute  to the attached bridge?

Forgive me so many doubts, it is really difficult to me  to understand these Spec.

i will greatly appreciate to any help.

intel-ethernet, 2015-12-31:
Probably best to read at least section 5.2.5 of http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs_v1.0.pdf ... which talks about the issues with domain detection and registration …

Fwiw.
ekm

marvinyane, 2016-01-04:
Thanks so much for your response.

According to section 5.2.5, Page 63
>Figure 19 and Figure 20 illustrate the local interactions with the SRP service for a Talker and a Listener. Figure 19 also shows the case where a Domain attribute is advertised from the peer and is available from SRP when the client starts up. However, by the protocol, there is no rule which side advertises Domain attributes first. Endpoints should never wait for the bridge to declare first, as a simple endpoint-to-endpoint connection without an intermediate bridge will never establish a Domain and therefore never stream any data. The endpoint should declare a Domain attribute with AVB default values, and only when the peer advertises a mis-match, the end node can decide whether to re-declare the Domain attribute with the non-default settings from the peer. It is assumed endpoint-to-endpoint configurations will resolve to the AVB default values for the Domain attributes.

**The endpoint should declare a Domain attribute with AVB default values**

But when I test with simple_talker and simple_listener in endpoint-to-endpoint connection mode,  neither of talker and listener  declare Domain attribute first.

I have read the source code,  and I does not find anywhere to declare a domain attribute before received the peer's  attribute propagation. I know it is very rude, but could you please show me the source code of declare the default domain attribute?

i am very appreciated for all your any response.

Best Regards.

andrew-elder, 2016-01-04:
Marvin,

When a client "connects" to the mrpd daemon, the daemon will asynchronously 
notify the client of relevant MRP events. The client application should 
check the Domain notifications being received from mrpd to determine 
whether or not domain attribute properties are as expected. Note - 
examples/mrp_client/mrpdhelper.c contains code for decoding notification 
strings sent from mrpd.

Regarding steps to declare a Domain, have a look at mrpValidate.c in 
examples/mrp_client/.

I hope the above makes senses.

- Andrew

marvinyane, 2016-01-05:
Hi Andrew, 

Thank you so much for your help.

After look mrd_client example code,  I have get how to declare domain attribute. Thanks so much.

In endpoint-to-endpoint mode,  one of each endpoint should send S+D to decalre domain attribute before send S?? to get domain attribute,  so I donot think the simple_talker and simple_listener work correct in endpint- to-endpint mode. Maybe when endpoints connect with AVB bridge, the bridge would declare domain attribute first, and then endpoints wait for getting domain attribute in LAN.

In avtp_pipeline library, it call *mrp_get_domain* first to get the domain attribute too.

*/lib/avtp_pipeline/platform/Linux/endpoint/openavb_endpoint_osal.c* line *274*.
```
    err = mrp_get_domain(&class_a_id,&a_priority, &a_vid, &class_b_id, &b_priority, &b_vid);
	if (err) {
		AVB_LOG_DEBUG("mrp_get_domain failed");
		goto error;
	}

	AVB_LOGF_INFO("detected domain Class A PRIO=%d VID=%04x...", a_priority, (int)a_vid);
	AVB_LOGF_INFO("detected domain Class B PRIO=%d VID=%04x...", b_priority, (int)b_vid);

	err = mrp_register_domain(&domain_class_a_id, &domain_class_a_priority, &domain_class_a_vid);
	if (err) {
		AVB_LOG_DEBUG("mrp_register_domain failed");
		goto error;
	}

	err = mrp_register_domain(&domain_class_b_id, &domain_class_b_priority, &domain_class_b_vid);
	if (err) {
		AVB_LOG_DEBUG("mrp_register_domain failed");
		goto error;
	}
```

But as According to section 5.2.5, Page 63 said, 

>> Endpoints should never wait for the bridge to declare first, as a simple endpoint-to-endpoint connection without an intermediate bridge will never establish a Domain and therefore never stream any data.

Maybe I misunderstand with the document, and please correct me if I am wrong.

Best regards.

pinealservo, 2016-01-05:
I think you understand correctly how things should work. Endpoints are required to declare domain attributes and *then* wait to hear the domain attribute of their peer. If they match, then the domain is established and they may proceed. If not, the endpoint may re-declare its domain attribute to match its link peer and then continue, or follow some sort of error condition behavior.

Unfortunately, while simple_talker and simple_listener are effective in showing how some of the API calls can be used, they are not fully compliant AVB endpoints. Your guess that when used with an AVB switch, an endpoint can usually get away with just waiting for the switch to declare its domain attribute is also correct. The simple_talker/simple_listener programs evolved from test utilities, and full compliance was not as important to that role as the ability to make them do their work without compliant peers and correct implementations of other protocols in place.

Fortunately, we now have the avtp_pipeline code, which *is* designed to be a compliant implementation. Although it's a much larger code base, it should be referred to for examples relating to how components work together instead of simple_talker/simple_listener.  However, we're open to contributions that can help simple_talker/simple_listener be better examples of compliant behavior; we just don't currently have the resources to focus on that right now.

marvinyane, 2016-01-08:
Thank you so much for your detailed explanation, and now I have no doubt about this.

Thanks for everyone, I really appreciate for your patience.

closed 2016-01-08 by marvinyane