Avnu/OpenAvnu #337: Synchronize system clock with gPTP

davidcemin, 2016-01-07:
There is a question on sourceforge Open-AVB that im replicating below so more people can contribute. 

Source: http://sourceforge.net/p/open-avb/discussion/general/thread/5feefbce/?limit=25#bf17

> Hi,
> I am experimenting a testbed on AVB. We have Intel I210 ethernet cards supporting gPtp with Extreme > Network switched supporting AVB. The Liunx/I210 end systems are detected by the Ethernet switches. > So gPtp is operational.
> My question is simple: how to synchronize a Linux System clock with the one provided by the gptp protocol ?
> With ptp4l, a phc2sys command was used to synchronize the system clock with ptp. Is there any ?equivalent command for gPtp/AVB protocol ?
> Best regards,
> Laurent George
> Researcher at ESIEE Paris engineering school.

christopher-s-hall, 2016-01-07:
Thanks David for reposting this. There’s no reason that phc2sys can’t be used with the Open-AVB gPTP daemon. The default behavior of gPTP is different from that of ptp4l.  In order, to replicate that behavior you should use the –S flag when starting the daemon. I don’t regularly test this code (syntonization) and I’m unsure of the state of it. In the past, it has worked quite well with back-to-back I210s and several switches that I tested with.

Chris

davidcemin, 2016-01-07:
Thank you Chris for answering it. 
Closing the issue.

closed 2016-01-07 by davidcemin