Avnu/OpenAvnu #346: Can I use multiple I210 NIC on same HOST

ksrivastava11, 2016-02-06:
Hi,

Let me start by telling you about my setup: I am having 2 Intel I-210 Gigabit Ethernet Adapter, PCIe Mini cards my Host machine(ubuntu 14.04), I have a Talker application that opens /dev/igb_avb (driver interface provided by the adapter after loading kernel module) file and does some stuff over it. I want to run two instances of this application on single host.

Problem: when I am trying to load the kernel module for these card through insmod igb_avb.ko, it creates two network interfaces like eth0 and eth1 but creates single device node under /dev directory with /dev/igb_avb and both the cards are using the same network driver. I verified with lspci -v. 

Is it possible to create /dev/igb_avb0 for NIC 1 and /dev/igb_avb1 for NIC 2 through igb driver code changes or through any udev rules, so that I can open specific device from Talker Application.

Thanks,
Krishna

intel-ethernet, 2016-02-07:
HI Krishna –

You are correct that the two devices each get a netdev instance (e.g. eth0 and eth1). Each of these netdev instances are created by the single loaded igb driver, which also creates the igb_avb device interface. If you want to access both device instances, you need to bind to each physical device by passing in the explicit PCI device path to the igb_attach() API.

As an example -  if one device is “0000:01:00.0” and another device is “0000:02:00.0” you call igb_attach() twice with each path.
igb_attach(“0000:01:00.0”, &dev1);
igb_attach(“0000:02:00.0”, &dev2);

You can also call igb_attach() from different threads or processes as well. Each igb_attach initializes the ‘adapter’ member in the ‘device_t *pdev’ structure allocated by the application and passed into the various routines.

Hope this helps.
ekm

davidcemin, 2016-07-19:
I assume this issue can be closed, since its a question and it was apparently answered. Comments?

andrew-elder, 2016-07-20:
Question answered.

closed 2016-07-20 by andrew-elder