Avnu/OpenAvnu #353: OPEN AVB FOR ARM

PRABIII, 2016-03-03:
Hi All
Im Working On Open AVB project for Ls1021A-IOT board. I need to cross compile this Open AVB for LS1021A-IOT , I'm able to compile mrpd and mrpd_client for LS1021A-IOT, but I'm not able to compile gptp for LS1021A-IOT. IS this gptp on the Open AVB will support ARM architecture, For compiling it for ARM, do I need to do any customisation in CMAKE or MAKE file.?
THANKS IN ADVANCE

andrew-elder, 2016-03-03:
I'm not aware that anyone has run gptp from OpenAVB on ARM. You may have to do some investigation on how best to do that.

PRABIII, 2016-03-14:
Thank you andrew-elder, foud ptp for ARM ie linuxptp. Also cross compile gPTP  and MRPD daemon for ARMv7(LS1021A-IOT ).

davidcemin, 2016-03-21:
Is this question still open ?

tobma71, 2016-05-04:
Hi
I also have a question to the ARM support of the openAVB stack.
My goal is to use the simple talker and listener on an ARMv7 architecture. 
So for that I need to compile the mrpd and gptp daemon for an ARMv7 architecture. I read that it should be possible to compile the mrpd daemon for ARM but it is not possible for me. On my host computer (ubuntu 12.04) is the ARM cross compiler tool-chain installed and I set:

export ARCH=arm
export CROSS_COMPILE=/home/user/gcc-linaro-arm-linux-gnueabihf-4.7-2012.11-20121123_linux/bin/arm-linux-gnueabihf-gcc

But when I use the "make" command the mrpd daemon is always compiled for the x86 architecture. Can anybody help me to compile mrpd and gptp for ARM? 

Thanks

andrew-elder, 2016-05-04:
I just go 
make ARCH=arm
in the /Open-AVB/daemons/mrpd directory and it compiles for ARM just fine.

There is another open issue on getting gptp to compile on ARM.