Avnu/OpenAvnu #354: Request Creation of Feature Branch: feature-ap-for-gptp

kencarlino, 2016-03-07:
As the title states, requesting the creation of a feature branch that will get AVnu Automotive Profile changes to gPTP.  

See the attached Readme file for changes and additions in this contribution offer. This contribution will be from Harman

[README_AVNU_AP.txt](https://github.com/AVnu/Open-AVB/files/161851/README_AVNU_AP.txt)

If agreeable this feature branch should be created from the open-avb-next branch.

Regards,
Ken

christopher-s-hall, 2016-03-07:
An automotive branch is a fine idea from my perspective, but “ap” makes me think access point not automotive profile.

andrew-elder, 2016-03-07:
What is the intended timing? We have a bunch of gPTP bugfixes posted to feature-gptp-next that are awaiting further work before being pulled in to open-avb-next. I'm thinking we are going to generate significant conflict resolution challenges if we have both efforts active at once. If we aim to wrap up the gPTP bugfixes by the end of this month, will that work out?

Note, I'm 100% in favor of the list of changes presented in the README file. I didn't see anything that looked like a red flag. The README reads a little bit like this work has already been completed. Is that the case?

Thanks,
Andrew

kencarlino, 2016-03-07:
Yes the work has already been completed. Sorry for not doing a sooner notice, we needed approval for the submission and wanted to get that before suggesting that we could contribute this work. 

The naming of the branch could be different, it was just a suggestion. 

Once the branch was created ideally I could issue a pull request within a week. A 3rd party will be pulling from that commit so ideally it wouldn't be too long out. 

Regards,
Ken

kbstanton, 2016-03-08:
Since "ap" collides with "access point", an 802.11 term (and relevant to
gptp), to avoid confusion, perhaps "feature-avnuautoprof-for-gptp" or
something similarly less cryptic.

andrew-elder, 2016-03-08:
My thinking is to just go ahead and create the branch. The task of reviewing potential gPTP conflict resolution exists whether or not this work is present on OpenAVB and I believe the AVnu/AVB/TSN  "cause" is better served having code publicly available in the AVnu OpenAVB for other folks to review/test and contribute to. I have just created https://github.com/AVnu/Open-AVB/tree/feature-gptp-avnu-automotive-profile. In case it isn't clear, the branch name is gptp-avnu-automotive-profile.

closed 2016-05-17 by pinealservo