Avnu/OpenAvnu #362: Minimum hardware requirements

MikeLoh, 2016-03-24:
We have been watching AVB (Time-Sensitivie networking) for our professional audio applications for some time now. I was really excited to see the launch of Open AVB and have been monitoring progress for a while.

We are now engaging in projects that will ultimately utilize AVB - been skimming at this point and plan to dig deeper over next week - but need a quick answer to questions:

What is the minimum system requirements to implement AVB?
Is it just having 1588 and then having sufficient memory and processing power to execute?
Is there a list of people of group on which processors they have successfully implemented AVB stack on?

Thanks for any time anyone can share.

andrew-elder, 2016-03-24:
I don't think there is a list of processors. I'm sure folks on the group have worked with various processors. Apart from 1588, you need to have a way to launch media packets at the correct time to comply with FQTSS. Multiple packet queues for different protocol types is also a useful feature.

MikeLoh, 2016-03-24:
Thanks for the quick reply - it does seem to be 1588 is the key, other elements are function of firmware and enough processing power to handle.

I will continue to review, but your input is very timely and helpful - thanks again. Hopefully more from me, including contributions, soon.

eric-mann, 2016-03-24:
I'd recommend reviewing the AVnu public available best practices paper at http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs-v1.3-SWAVI-Updates_Final_Approved_Clean.pdf . 

There are several ways to partition an AVB solution to meet various power, price and performance targets. It also depends on what you want to do - whether you want to sample and launch data into an established stream, receive/transform/send back out a stream, or simply receive a stream. 

Shameless plug, but you can implement an AVB sw stack using virtually any available Intel processor + Intel I210 adapter.

MikeLoh, 2016-03-24:
Eric - most excellent white paper - thanks. Answers a lot for me int this preliminary step.

pinealservo, 2016-03-24:
Hi Mike,

If you are interested in microcontroller-based solutions, there are also
offerings from Freescale (now NXP), Atmel, and Broadcom that have
more-than-just-1588 AVB support. XMOS has some interesting AVB support as
well, although their architecture is significantly different from the
traditional microcontroller. While TI hasn't (that I'm aware of, anyway)
got as thorough AVB support in their hardware yet, the versions of their
SoCs that contain PRU-ICSS modules provide a dedicated coprocessor that you
can offload a lot of timing-sensitive network-related work to. There may be
others as well that I haven't investigated yet.

In my experience, implementing AVB with *only* 1588 support provided by
hardware is difficult and limited in the features that can be reliably
supported. If you need to do tight synchronization of recovered media
clocks, don't forget to look for features that allow accurate timestamping
of media clock edges and fine-grained PLL adjustment of media clocks. These
are starting to get integrated into AVB-supporting SoCs, but for 1588-only
devices you'll most likely need to add some external circuitry and make
clever use of other peripherals for measurement. There is significant
effort and risk involved in getting 1588-only hardware to fully support
AVB! Be careful about your cost analysis here, and make sure you understand
what's involved in meeting your clock recovery and packet scheduling
requirements in terms of hardware external to the MCU and software
expertise (i.e. sufficient knowledge of time/clock related issues and
control systems to come up with reasonable filtering, recovery, and
scheduling algorithms given the limitations of the hardware you'll be
working with).

Also note that raw processing power and memory are not necessarily the most
important aspects to a good implementation in the absence of hardware
assistance.  AVB is essentially about making real-time guarantees, so
consistency of execution timing and minimal task switching overhead become
increasingly important as you start taking away hardware support for
precisely-timed scheduling. You can bring a fast embedded applications
processor to its knees with task switching and scheduling overhead and
still end up with a difficult job of keeping the timing in-bounds, while a
simpler implementation on a dedicated slower but more predictable
microcontroller-style core could easily handle the load.  More consistency
in execution can also mean a smaller memory footprint due to reduced
buffering needs.

I've been involved in implementing and working with AVB stacks for various
hardware and software combinations for nearly 10 years now, ranging across
fully FPGA-based custom solutions, XMOS-based solutions, SoCs with only
1588 support, SoCs with more thorough AVB support, and OpenAVB on Intel
with the i210. I *strongly* recommend getting some hardware with better
than 1588-only support for your first attempt. With the current landscape,
the NXP iMX 6.SoloX and iMX 7, XMOS xCore-200, and Broadcom BCM5830x are
the ones with the most thorough AVB-on-Gigabit support I'm aware of. If
Fast Ethernet is sufficient, the Atmel SAM V7x and earlier XMOS-based
solutions look good. I believe Intel has some embedded solutions that would
be one of the most straightforward ways to directly use OpenAVB, but I
don't know the details about which parts you'd want to look at. And of
course OpenAVB + Intel PCs with i210 NICs are the main development platform
for OpenAVB at present.

If you are planning to use OpenAVB, we are excited to hear that!  But keep
in mind that it will need some work to adapt to platforms other than Intel
with i210. Different components have different ranges of current OS
portability as well. Everything will build in Linux, the gPTP module will
also build in Windows, and the AVTP Pipeline code is extracted from a stack
that has been used across many platforms but I'm not sure whether it will
build anywhere but Linux in its current state without some extra code. We
welcome contributions to the project!

        --Levi

vitorroriz, 2017-06-20:
Hello guys, I am very excited to start investigating and using AVB. In my current setup I am using a NXP IMX7 dual Sabre board which has more-than-just-1588 AVB support as mentioned by @pinealservo . I have no experience yet but I am willing to learn. I would like to ask if someone has previous experiences implementing AVB on other plataform than x86 and Intel i210 controller. I am now just trying to run the simple_talker/simple_listener example. I can cross compile the mrpd daemon for ARM but I am not sure which gPTP daemon to use for my ethernet controller, since the one from the OpenAvnu project seems to support  ($(ARCH),I210) and ($(ARCH),IntelCE). I read that if other architecture is defined "linux_hal_generic" will be used but I don't really know if it is enough.

Any hints and documentation about the implementation on this kind of setup is very welcome.
Thanks!

pinealservo, 2017-06-20:
The gPTP daemon will work with the i.MX 7Dual SoC; we are using one now as well. I think the build options that using the i210 and IntelCE arch flags would enable are things that are now supported with standard kernel APIs, so you shouldn't be missing out on anything.

You won't be able to get simple_talker or simple_listener to work; they rely on the i210 and the custom driver/library for it. The newer NXP ethernet MAC in the i.MX 6SoloX, i.MX 7, and i.MX 8 has analogous hardware capabilities to those in the i210, but there is currently no support in the upstream kernel driver for them and we don't have any custom kernel code in the OpenAvnu repository aside from the i210 driver, which is not done in a way that would be acceptable as part of the upstream kernel and is otherwise way behind the current upstream one.

It is possible to get the avtp_pipeline code (see lib/avtp_pipeline) working on the i.MX 7Dual, and it is far more complete than simple_talker/simple_listener, but it is a bit more complicated to cross-compile and set up and we don't have a generic build solution for cross-compile situations, since there are a lot of different ways people want to do it. There have been a few recent issues from others that have been cross-compiling for i.MX platforms that you should browse through. I will look at improving the cross-compile situation and documentation, but it may take a while.

We are not currently taking advantage of any of the special hardware support of the NXP ethernet core aside from 1588 timestamping, but it's possible to get reasonably good performance with one or two streams with only software shaping. There's work in progress to get some standard kernel interfaces in place for things like timed-launch and credit-based shaping, but currently most hardware vendors (mostly in the automotive space right now) seem to be content to provide custom kernel drivers to work with proprietary AVB stacks rather than push to get a common API implemented.

If you get to the point where you are comfortable with the project and want to help get some kernel support for the NXP ethernet features in places, we'd love to collaborate on that.

ahogen, 2017-06-20:
Also note some of @jdkoftinoff's notes on the subject of AVB/TSN hardware here: 

- http://avb.statusbar.com/page/developer-faq/#what-are-the-minimum-requirements-for-an-avb-device

vitorroriz, 2017-06-21:
Thanks for the answers. I am still in the phase of reading the documentation (I am going through http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs-v1.3-SWAVI-Updates_Final_Approved_Clean.pdf) since this is all very new to me, thus, unfortunately I don't have the necessary knowledge to collaborate yet @pinealservo.

I am not sure if it is the best thread for that, but I finally managed to compile the gPTP daemon for the imx7 Dual Sabre board. It may look pretty obvious for the readers here but for other beginners like me it might help:

I overwrote the definitions for CC and CX in the makefile program for the ones I was using (arm-linux-gnueabi-gcc and arm-linux-gnueabi-g++). However the target device couldn't execute the daemon. It was due to shared libraries problems. I then tried to cross compiled with "-static" but then a lot of missing references appeared. I figured out that the right toolchain to use for the imx7  would be armhf instead. Here is the explanation: https://stackoverflow.com/questions/31929092/trying-to-run-a-cross-compiled-executable-on-target-device-fails-with-no-such-f

So I just installed the new toolchains:

**$ sudo apt-get install gcc-arm-linux-gnueabihf
$ sudo apt-get install g++-arm-linux-gnueabihf**

I could then execute the gPTP daemon in the target.